espressif/esp-idf #15587: esp_http_server: async_handlers example running error.

lala-666, 2025-03-17:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.5-dev-2375-g1f46216a72

### Espressif SoC revision.

ESP32-PICO-V3-02 (revision v3.0)

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom Board

### Power Supply used.

USB

### What is the expected behavior?

Run async_handlers example without error.

### What is the actual behavior?

1. assert failed after long_asyn()
`assert failed: block_next tlsf_block_functions.h:161 (!block_is_last(block))`
2. browser does not response.

### Steps to reproduce.

1. copy async_handlers example to a new folder.
2. build example.(I shortened the duration in long_async() to avoid wasting time).

> for (int i = 0; i < 10; i++) {

4. flash and run.
5. Accessing the 'long handler' link in the browser.
6. crash happened in function httpd_req_async_handler_complete().

### Debug Logs.

```plain
I (4378) example_connect: Got IPv4 event: Interface "example_netif_sta" address: 192.168.31.199
I (4528) example_connect: Got IPv6 event: Interface "example_netif_sta" address: fe80:0000:0000:0000:4e75:25ff:fea4:47f4, type: ESP_IP6_ADDR_IS_LINK_LOCAL
I (4528) example_common: Connected to example_netif_sta
I (4538) example_common: - IPv4 address: 192.168.31.199,
I (4538) example_common: - IPv6 address: fe80:0000:0000:0000:4e75:25ff:fea4:47f4, type: ESP_IP6_ADDR_IS_LINK_LOCAL
I (4548) example: starting async req task worker
I (4558) example: starting async req task worker
I (4558) example: Starting server on port: '80'
I (4568) example: Registering URI handlers
I (4568) main_task: Returned from app_main()
I (4678) example: uri: /
I (7038) example: uri: /long
I (7038) example: invoking /long
I (7038) example: running: /long

assert failed: block_next tlsf_block_functions.h:161 (!block_is_last(block))

Backtrace: 0x40081cdd:0x3ffc7aa0 0x40088aa1:0x3ffc7ac0 0x400909b1:0x3ffc7ae0 0x4008dd2c:0x3ffc7c00 0x4008dba4:0x3ffc7c20 0x400824b7:0x3ffc7c40 0x400909e1:0x3ffc7c60 0x400e29a6:0x3ffc7c80 0x400d88b0:0x3ffc7ca0 0x40089545:0x3ffc7cd0
--- 0x40081cdd: panic_abort at /root/esp-idf/components/esp_system/panic.c:468
0x40088aa1: esp_system_abort at /root/esp-idf/components/esp_system/port/esp_system_chip.c:87
0x400909b1: __assert_func at /root/esp-idf/components/newlib/src/assert.c:80
0x4008dd2c: block_next at /root/esp-idf/components/heap/tlsf/tlsf_block_functions.h:161 (discriminator 1)
 (inlined by) block_link_next at /root/esp-idf/components/heap/tlsf/tlsf_block_functions.h:168 (discriminator 1)
 (inlined by) block_mark_as_free at /root/esp-idf/components/heap/tlsf/tlsf_block_functions.h:176 (discriminator 1)
 (inlined by) tlsf_free at /root/esp-idf/components/heap/tlsf/tlsf.c:630 (discriminator 1)
0x4008dba4: multi_heap_free_impl at /root/esp-idf/components/heap/multi_heap.c:233
0x400824b7: heap_caps_free at /root/esp-idf/components/heap/heap_caps_base.c:75
0x400909e1: free at /root/esp-idf/components/newlib/src/heap.c:34
0x400e29a6: httpd_req_async_handler_complete at /root/esp-idf/components/esp_http_server/src/httpd_txrx.c:677
0x400d88b0: worker_task at /root/esp-projects/async_handlers/main/main.c:150
0x40089545: vPortTaskWrapper at /root/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:139

ELF file SHA256: 0ff225788

Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 271414342, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:6368
load:0x40078000,len:15880
load:0x40080400,len:3860
```

### More Information.

My English is not good, please forgive me

It's the same on my another chip esp32s3.

In httpd_req_async_handler_complete():

> free(ra->scratch);

it freeed ra->scratch.

bnt it httpd_req_async_handler_begin():
ra->scratch is not malloc ,it was a shallow copy.

then i commented `free(ra->scratch);`

chip does not crash after httpd_req_async_handler_complete() return. But I can not open a session and refreshing the web page. Open a new tag caused a stack overflow.

> I (4548) example: starting async req task worker
I (4558) example: starting async req task worker
I (4558) example: Starting server on port: '80'
I (4568) example: Registering URI handlers
I (4568) main_task: Returned from app_main()
I (44978) example: uri: /long
I (44988) example: invoking /long
I (44988) example: running: /long
I (297528) example: uri: /long
I (297528) example: invoking /long
I (297528) example: running: /long
I (316488) example: uri: /
I (316498) example: uri: /long
I (316498) example: invoking /long
I (316508) example: running: /long
W (317508) httpd_txrx: httpd_sock_err: error in send : 104

> ***ERROR*** A stack overflow in task async_req_worke has been detected.

I increased stack size.But it doesnot help.

> I (4558) example: starting async req task worker
I (4558) example: starting async req task worker
I (4568) example: Starting server on port: '80'
I (4568) example: Registering URI handlers
I (4568) main_task: Returned from app_main()
I (4658) example: uri: /long
I (4658) example: invoking /long
I (4658) example: running: /long
I (21738) example: uri: /
I (21748) example: uri: /long
I (21748) example: invoking /long
I (21748) example: running: /long
W (22748) httpd_txrx: httpd_sock_err: error in send : 104
W (23748) httpd_txrx: httpd_sock_err: error in send : 128
I (24308) example: uri: /long
I (24308) example: invoking /long
I (24308) example: running: /long
W (24748) httpd_txrx: httpd_sock_err: error in send : 128
W (25748) httpd_txrx: httpd_sock_err: error in send : 128
W (26748) httpd_txrx: httpd_sock_err: error in send : 128
W (27748) httpd_txrx: httpd_sock_err: error in send : 128
W (28748) httpd_txrx: httpd_sock_err: error in send : 128
W (29748) httpd_txrx: httpd_sock_err: error in send : 128
W (30748) httpd_txrx: httpd_sock_err: error in send : 128
W (31748) httpd_txrx: httpd_sock_err: error in send : 128
W (31748) httpd_txrx: httpd_sock_err: error in send : 128

It seems that the previous session is not closed.

I read the esp http server code in detail.
Also in httpd_req_async_handler_complete():

    ra->sd->for_async_req = false;

It make it false but does not close the session.
So i tried adding one line:

    httpd_sess_close_lru((struct httpd_data *)ra->sd->handle);

It works perfectly no matter how i request.

The final httpd_req_async_handler_complete():

```
esp_err_t httpd_req_async_handler_complete(httpd_req_t *r)
{
    if (r == NULL) {
        return ESP_ERR_INVALID_ARG;
    }

    struct httpd_req_aux *ra = r->aux;
    ra->sd->for_async_req = false;
    httpd_sess_close_lru((struct httpd_data *)ra->sd->handle);
    // free(ra->scratch);
    // ra->scratch = NULL;
    // ra->scratch_cur_size = 0;
    // ra->scratch_size_limit = 0;
    free(ra->resp_hdrs);
    free(r->aux);
    free(r);

    return ESP_OK;
}
```

lala-666, 2025-03-17:
This is my first time posting on GitHub, please give me more guidance

hrushikesh430, 2025-03-24:
Hi @lala-666, I reproduced your issue on my local machine. Thank you for reporting this issue, we will trying to fix it as early as possible. This issue can be fixed by increasing the stack size `ASYNC_WORKER_TASK_STACK_SIZE` to 3k or more. And remove the `ra->scratch` buffer related lines from the `httpd_req_async_handler_complete`

closed 2025-05-06 by espressif-bot

hrushikesh430, 2025-05-06:
Hi @lala-666 , The issue is being fixed, you may refer [this](https://github.com/espressif/esp-idf/commit/b6d4fa2c2eb77dc6613ea8635c384670b38a2a75) change on master. Thank you.