espressif/esp-idf #15588: RMT Module - Guru Meditation Error: Core  / panic'ed (Cache disabled but cached memory region accessed).

Dawid-Wawrzynczyk, 2025-03-17:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

ESP-IDF 5.4.0-dirty

### Espressif SoC revision.

ESP32-S3

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

ESP32-S3-WROOM-1

### Power Supply used.

USB

### What is the expected behavior?

When I click a button the motor should start rotating at the expected speed using the RMT Module - When I click the button again the motor should come to a stop.

### What is the actual behavior?

The motor does indeed start and stop accordingly. The issue is; sometimes, if the timing is right, during quick ON / OFF switch, I get the:

> Guru Meditation Error: Core  / panic'ed (Cache disabled but cached memory region accessed). 
> MMU entry fault error occurred while accessing the address 0x3ceb7c2c (invalid mmu entry)

There is no clear pattern to when it happens, sometimes after few clicks, sometimes after many clicks.

### Steps to reproduce.

It's really hard to clearly point out where this bug resides. The final stack line error is always at 'rmt_encoder.c:210' with few lines following below it present in the debug log.

I'll present the chain of functions that have anything to do with the error.

Here is the main task that receives a command and performs the RMT communication:
`void IRAM_ATTR motor_move_task(void *pvParameters)
{
    void **data = (void**)pvParameters;
    motor_handle_t *motor_handle = (motor_handle_t*)data[0];
    SemaphoreHandle_t semaphore = *((SemaphoreHandle_t*)data[1]);
    bool *new_command_flag = ((bool*)data[2]);

    while (1)
    {
        if (xSemaphoreTake(semaphore, portMAX_DELAY) == pdTRUE)
        {
            uint32_t base_frequency = calculate_frequency(motor_handle->steps_per_rev, 1);

            if ((motor_handle->command == STOP || motor_handle->command == INSTANTSTOP) && motor_handle->motor_state == STATIONARY) continue;

            if (motor_handle->motor_state == MOVING)
[... 460 lines cut ...]
D (999) rmt: new tx channel(0,0) at 0x3c0b7f70, gpio=21, res=10000000Hz, hw_mem_base=0x60016800, dma_mem_base=0x0, dma_nodes=0x0, ping_pong_size=48, queue_depth=5
I (1009) MC: Initialised Motor using MC
I (1019) LVGL: Starting LVGL task
I (1019) gpio: GPIO[7]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (1029) ST7796: Installed panel IO
I (1029) st7796: version: 1.3.1
I (1029) gpio: GPIO[9]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (1039) st7796_general: LCD panel create success, version: 1.3.1
I (1279) ST7796: Successfully Installed Display Driver
I (1309) ST7796: Added LVGL Display Instance
I (1309) gpio: GPIO[14]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:2
E (1309) gpio: gpio_install_isr_service(502): GPIO isr service already installed
I (1319) ST7796: Initialised touch controller XPT2046
I (1319) ST7796: Successfully Finished GUI Initialisation
```

### More Information.

I am using the LVGL 9.3 for the user interface with ST7796 LCD Display over SPI.

The log output has a stack from LVGL for some unknown reason, it doesn't connect to my code anywhere and to be honest I don't even know why it's started appearing there, It didn't use to, but now I can't really get rid of it, so I'm including it just in case.

I've disabled the 'CONFIG_SPI_MASTER_ISR_IN_IRAM' in case it's the problem with the LCD, I have also tried decorating every single method that I showcased here in this report with the 'IRAM_ATTR', the stack might change a little at the entry point, but the rest is the same.

I've spent days trying to fix this bug and anyone here willing to help is my last resort.

suda-morris, 2025-03-17:
So you enabled  `RMT_ISR_IRAM_SAFE` Kconfig? If so, all RMT ISR callbacks must be placed in the IRAM, including your customized encoder function. Have you placed the `rmt_encode_stepper_motor_curve` into the IRAM?

Dawid-Wawrzynczyk, 2025-03-17:
> So you enabled `RMT_ISR_IRAM_SAFE` Kconfig? If so, all RMT ISR callbacks must be placed in the IRAM, including your customized encoder function. Have you placed the `rmt_encode_stepper_motor_curve` into the IRAM?

So I've tried enabling the 'CONFIG_RMT_ISR_IRAM_SAFE' but I disabled it because I started having memory problems when it's enabled, basically it seemed like I don't have enough memory anymore and I wasn't sure why, and yes I've tried putting the 'rmt_encode_stepper_motor_curve' into IRAM using the 'IRAM_ATTR' pretty much the same thing still happens.

Dawid-Wawrzynczyk, 2025-03-17:
> So I've tried enabling the `CONFIG_RMT_ISR_IRAM_SAFE` but I disabled it because I started having memory problems when it's enabled, basically it seemed like I don't have enough memory anymore and I [...]

Actually, update on the `CONFIG_RMT_ISR_IRAM_SAFE`, I played around with it, a little and the out of memory was only to do with a memory leak. I have fixed this now but the main issue still persists.

suda-morris, 2025-03-18:
I would suggest you to strip your project to only contain the RMT + stepper motor driver and get rid of the LVGL layer.

Dawid-Wawrzynczyk, 2025-03-18:
> I would suggest you to strip your project to only contain the RMT + stepper motor driver and get rid of the LVGL layer.

I'll try that, thank you.

Kainarx, 2025-08-22:
Close the issue due to inactivity. Feel free to reopen if the issue still happens or with more updates.

closed 2025-08-22 by suda-morris