espressif/esp-idf #15591: panic on esp_event_loop_delete on esp_websocket_destroy

kzoltaan, 2025-03-18:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.4-dirty

### Espressif SoC revision.

ESP32-WROOM32E

### Operating System used.

Linux

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom board

### Power Supply used.

External 5V

### What is the expected behavior?

No panic

### What is the actual behavior?

Sometimes on esp_websocket_client_destroy I get panic. No communication in progress...
 

### Steps to reproduce.

1. Create a websocket connection
2. Destroy.
3. Go to step 1.

### Debug Logs.

```plain
assert failed: xQueueSemaphoreTake queue.c:1713 (pxQueue->uxItemSize == 0)

Backtrace: 0x40081c49:0x3fff0df0 0x4008aba1:0x3fff0e10 0x40091cb5:0x3fff0e30 0x4008b312:0x3fff0f50 0x4008b43a:0x3fff0f90 0x40186d65:0x3fff0fb0 0x400e9d4a:0x3fff0ff0 0x400ea00b:0x3fff1010
--- 0x40081c49: panic_abort at /esp/v5.4/esp-idf/components/esp_system/panic.c:454
0x4008aba1: esp_system_abort at /esp/v5.4/esp-idf/components/esp_system/port/esp_system_chip.c:92
0x40091cb5: __assert_func at /esp/v5.4/esp-idf/components/newlib/assert.c:80
0x4008b312: xQueueSemaphoreTake at /esp/v5.4/esp-idf/components/freertos/FreeRTOS-Kernel/queue.c:1713 (discriminator 1)
0x4008b43a: xQueueTakeMutexRecursive at /esp/v5.4/esp-idf/components/freertos/FreeRTOS-Kernel/queue.c:835
0x40186d65: esp_event_loop_delete at /esp/v5.4/esp-idf/components/esp_event/esp_event.c:644
0x400e9d4a: destroy_and_free_resources at /digitalio/trunk/managed_components/espressif__esp_websocket_client/esp_websocket_client.c:432
0x400ea00b: esp_websocket_client_task at /digitalio/trunk/managed_components/espressif__esp_websocket_client/esp_websocket_client.c:1142
```

filzek, 2025-03-18:
show you code!@

kzoltaan, 2025-03-19:
I have a system what handle multiple connections. Sometimes it crash on connection stop. I have statemachines for connection handling, and I run it from esp_timer in every 10 ms. This is stable, it handling wifi lost events, start/stop from other modules, but sometimes it crash. Free heap is approx 44k.  blocking delay not used.... This is the related code.
```

    static void WSRun(WS_CONNECTION_T * client )
    {
    
        switch (client->State)
        {
        case WSC_IDLE:
    
            break;
        case WSC_CONNECT:
    
            if (!client->Enabled)
                client->State = WSC_IDLE;
    
            if (!timer_expired(&client->ConnectTimer))
                break;
    
            if (SysInfo.Wifi.STAState != STA_CONNECTED)
                break;
    
            LOGI(TAG, "Connecting: %s", client->Name);
    
            InitWSClient(client);
    
            client->State = WSC_CONNECTING;
            client->Counter = 0;
            break;
    
        case WSC_CONNECTING:
    
            break;
    
        case WSC_CONNECTED:
    
            if (client->EnabledPeriodical)
            {
                if (timer_expired(&client->PollTimer))
                {
                    timer_stop(&client->PollTimer);
                    WS_Upload(client);
                }
                else if (client->SendRequest)
                {
                    WS_Upload(client);
                    client->SendRequest = false;
                }
            }
    
            break;
    
        case WSC_SHUTDOWN:
            LOGI(TAG, "Shutting down... %s", client->Name);
    
            // to wait for finish sending if pending....
            timer_set(&client->ConnectTimer, 100); 
            client->State = WSC_DISCONNECT;
            break;
    
        case WSC_DISCONNECT:
    
            if (!timer_expired(&client->ConnectTimer))
                break;
    
            LOGI(TAG, "Game over... %s", client->Name);
    
    ////panic here.........
            esp_websocket_client_destroy(client->WSClient);
    
            LOGI(TAG, "esp_websocket_client_destroy %s", client->Name);
    
            timer_stop(&client->PollTimer);
    
            if (client->Enabled)
            {
                timer_set(&client->ConnectTimer, 500);
                client->State = WSC_CONNECT;
            }
            else
            {
                client->State = WSC_IDLE;
            }
    
            break;
        }
    }
    
    void WSClient_Run(void)
    {
    
        for (uint8_t i = 0; i < WS_CLIENT_COUNT; i++)
        {
            WSRun(&WSClient[i]);
        }
    }
    
    
    static void periodic_timer_callback(void *arg)
    {
    WSClient_Run();
    }
    
    static void InitClock()
    {
        esp_timer_create(&periodic_timer_args, &periodic_timer);
        esp_timer_start_periodic(periodic_timer, 10000);
    }

```

kzoltaan, 2025-03-20:
Sometimes I get this error... 
```
--- 0x4008b64e: xt_utils_get_raw_core_id at /esp/v5.4/esp-idf/components/xtensa/include/xt_utils.h:52
 (inlined by) spinlock_acquire at /esp/v5.4/esp-idf/components/esp_hw_support/include/spinlock.h:89
 (inlined by) xPortEnterCriticalTimeout at /esp/v5.4/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:479
0x4008bd86: vPortEnterCritical at /esp/v5.4/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/include/freertos/portmacro.h:567
 (inlined by) xEventGroupClearBits at /esp/v5.4/esp-idf/components/freertos/FreeRTOS-Kernel/event_groups.c:499
0x400ea40d: esp_websocket_client_destroy at managed_components/espressif__esp_websocket_client/esp_websocket_client.c:788 (discriminator 1)
```
I have no idea why....

kzoltaan, 2025-03-26:
> LOGI(TAG, "%s 1 null? %s ",client->Name, client->WSClient ? "Nope" : "Yep");         esp_websocket_client_destroy(client->WSClient);
> LOGI(TAG, "%s 2 null? %s ",client->Name, client->WSClient ? "Nope" : "Yep");

output is:

> 
> (06:49:24.409) WS-CLIENT: (WSRun:433) User-WS 1 null? Nope 
> I (06:49:24.637) WS-CLIENT: (WSRun:435) User-WS 2 null? Nope 

The `destroy_and_free_resources(client);` in `esp_websocket_client_destroy` ends with ` client = NULL;`

When I made it NULL, also indicate exception....

What am I doing wrong?

filzek, 2025-03-26:
the log do not show too much information, but could be relared to process not yet finished in the websocket task, maybe there is something still running and not yet true cleaned at all.
did you manage to set your struct to a intermediate state and only nullify it later, like 3 seconds to do it? so, its more accurate to understand what actions are being done inside the websocket.

glmfe, 2025-04-17:
Hi @kzoltaan, First of all, thank you for submitting this ticket. Unfortunately, I wasn't able to reproduce the described behavior. Could you please provide a complete reproducer for this? You can fork the esp-protocols repository and extend the esp_websocket_client example to reproduce the issue.