espressif/esp-idf #15592: [esp32s3] i2c slave callbacks not triggering and unexpected data read from TX buffer

Daphne-HU, 2025-03-18:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.4-690-gd4aa25a38e-dirty

### Espressif SoC revision.

ESP32-S3 (QFN56) (revision v0.1)

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-S3-DevKitC-1 V1.0

### Power Supply used.

USB

### What is the expected behavior?

- i2c slave callback get triggered on request and on receive after setting i2c_slave_register_event_callbacks
- the tx buffer reads 1 

### What is the actual behavior?

- Callbacks do not get triggered
- master reads {1,61,1,1} in repeat

### Steps to reproduce.

// set I2C_ISR_IRAM_SAFE and I2C_ENABLE_SLAVE_DRIVER_VERSION_2 in menuconfig to true
// (clock stretching is disabled)

// H
typedef enum {
	I2C_RX,
	I2C_TX
} i2cSlaveEvent_t;

using message_t = std::array<uint8_t, 32>;

class CasambiService {
  public:
	CasambiService();
	TaskHandle_t StartServiceTask();
	void Start();

  private:
	static void I2CTask(void* arg);
	static bool IRAM_ATTR i2cSlaveRequestCB(i2c_slave_dev_handle_t i2c_slave, const i2c_slave_request_event_data_t* evt_data, void* arg);
	static bool IRAM_ATTR i2cSlaveReceivedCB(i2c_slave_dev_handle_t i2c_slave, const i2c_slave_rx_done_event_data_t* evt_data, void* arg);
	void i2cSlaveWrite(message_t message, uint8_t messageLength);

	static void timerCallback(void* arg);

	uint32_t m_writeLen{};
	static DRAM_ATTR QueueHandle_t m_i2cQueue;
	i2c_slave_config_t i2cSlaveConfig{};
	i2c_slave_dev_handle_t m_i2cHandle;
};

// CPP
CasambiService::CasambiService() {
	m_i2cQueue = xQueueCreate(20, sizeof(i2cSlaveEvent_t));

	gpio_config_t io_conf{};
	io_conf.pin_bit_mask = (1ULL << I2C_SLAVE_PENDING_SIGNAL_PIN);
	io_conf.mode = GPIO_MODE_OUTPUT;
	io_conf.pull_up_en = GPIO_PULLUP_DISABLE;
	io_conf.pull_down_en = GPIO_PULLDOWN_DISABLE;
	io_conf.intr_type = GPIO_INTR_DISABLE;
	ESP_ERROR_CHECK(gpio_config(&io_conf));

	i2cSlaveConfig.i2c_port = -1;
	i2cSlaveConfig.sda_io_num = I2C_SLAVE_SDA_PIN;
	i2cSlaveConfig.scl_io_num = I2C_SLAVE_SCL_PIN;
	i2cSlaveConfig.clk_source = I2C_CLK_SRC_DEFAULT;
	i2cSlaveConfig.send_buf_depth = 100;
	i2cSlaveConfig.receive_buf_depth = 100;
	i2cSlaveConfig.slave_addr = I2C_SLAVE_ADDRESS;
	i2cSlaveConfig.addr_bit_len = I2C_ADDR_BIT_LEN_7;
	i2cSlaveConfig.intr_priority = 3;
	i2cSlaveConfig.flags.allow_pd = 0; // Allow power-down mode
	i2cSlaveConfig.flags.enable_internal_pullup = 1;
	i2cSlaveConfig.flags.broadcast_en = 0;

	ESP_ERROR_CHECK(i2c_new_slave_device(&i2cSlaveConfig, &m_i2cHandle));

	i2c_slave_event_callbacks_t cbs = {
		.on_request = i2cSlaveRequestCB,
		.on_receive = i2cSlaveReceivedCB,
	};
	esp_err_t cb_err = i2c_slave_register_event_callbacks(m_i2cHandle, &cbs, &m_i2cQueue);
	if(cb_err != ESP_OK) {
		ESP_LOGE(LOG_TAG, "Failed to register I2C callbacks: %s", esp_err_to_name(cb_err));
	}
}

TaskHandle_t CasambiService::StartServiceTask() {
	ESP_LOGI(LOG_TAG, "Start casambi service");
	TaskHandle_t taskHandle = nullptr;
	xTaskCreate(I2CTask, "i2cTask", 2 * 2048, this, 6, &taskHandle);
	return taskHandle;
};

void CasambiService::Start() {
	const esp_timer_create_args_t timerArgs = {
		.callback = &timerCallback,
		.arg = this,
		.dispatch_method = ESP_TIMER_TASK,
		.name = "timer",
		.skip_unhandled_events = false};
	esp_timer_handle_t timer;
	ESP_ERROR_CHECK(esp_timer_create(&timerArgs, &timer));
	ESP_ERROR_CHECK(esp_timer_start_periodic(timer, 1000'000));
}

void CasambiService::I2CTask(void* arg) {
	CasambiService* context = (CasambiService*)arg;
	i2cSlaveEvent_t message;
	for(;;) {
		if(xQueueReceive(context->m_i2cQueue, &message, portMAX_DELAY)) {
			switch(message) {
			case I2C_TX: {
				ESP_LOGI(LOG_TAG, "TX callback");
				// write following byte to master
				break;
			}
			case I2C_RX: {
				ESP_LOGI(LOG_TAG, "RX callback");
				break;
			}
			default: {
				static_assert(true, "received unknown task callback ID in casambiService");
				break;
			}
			}
		}
	}
	vTaskDelete(NULL);
}

bool IRAM_ATTR CasambiService::i2cSlaveRequestCB(i2c_slave_dev_handle_t i2c_slave, const i2c_slave_request_event_data_t* evt_data, void* arg) {
	QueueHandle_t* queue = (QueueHandle_t*)arg;
	i2cSlaveEvent_t evt = I2C_TX;
	BaseType_t xTaskWoken = 0;
	xQueueSendFromISR(*queue, &evt, &xTaskWoken);
	return xTaskWoken == pdTRUE ? pdTRUE : pdFALSE;
}

bool IRAM_ATTR CasambiService::i2cSlaveReceivedCB(i2c_slave_dev_handle_t i2c_slave, const i2c_slave_rx_done_event_data_t* evt_data, void* arg) {
	QueueHandle_t* queue = (QueueHandle_t*)arg;
	i2cSlaveEvent_t evt = I2C_RX;
	BaseType_t xTaskWoken = 0;
	xQueueSendFromISR(*queue, &evt, &xTaskWoken);
	portYIELD_FROM_ISR();
	return xTaskWoken == pdTRUE ? pdTRUE : pdFALSE;
}

void CasambiService::i2cSlaveWrite(message_t message, uint8_t messageLength) {
	ESP_LOGI(LOG_TAG, "Wrote on buffer (numbers):");
	for(int i = 0; i < messageLength; i++) {
		ESP_LOGI(LOG_TAG, "%d", message[i]); // Print character as a number
	}
	esp_err_t err = i2c_slave_write(m_i2cHandle, message.begin(), messageLength, &m_writeLen, 100);
	if(err != ESP_OK) {
		ESP_LOGE(LOG_TAG, "Error writing to i2c write buffer: %s", esp_err_to_name(err));
		return;
	}
	if(m_writeLen > 0) {
		m_writeLen--;
		ESP_LOGI(LOG_TAG, "write first byte");
		gpio_set_level(I2C_SLAVE_PENDING_SIGNAL_PIN, 1);
		vTaskDelay(pdMS_TO_TICKS(1));
		gpio_set_level(I2C_SLAVE_PENDING_SIGNAL_PIN, 0);
	}
}

void CasambiService::timerCallback(void* arg) {
	CasambiService* context = (CasambiService*)arg;
	message_t test = {1, 1};
	context->i2cSlaveWrite(test, 2);
}

### Debug Logs.

```plain
nativeFlashTool flash_monitor /dev/tty.usbserial-14220 esp32s3
Flashing and starting monitor on port /dev/tty.usbserial-14220...
esptool.py --chip esp32s3 -b 460800 --port /dev/tty.usbserial-14220 --before default_reset --after hard_reset write_flash --flash_mode dio --flash_freq 80m --flash_size 2MB 0x0 bootloader/bootloader.bin 0x10000 M5StackTest.bin 0x8000 partition_table/partition-table.bin
esptool.py v4.8.1
Serial port /dev/tty.usbserial-14220
Connecting....
Chip is ESP32-S3 (QFN56) (revision v0.1)
Features: WiFi, BLE, Embedded PSRAM 8MB (AP_3v3)
Crystal is 40MHz
MAC: 7c:df:a1:e0:c8:b8
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Flash will be erased from 0x00000000 to 0x00005fff...
Flash will be erased from 0x00010000 to 0x00061fff...
Flash will be erased from 0x00008000 to 0x00008fff...
SHA digest in image updated
Compressed 21024 bytes to 13403...
Wrote 21024 bytes (13403 compressed) at 0x00000000 in 0.6 seconds (effective 288.6 kbit/s)...
Hash of data verified.
Compressed 335504 bytes to 188931...
Wrote 335504 bytes (188931 compressed) at 0x00010000 in 4.9 seconds (effective 542.4 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 103...
Wrote 3072 bytes (103 compressed) at 0x00008000 in 0.1 seconds (effective 486.2 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
I (2343) CasambiSevice: ote on buffe (numbes):
I (2343) CasambiSevice: 1
I (2343) CasambiSevice: 1
I (2343) CasambiSevice: wite fist byte
I (3343) CasambiSevice: ote on buffe (numbes):
I (3343) CasambiSevice: 1
I (3343) CasambiSevice: 1
I (3343) CasambiSevice: wite fist byte
I (4343) CasambiSevice: ote on buffe (numbes):
I (4343) CasambiSevice: 1
I (4343) CasambiSevice: 1
I (4343) CasambiSevice: wite fist byte
I (5343) CasambiSevice: ote on buffe (numbes):
I (5343) CasambiSevice: 1
I (5343) CasambiSevice: 1
I (5343) CasambiSevice: wite fist byte
I (6343) CasambiSevice: ote on buffe (numbes):
I (6343) CasambiSevice: 1
I (6343) CasambiSevice: 1
I (6343) CasambiSevice: wite fist byte
etc..
```

### More Information.

Observing the oscilloscope, I can see that the master reads a byte after the pending signal pin is triggered. However, only a single byte is read, as the TX callback does not get triggered. As a result, the buffer accumulates two data points but transmits only one, causing it to gradually fill up.

Additionally, the oscilloscope reveals that the data read by the master consistently follows the sequence {1,61,1,1}. However, I am unsure of the origin of the value 61.

For context: In this master configuration, {1,1} functions as a ping. When the last two bytes of {1,61,1,1} are transmitted, I receive a {1,2} pong response from the master. However, also the RX callback is not triggered in this process.

Daphne-HU, 2025-03-18:
Yes you are right!
And I need the slave to communicate with the master back and forth, the master sends one or more bytes (The master sends all bytes in a separate write, why they implement it like that is a question) And when I have data to send to the master I trigger the pending signal pin and then the master always reads one byte, so If I have a 3 byte message I need to trigger the pin 3 times

mythbuster5, 2025-04-11:
[patch.txt](https://github.com/user-attachments/files/19699123/patch.txt)

@Daphne-HU I added a non-stretch version support and related docs, would you please have a try to see whether this can meet your requirement?

mythbuster5, 2025-04-11:
[patch.txt](https://github.com/user-attachments/files/19701744/patch.txt)

@Daphne-HU Try this.

Alvin1Zhang, 2025-12-02:
@Daphne-HU Thanks for reporting, would you please help if any updates for the issue? Thanks.