espressif/esp-idf #15598: Peripheral drivers using esp_gpio_reserve() should have a way to invoke esp_gpio_revoke() for the peripheral.

kenh-sq, 2025-03-18:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Using release/v5.4 branch:

In our system, we install/remove the UART driver as needed. When installing we'll do calls like:
```c
uart_param_config( ... );
uart_set_pin( UART_NUM, PIN_USED_FOR_TX, PIN_USED_FOR_RX, UART_PIN_NO_CHANGE, UART_PIN_NO_CHANGE );
uart_driver_install( ... );
```

When removing, we do the following:
```c
uart_driver_delete( UART_NUM );
gpio_reset_pin( PIN_USED_FOR_TX );
gpio_reset_pin( PIN_USED_FOR_RX );
```

If we install, remove, then install again, we get the following warnings:
```
W (4599) uart: GPIO 16 is not usable, maybe used by others
W (4600) uart: GPIO 17 is not usable, maybe used by others
```

This is due to `uart_set_pin()` making use of `esp_gpio_reserve()` to mark the pin usage. However, there is no readily exposed way to invoke `esp_gpio_revoke()` for those pins. We could include esp_private/esp_gpio_reserve.h in our file and invoke it directly, but since it's supposed to be "private", the implication is we shouldn't be doing that.

`gpio_reset_pin()` could invoke `esp_gpio_revoke()` for the pin number passed to it. Is there a reason it shouldn't? If there's a good enough reason not to do so, I'd like there to be an opposing function to `uart_set_pin()` that would "unset" all pins used by a given UART instance. I know there are other drivers that make use of `esp_gpio_reserve()` that also do not have a way to "revoke" them.

Thank you.

skibu, 2025-03-22:
I just ran into this same problem, but for LEDC GPIO bits that are configured for PWM output. Turns out that when done with a channel there is no way to release it and also release the GPIO bit.

For now I call esp_gpio_revoke() manually in my code.

But it seems that there should be an official way to release a LEDC channel and have esp_gpio_revoke() be called by the system code instead of by the application code.

songruo, 2025-03-24:
We will include a call to `esp_gpio_revoke` inside `gpio_reset_pin`, so that it can be a general workaround to release the IO, regardless the peripheral drivers.

closed 2025-03-27 by espressif-bot