espressif/esp-idf #15599: I2C V5.4 Crash when using i2c_master_execute_defined_operations

rsnbusa, 2025-03-18:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

V5.4 release

### Espressif SoC revision.

ESP32S3

### Operating System used.

macOS

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

esp32s3

### Power Supply used.

USB

### What is the expected behavior?

First the downloaded version 5.4/release in VSC with espressif extension and it  **DOES NOT** have the i2c_master_execute_defined_operations call and types in the i2c_master.c file. Neither does 5.3.2. **Result** -> Wont compile.

 The github web page for the 5.4 release DOES HAVE it. _Whats wrong?_

So I then download the Master (development version 5.5.0 apparently) version which has the calls and struct types, etc and **hence compiles**.

A standard setup of i2c device/bus is done except for the address which is documented as to NONE (per instructions documentation I2C espressidf).

Set up the ops[] array to sequence required to read 10 bytes.

Using the i2c_master_execute_defined_operations(...) option to control the i2c due to speciall address management by the chip MB85RC16.
The operation is to read the first 10 bytes from address 0. **Result CRASH**

### What is the actual behavior?

```
#include <stdio.h>
#include <stddef.h>
#include <stdint.h>
#include <string.h>

#define I2C_NUMBER                    I2C_NUM_0
#define ESP_SLAVE_ADDRF        (0x50)
#define WRITE_BIT                       I2C_MASTER_WRITE              
#define READ_BIT                        I2C_MASTER_READ               

#include "driver/i2c_master.h"

i2c_master_dev_handle_t idev_handle;

void initi2c(int sda, int scl)
{
	i2c_master_bus_config_t i2c_mst_config;
	bzero(&i2c_mst_config,sizeof(i2c_mst_config));
[... 271 lines cut ...]
D (642) intr_alloc: Connected src 42 to int 8 (cpu 0)
Passed i2c_new_master_bus
Passed i2c_master_bus_add_device
Ready 0x3fceada4. --> my message of dev_handle pointer

assert failed: xQueueGenericReset queue.c:333 (pxQueue)

Backtrace: 0x403759d5:0x3fc983c0 0x40379c01:0x3fc983e0 0x40380565:0x3fc98400 0x40379d6a:0x3fc98520 0x42018edd:0x3fc98540 0x42019254:0x3fc98570 0x420193d0:0x3fc985b0 0x4201a027:0x3fc98600 0x4200a244:0x3fc986b0 0x4200a289:0x3fc98770 0x4201aa64:0x3fc98790 0x4037a639:0x3fc987c0
--- 0x403759d5: panic_abort at /Users/rsn/esp/master/esp-idf/components/esp_system/panic.c:468
0x40379c01: esp_system_abort at /Users/rsn/esp/master/esp-idf/components/esp_system/port/esp_system_chip.c:87
0x40380565: __assert_func at /Users/rsn/esp/master/esp-idf/components/newlib/src/assert.c:80
0x40379d6a: xQueueGenericReset at /Users/rsn/esp/master/esp-idf/components/freertos/FreeRTOS-Kernel/queue.c:333 (discriminator 1)
0x42018edd: s_i2c_send_commands at /Users/rsn/esp/master/esp-idf/components/esp_driver_i2c/i2c_master.c:470
0x42019254: s_i2c_transaction_start at /Users/rsn/esp/master/esp-idf/components/esp_driver_i2c/i2c_master.c:631
0x420193d0: s_i2c_asynchronous_transaction at /Users/rsn/esp/master/esp-idf/components/esp_driver_i2c/i2c_master.c:847
0x4201a027: i2c_master_execute_defined_operations at /Users/rsn/esp/master/esp-idf/components/esp_driver_i2c/i2c_master.c:1377
0x4200a244: i2c_master_read_slave at /Users/rsn/Desktop/i2cfail/main/main.c:74
0x4200a289: app_main at /Users/rsn/Desktop/i2cfail/main/main.c:84
0x4201aa64: main_task at /Users/rsn/esp/master/esp-idf/components/freertos/app_startup.c:208
0x4037a639: vPortTaskWrapper at /Users/rsn/esp/master/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:139
```

### More Information.

Thanks for any help.

mythbuster5, 2025-03-19:
It's nothing to do with `i2c_mst_config.trans_queue_depth=20;` I can reproduce this, and will fix. Thanks for this report..

mythbuster5, 2025-03-19:
Same issue with https://github.com/espressif/esp-idf/issues/15583 The fix will sync to github recently.

rsnbusa, 2025-03-19:
> Remove i2c_mst_config.trans_queue_depth=20; and try again.

Hi Simon_Cao.

Same result. Crash. I actually added that due to it being the only
parameter remotely associated with a Queue, which resulted in a collateral
damage routine not the direct cause.

From the iser #15583 it appears to be a FIXED array or max array out of
bound  problem.

So.... I changed the #define I2C_STATIC_OPERATION_ARRAY_MAX 10 // was 6 in
the i2c_private.h and now it works.

let me know if there is going to be a fix in the official repo.

Thanks.

mythbuster5, 2025-03-20:
I already merge the fix, it will sync to github soon.

mythbuster5, 2025-03-24:
Fixed https://github.com/xiv3r/esp-idf/commit/757f776f9e4d43eb67f819823a1c0f4ab1f2c5fb