espressif/esp-idf #15602: When the first line of lyrics appears on the Bluetooth music album cover, the receiving process will start. However, if it's pure music, the album cover may not be obtained continuously.

hyzgit, 2025-03-19:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

When the first line of lyrics appears on the Bluetooth music album cover, the receiving process will start. However, if it's pure music, the album cover may not be obtained continuously.

esp-lrh, 2025-03-20:
Hi, @hyzgit 

Thank you for your feedback, but we need more information to locate the issue, can you provide more information about this issue ? For example, IDF commit id, console log, peer device version (Android version or IOS version) etc.

By the way, can the issue being  reproduced in IDF example `a2dp_sink` ?

Alvin1Zhang, 2025-03-26:
@hyzgit Thanks for reporting, would you please help share if any updates for the issue? Thanks.

hyzgit, 2025-03-27:
> Hi, [@hyzgit](https://github.com/hyzgit)
>
> Thank you for your feedback, but we need more information to locate the issue, can you provide more information about this issue ? For example, IDF commit [...]

The a2dp_sink example of ESP-IDF5.4 is used and can be repeated.

esp-lrh, 2025-03-28:
> The a2dp_sink example of ESP-IDF5.4 is used and can be repeated.

Got it, since the problem can be reproduced on `a2dp_sink` example, let's first rule out non-code-related factors.

1. what we retrieve is music cover art, it is related to a specific song, if **the song does not have a cover art image**, then we will failed to retrieve, you may see log like `RC_CT: AVRC metadata rsp: attribute id 0x80,`. please test more pure music song about this.
2. the example only initiate get cover art progress when track change event come, you can add a timer to trig the progress to test this (a simple way is to call `bt_av_new_track()` in timer callback).
3. for some android music player, it need some delay to get cover art image after the track change event come, we can add a delay or retry the operation after one or two second to workaround this.

Please check the description above, make the appropriate changes and test again, thanks

closed 2025-07-02 by Alvin1Zhang