espressif/esp-idf #15606: 为啥克隆都有问题啊？ 强制更新都解决不了呢？

durongze, 2025-03-19:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

最新版

### Operating System used.

Windows

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

root@SNSVP11:~/code/esp/esp-idf# vi .gitmodules
root@SNSVP11:~/code/esp/esp-idf# git submodule update --remote --recursive
Submodule path 'components/bootloader/subproject/components/micro-ecc/micro-ecc': checked out '541b3a78026420a3e369c4c9281c396b5e531113'
Submodule path 'components/bt/controller/lib_esp32c2/esp32c2-bt-lib': checked out '3ce1675e0e0154345819eb79793d53bea522b29a'
Submodule path 'components/bt/controller/lib_esp32c3_family': checked out 'e668c2d101ee46ee1950819607694fb852aecae0'
Submodule path 'components/bt/controller/lib_esp32c5/esp32c5-bt-lib': checked out 'b4d27797cdfa3b21e44c2bdfdf3a09c48f7700be'
Submodule path 'components/bt/controller/lib_esp32c6/esp32c6-bt-lib': checked out '55778309611becb63829929912df61d4ff74bb89'
Submodule path 'components/bt/controller/lib_esp32h2/esp32h2-bt-lib': checked out 'e9d636c148ffb31b9f28563b2674b35ef09f4293'
remote: Enumerating objects: 15, done.
remote: Counting objects: 100% (15/15), done.
remote: Compressing objects: 100% (4/4), done.
remote: Total 9 (delta 2), reused 9 (delta 2), pack-reused 0 (from 0)
Unpacking objects: 100% (9/9), 727.33 KiB | 1.85 MiB/s, done.
From https://github.com/espressif/esp-ble-mesh-lib
   47ee135..480089d  main       -> origin/main
Submodule path 'components/bt/esp_ble_mesh/lib/lib': checked out '480089db598967048121942a12dc0ff24807d2ee'
remote: Enumerating objects: 12, done.
remote: Counting objects: 100% (12/12), done.
remote: Compressing objects: 100% (8/8), done.
remote: Total 12 (delta 2), reused 12 (delta 2), pack-reused 0 (from 0)
Unpacking objects: 100% (12/12), 61.42 KiB | 1.36 MiB/s, done.
From https://github.com/espressif/esp-nimble
   149eec70..0a221c06  nimble-1.6.0-idf -> origin/nimble-1.6.0-idf
Submodule path 'components/bt/host/nimble/nimble': checked out '0a221c0693af1c67622d0d10a9a96c0bd976b83a'
Submodule path 'components/cmock/CMock': checked out '204117cb59bafb7810946811688da8300b3b0927'
Submodule path 'components/cmock/CMock/vendor/c_exception': checked out '37673b75954b33ebd593d180993c64d3bef45bc0'
Submodule path 'components/cmock/CMock/vendor/unity': checked out 'cdf1d0297effc2736ee847e557b4275b4f02310b'
remote: Enumerating objects: 69, done.
remote: Counting objects: 100% (69/69), done.
remote: Compressing objects: 100% (14/14), done.
remote: Total 69 (delta 55), reused 69 (delta 55), pack-reused 0 (from 0)
Unpacking objects: 100% (69/69), 2.40 MiB | 1.38 MiB/s, done.
From https://github.com/espressif/esp32-wifi-lib
   b328e5ed..f6b785ec  master     -> origin/master
fatal: Unable to find refs/remotes/origin/HEAD revision in submodule path 'components/esp_wifi/lib'
root@SNSVP11:~/code/esp/esp-idf# make clean
make: *** No rule to make target 'clean'.  Stop.
root@SNSVP11:~/code/esp/esp-idf# git clean -xfd
root@SNSVP11:~/code/esp/esp-idf# git clean -fd
root@SNSVP11:~/code/esp/esp-idf# git status
fatal: not a git repository: components/lwip/lwip/../../../.git/modules/components/lwip/lwip
root@SNSVP11:~/code/esp/esp-idf# git remote -v
origin  https://github.com/espressif/esp-idf.git (fetch)
origin  https://github.com/espressif/esp-idf.git (push)
root@SNSVP11:~/code/esp/esp-idf#

[... 76 lines cut ...]
Submodule path 'components/cmock/CMock/vendor/unity': checked out 'cdf1d0297effc2736ee847e557b4275b4f02310b'
remote: Enumerating objects: 69, done.
remote: Counting objects: 100% (69/69), done.
remote: Compressing objects: 100% (14/14), done.
remote: Total 69 (delta 55), reused 69 (delta 55), pack-reused 0 (from 0)
Unpacking objects: 100% (69/69), 2.40 MiB | 1.38 MiB/s, done.
From https://github.com/espressif/esp32-wifi-lib
   b328e5ed..f6b785ec  master     -> origin/master
fatal: Unable to find refs/remotes/origin/HEAD revision in submodule path 'components/esp_wifi/lib'
root@SNSVP11:~/code/esp/esp-idf# make clean
make: *** No rule to make target 'clean'.  Stop.
root@SNSVP11:~/code/esp/esp-idf# git clean -xfd
root@SNSVP11:~/code/esp/esp-idf# git clean -fd
root@SNSVP11:~/code/esp/esp-idf# git status
fatal: not a git repository: components/lwip/lwip/../../../.git/modules/components/lwip/lwip
root@SNSVP11:~/code/esp/esp-idf# git remote -v
origin  https://github.com/espressif/esp-idf.git (fetch)
origin  https://github.com/espressif/esp-idf.git (push)
root@SNSVP11:~/code/esp/esp-idf#

### Build or installation Logs.

```plain

```

dobairoland, 2025-03-24:
Hi @durongze. 

This is the right command to run:
```sh
git submodule update --init --recursive
```

The `--init` argument is required for submodules which haven't been checked out yet.

If you have issue with your Internet then:
- it is possible to download a ZIP-ed ESP-IDF from Github releases which contains submodules as well,
- there are other mirrors like gitee.com/EspressifSystems/esp-idf.git or jihulab.com/esp-mirror/espressif/esp-idf/ which might help if you have trouble accessing github.com.

closed 2025-03-31 by dobairoland

durongze, 2025-04-04:
感谢回复，我后来删除了整个仓库，重新克隆就恢复了。