espressif/esp-idf #15609: I want to protect my IoT device from WiFi Deauth attacks in case of management frames are unencrypted in WPA2 network

aggaddam, 2025-03-20:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Espressif Support Team,

I am currently working on an IoT device based on the ESP32-C3 and would like to ensure its security against Wi-Fi de-authentication attacks, particularly when operating in WPA2-Personal networks where management frames are unencrypted.

Given that many customer environments still rely on WPA2 routers, I want to implement the best possible security measures without compromising compatibility with existing routers. My key concerns are:

    Protection Against Deauthentication Attacks: Since WPA2-Personal does not inherently encrypt management frames, what are Espressif’s recommended configurations for enabling Protected Management Frames (PMF) while ensuring broad router compatibility?

    Authentication Mode Recommendation: To maximize security while maintaining connection reliability across different routers, should I strictly use WIFI_AUTH_WPA2_PSK, or is there an alternative configuration that balances security and compatibility?  I currently plan to set PMF as "capable but not required" (pmf_cfg.capable = true, pmf_cfg.required = false). Would this configuration provide adequate protection against de-authentication attacks while allowing connectivity to routers that do not support PMF?

```
    wifi_config_t wifi_config    = {0};
    wifi_config.sta.scan_method  = WIFI_ALL_CHANNEL_SCAN;
    wifi_config.sta.sort_method  = WIFI_CONNECT_AP_BY_SIGNAL;
    wifi_config.sta.threshold.authmode = WIFI_AUTH_WPA2_PSK;
    wifi_config.sta.pmf_cfg.capable = true;  // Enables PMF if supported
    wifi_config.sta.pmf_cfg.required = false; // Allows connection even if PMF is not supported
```

    WPA3 Considerations: While WPA3 offers better security with mandatory PMF, not all routers support it. Would you recommend implementing WPA3 as an optional mode and falling back to WPA2 if necessary? What auth mode do I need to use in this case? I tried the below code but it didn't work with WPA2 router
```

    wifi_config_t wifi_config    = {0};
    wifi_config.sta.scan_method  = WIFI_ALL_CHANNEL_SCAN;
    wifi_config.sta.sort_method  = WIFI_CONNECT_AP_BY_SIGNAL;
    wifi_config.sta.threshold.authmode = WIFI_AUTH_WPA2_WPA3_PSK;
    wifi_config.sta.pmf_cfg.capable = true;  // Enables PMF if supported
    wifi_config.sta.pmf_cfg.required = false; // Allows connection even if PMF is not supported
```

I appreciate any guidance or best practices Espressif can share on securing IoT devices in real-world WPA2/WPA3 networks.

Looking forward to your expert advice.

kapilkedawat, 2025-03-20:
Hi @aggaddam,

Espressif gives the highest priority to security and has enabled best practices by default to ensure that users don't miss them. If a user wants to use lower security settings, they must explicitly configure them; otherwise, the default secure settings will remain enabled.

PMF : PMF is always enabled in station mode. If the AP supports PMF, it will always be used during the connection. Please see the [capable](https://docs.espressif.com/projects/esp-idf/en/v5.1.6/esp32/api-reference/network/esp_wifi.html?highlight=pmf_cfg#_CPPv4N17wifi_pmf_config_t7capableE) configuration for more details.

Authmode Threshold : [Authmode](https://docs.espressif.com/projects/esp-idf/en/v5.1.6/esp32/api-reference/network/esp_wifi.html?highlight=pmf_cfg#_CPPv4N21wifi_scan_threshold_t8authmodeE) is set to WIFI_AUTH_WPA2_PSK by default. Note that this minimum threshold—if a WPA3 AP is available, it will automatically use WPA3 security.

Please let us know if you have any further questions.

kapilkedawat, 2025-03-20:
@aggaddam please close the issue if this answers your queries.

closed 2025-03-21 by aggaddam