espressif/esp-idf #15612: IDF Release v5.4 incorrect issue link #14095

mmrein, 2025-03-20:
Just found that [IDF v5.4 Release](https://github.com/espressif/esp-idf/releases/tag/v5.4) changelog has incorrect issue linked : 

> I2C: Fixed I2C return without release the lock (https://github.com/espressif/esp-idf/issues/14095) (https://github.com/espressif/esp-idf/commit/944329fff4699b021eeff0d777c1d13f9c4c08c7)

As you can see, it links to **(IDFGH-13154) #14095** instead of **(IDFGH-14095) #14906**

Found it because I'm waiting for actual #14095 to be released, which was not ready at the time of first v5.4 Release.

Alvin1Zhang, 2025-03-21:
Thanks for reporting, and sorry for the inconvenience. 

The fix for https://github.com/espressif/esp-idf/issues/14095 has been merged internally, and fix on release/5.4 is available at https://github.com/espressif/esp-idf/commit/e347bdd44b69d5650410fca9fbec29c934adb640, and will be included in v5.4.1 release. Regarding the release of IDF v5.4.1, please help refer to https://github.com/espressif/esp-idf/blob/master/ROADMAP.md. Thanks.

ginkgm, 2025-03-21:
Thanks for reminder, the commit in the change list is correct while the github link is wrong. We can't easily update the release notes after it's published. Sorry for inconvenience.

Issue closed.

closed 2025-03-21 by ginkgm