espressif/esp-idf #15613: How can I re-init provisioning after provisioning fail (Not reset MCU)?

tranthanh1699, 2025-03-20:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hello everyone, I'm facing two issues:

When using the BLE provisioning example, if I provide incorrect Wi-Fi information (like a wrong password), I'm unable to retry. I tried deinitializing and reinitializing the provisioning, but it results in a reset. Could someone share sample code or a solution to address this issue?

How can I register an event in the ESP-IDF SDK so that it detects when a phone connects via BLE to the ESP32 during provisioning?

nopnop2002, 2025-03-21:
>Could someone share sample code or a solution to address this issue?

Can you provision again now?
```
idf.py erase-flash
idf.py flash monitor
```

tranthanh1699, 2025-03-22:
> > Could someone share sample code or a solution to address this issue?
> 
> Can you try provisioning again with this?
> 
> ```
> idf.py erase-flash
> idf.py flash monitor
> ```

Yes, I know this command, but my problem is I create the Device and when end user Provisioning device and provide incorrect wifi pass, I want to provisioning again (Not reset device)

nopnop2002, 2025-03-23:
The result of BLE provisioning is stored in the NVS and the WiFi mode is switched from AP mode to Station mode.

If you enter incorrect Wi-Fi information (such as an incorrect password), there is no function to switch back to AP mode.

hrushikesh430, 2025-03-24:
Hi @tranthanh1699, Cloud you please send me your minimal build application, which will help me to reproduce the issue?

tranthanh1699, 2025-03-25:
> Hi [@tranthanh1699](https://github.com/tranthanh1699), Cloud you please send me your minimal build application, which will help me to reproduce the issue?

My issue has been solve in the new SDK V5.4.0, thank for your support

closed 2025-03-25 by tranthanh1699