espressif/esp-idf #15614: ESP32S3 - PDM (Master Rx) - Over 104kHz not working (originally -- IDFGH-13234)

timsterc, 2025-03-20:
I have made your suggested change to the pdm driver, but the audio still doesn't seem to be working correctly.  

Any chance you could try this on one of your development boards that includes a PDM microphone?  Maybe you could use the PDM Microphone Recording example and modify it to work at 192,000Hz ?
 
Thanks!
-Tim 

 _Originally posted by @timsterc in [#14169](https://github.com/espressif/esp-idf/issues/14169#issuecomment-2600333079)_

L-KAYA, 2025-03-21:
Hi @timsterc, could you describe more detailedly about the issue? Like, did it can run successfully but the sampled audio seems not correct? Or the program still report error when setting the sample rate to 192KHz?

From the information I collect, there are several challenges for your scheme (i.e., sample ultrasonic by normal PDM microphone): 
1. Frequency response range limitation (most electret microphones have an effective range of ≤ 50kHz, but the bat's ultrasonic might reach 200KHz)
2. Low ultrasonic sensitivity (may require preamplifier)

So, I guess you may need a preamplifier and filter to get the expected data.

Another possible scheme is to use an ultrasonic receiver to replace the PDM microphone, and also need a preamplifier and filter to get the proper analog data, then convert it to digital by an ADC (whose sample rate better to be higher than 400 KHz), lastly, read the digital data from ADC via SPI or I2S interface.

timsterc, 2025-03-21:
@L-KAYA -- on one of your EPS32S3 dev boards that has a PDM microphone (I believe there are several) -- could you try running at 102kHz, then at 106kHz.  I don't think the microphone suddenly stops working correctly right at 104kHz.

L-KAYA, 2025-03-24:
@timsterc Glad to hear that, this issue will keep open~ I'm looking forward to your good news!

timsterc, 2025-03-26:
@L-KAYA Next question -- I'm wondering if it is an option to bypass the PDM to PCM conversion?  

I made this modification in the i2s_ll.h file in esp-idf-5.4 -- in hopes that I might be able to gain access to the raw PDM bit stream, so that I could perform my own FIR filtering / decimation. 

/**
 * @brief Enable RX PDM mode.
 *
 * @param hw Peripheral I2S hardware instance address.
 */
static inline void i2s_ll_rx_enable_pdm(i2s_dev_t *hw)
{
    hw->rx_conf.rx_pdm_en = true;
    hw->rx_conf.rx_tdm_en = false;
    //hw->rx_conf.rx_pdm2pcm_en = true;
    hw->rx_conf.rx_pdm2pcm_en = false;
}

L-KAYA, 2025-03-26:
@timsterc [This feature](https://github.com/espressif/esp-idf/blob/master/components/esp_driver_i2s/include/driver/i2s_pdm.h#L78) has been supported on the master branch, you can take a try with it first. It will be released in the next minor version v5.5.

timsterc, 2025-03-27:
@L-KAYA -- could you elaborate on the RAW format?  I have (2) microphones -- Left and Right -- does the RAW configuration respect MONO or would it always be L/R interleaved?

L-KAYA, 2025-03-28:
@timsterc  Sure, there is no much difference on the configuration compare to the current one, you can choose either both slots or any one of them. 

But if you use the raw PDM format, the PDM2PCM converter will be bypassed, and the converter coefficients will no longer take effect.

And you will still receive the data in `16 bits` for each slot, but the catenated raw data will look like `0001100110111101 1111110011110000 1100000010000000`, whose value is represented by the density of `1`. You need to realize a FIR filter in the software to convert it into PCM format.

timsterc, 2025-03-29:
@L-KAYA -- I'm making some progress, using the raw format, but would be curious if you're able to share any details on the FIR / decimation scheme that is employed by the on-chip PDM to PCM converter.  

I've tried a number of different approaches, but am continuing to get a fairly large amount of higher frequency noise.

In this example I'm speaking a few words, which can be seen down ~1kHz.

[image]

timsterc, 2025-05-04:
@L-KAYA -- just a ping to see if you have any thoughts?

L-KAYA, 2025-05-06:
TBH, I did find a good enough FIR filter yet, I tried some simple FIR filter but not perform well. It needs far more effort to finalize the algorithm, which is supposed to be universal, flexible and well performed.

timsterc, 2025-08-28:
@L-KAYA -- just bumping in hopes of a comment.

Alvin1Zhang, 2025-11-26:
Thanks for reporting, will close due to inactivity, feel free to reopen with more updates, thanks.

closed 2025-11-26 by Alvin1Zhang