espressif/esp-idf #15617: nimble ble_gap_master_set_timer compile fail if CONFIG_NIMBLE_ROLE_OBSERVER=y

bryghtlabs-richard, 2025-03-20:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1.6, v5.1.5-647-g346d2367ef

### Espressif SoC revision.

ESP32-S3

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

CMD

### Development Kit.

ESP32-S3-DevKitC-1 v1.0

### Power Supply used.

USB

### What is the expected behavior?

I expected to be able to update a project built using the BLE observer role.

### What is the actual behavior?

`idf.py build` fails with warnings and errors.

### Steps to reproduce.

1. Use examples/bluetooth/nimble/bleprph as example
2. Configure these:

```
CONFIG_BT_NIMBLE_ROLE_CENTRAL=n
CONFIG_BT_NIMBLE_ROLE_PERIPHERAL=y
CONFIG_BT_NIMBLE_ROLE_BROADCASTER=y
CONFIG_BT_NIMBLE_ROLE_OBSERVER=y
```

3. Then you will hit the build error

### Debug Logs.

```plain
C:/Espressif/frameworks/esp-idf-v5.1.6/components/bt/host/nimble/nimble/nimble/host/src/ble_gap.c: In function 'ble_gap_disc':
C:/Espressif/frameworks/esp-idf-v5.1.6/components/bt/host/nimble/nimble/nimble/host/src/ble_gap.c:5684:9: error: implicit declaration of function 'ble_gap_master_set_timer'; did you mean 'ble_gap_master_timer'? [-Werror=implicit-function-declaration]
 5684 |         ble_gap_master_set_timer(duration_ticks);
      |         ^~~~~~~~~~~~~~~~~~~~~~~~
      |         ble_gap_master_timer
cc1.exe: some warnings being treated as errors
```

### Diagnostic report archive.

This functionality is not available on 5.1.6.

### More Information.

Looks to be the same as #15157

rahult-github, 2025-03-21:
Hi @bryghtlabs-richard  ,

A [fix](https://github.com/espressif/esp-idf/commit/65aaf217d8eaab6d50bff3c034d69892eb98f7d2) for this issue is already present on release/v5.1 .   You can cherry-pick it at your end.

bryghtlabs-richard, 2025-03-21:
Thanks @rahult-github - it's working for me now.

closed 2025-03-21 by bryghtlabs-richard