espressif/esp-idf #15619: ESP32P4的SPI2和SPI3分别可以挂多少个device？

beautifulpeer, 2025-03-21:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

1. 测试过程中发现，使用SPI3挂着3个device可以，挂载第四个就会出错

[image]

2.尝试将SPI3改为SPI2，测试挂载第四个device没问题

3.ESP32P4的SPI2和SPI3分别可以挂多少个device？官方文档中也没找到

suda-morris, 2025-03-21:
On ESP32-P4, the number of hardware CS signals in GPSPI3 is only 3, and the number of hardware CS signals in GPSPI4 is 6.

beautifulpeer, 2025-03-21:
[image]
where is GPSPI4?

beautifulpeer, 2025-03-21:
1. 测试 SPI2 挂4个没问题， 但测试SPI3挂3个可以，挂4个就不行

closed 2025-03-21 by beautifulpeer