espressif/esp-idf #15621: How to properly de-init ledc channel?

7aman, 2025-03-21:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Here’s a more grammatically refined version of your GitHub issue description:

---

**Title:** Issue with reinitializing LEDC channel after deinitialization

---

I use the following functions to initialize and deinitialize LEDC channels:

```c
static void ledc_init(const gpio_num_t pin, const ledc_channel_t ch, const ledc_timer_t tim)
{
    const ledc_timer_config_t timer_config = {
        .speed_mode = LEDC_LOW_SPEED_MODE,
        .timer_num = tim,
        .duty_resolution = LEDC_TIMER_12_BIT,
        .freq_hz = 5000,
        .clk_cfg = LEDC_AUTO_CLK
    };
    ESP_ERROR_CHECK(ledc_timer_config(&timer_config));

    const ledc_channel_config_t channel_config = {
        .speed_mode = LEDC_LOW_SPEED_MODE,
        .channel = ch,
        .timer_sel = tim,
        .intr_type = LEDC_INTR_DISABLE,
        .gpio_num = pin,
        .duty = OFF_DUTY,
        .hpoint = 0
    };
    ESP_ERROR_CHECK(ledc_channel_config(&channel_config));
}

static void ledc_deinit(const gpio_num_t pin, const ledc_channel_t ch, const ledc_timer_t tim)
{
    ESP_ERROR_CHECK(ledc_timer_pause(LEDC_LOW_SPEED_MODE, tim));
    const ledc_timer_config_t timer_config = {
        .speed_mode = LEDC_LOW_SPEED_MODE,
        .timer_num = tim,
        .deconfigure = true,
    };
    ESP_ERROR_CHECK(ledc_timer_config(&timer_config));
    ESP_ERROR_CHECK(ledc_stop(LEDC_MODE, ch, 0));
    ESP_ERROR_CHECK(gpio_reset_pin(pin));
}
```

After the first de-initialization, when I try to re-initialize the channel, I get the following runtime warning:

```plaintext
ledc: GPIO 26 is not usable, maybe conflict with others
```

The issue seems to stem from LEDC reserving the pin in `ledc_channel_config`, but there is no API provided to reset or delete this reservation. The only available function to reset the GPIO is `gpio_reset_pin`, but it doesn’t fully revoke the pin either.

By "reserve" and "revoke", I refer to the internal APIs used in the `esp_gpio_reserve` header, which seem to be responsible for pin management. There is no corresponding API to completely remove or revoke GPIO reservations made by LEDC.

Would it be possible to add an API to properly release the GPIO pins reserved by LEDC or a way to reset the pin fully?

chegewara, 2025-03-24:
What board, what esp-idf version?

I am using ledc together with `gpio level` on the same pin, because in some situations i need to keep pin high in light sleep mode, which is not working with ledc/pwm.
I dont have any problems to reuse the same pin as gpio and ledc.
I dont actually do `ledc_stop`, all i do is `gpio_reset_pin`.

7aman, 2025-03-24:
> What board, what esp-idf version?

esp-idf v5.4  
custom board using `ESP32-­WROOM­-32E`  

I do not intend to use the pin directly after de-initialization. I just want to know how to properly release the resources acquired by LEDC. Most examples include a de-initialization procedure, but I couldn't find one for LEDC.

songruo, 2025-03-31:
Please check the comment in https://github.com/espressif/esp-idf/issues/15666. I will backport the workaround to v5.4.

7aman, 2025-03-31:
thank you. i close this issue.

closed 2025-03-31 by 7aman