espressif/esp-idf #15624: Race condition in I2C Master

JD-MLC, 2025-03-22:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.4

### Espressif SoC revision.

ESP32-C6

### Operating System used.

Linux

### How did you build your project?

Command line with CMake

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom Board

### Power Supply used.

USB

### What is the expected behavior?

Same issue as https://github.com/espressif/esp-idf/issues/15444

User gomatteo pointed out part of the issue. You should be able to do I2C and not have the system crash.

The fix they posted was not enough to fully fix the problem for me.

// Initialise event queue.
    xQueueReset(i2c_master->event_queue);
    i2c_master->event = I2C_EVENT_ALIVE;
    while (i2c_master->i2c_trans.cmd_count) {
        if (xSemaphoreTake(i2c_master->cmd_semphr, ticks_to_wait) != pdTRUE) {
            // Software timeout, clear the command link and finish this transaction.
            atomic_store(&i2c_master->status, I2C_STATUS_TIMEOUT);
            i2c_master->cmd_idx = 0;
            i2c_master->trans_idx = 0;
            ESP_LOGE(TAG, "I2C software timeout");
            xSemaphoreGive(i2c_master->cmd_semphr);
            return;
        }

I think it is still correct, but I also had to modify the code below:

#if !SOC_I2C_STOP_INDEPENDENT
    else {
        i2c_operation_t *i2c_operation = &i2c_master->i2c_trans.ops[i2c_master->read_buf_pos];
        portENTER_CRITICAL_ISR(&i2c_master->base->spinlock);
        i2c_ll_read_rxfifo(hal->dev, i2c_operation->data + i2c_operation->bytes_used, i2c_master->read_len_static);
        i2c_ll_read_rxfifo(hal->dev, i2c_master->i2c_trans.ops[i2c_master->read_buf_pos + 1].data, 1);

adding a simple check between the two i2c_ll_read_rxfifo()

        i2c_ll_read_rxfifo(hal->dev, i2c_operation->data + i2c_operation->bytes_used, i2c_master->read_len_static);
        // Check if the read buffer is allocated.
    	if (i2c_master->i2c_trans.ops[i2c_master->read_buf_pos + 1].data == NULL) {
    		// Mark the transaction as error so higher layers can detect the failure.
    		i2c_master->status = I2C_STATUS_ACK_ERROR;  // maybe replace with a more appropriate error
    		portEXIT_CRITICAL_ISR(&i2c_master->base->spinlock);
    		return;
    	}
        i2c_ll_read_rxfifo(hal->dev, i2c_master->i2c_trans.ops[i2c_master->read_buf_pos + 1].data, 1);

I have a board with a few I2C slaves that have bad I2C state machines, so they mess up the bus a lot and I have to reset it every few hundred transactions using a mix of i2c_master_bus_reset() and i2c_master_bus_rm_device() + i2c_del_master_bus() + setup again. Not sure if that is related, but either way, the I2C slaves suck and constantly have timing issues so it is great for testing I2C code! The ESP32C6 as the I2C master was crashing in i2c_ll_read_rxfifo() because it was passed a NULL pointer. It crashed about once every 20K transactions without any changes, every 30-40K with the previous fix, and with my fix I am testing and now have over 100K transactions without crashing yet. Back of the napkin statistics say 99% confident it was fixed, but I'll keep it running for a few hundred thousand more.

### What is the actual behavior?

It crashes. See the other issue.

### Steps to reproduce.

1. Step: See other issue
2. Step
3. Step
...

### Debug Logs.

```plain

```

mythbuster5, 2025-03-24:
@JD-MLC  Ok. I see. Both of your issue is that when the i2c is not at a correct state, but interrupt want to receive some data. So this is a race condition indeed. So I'm thinking to make more strict judge branch. Would you please help try and test this code?

[image]

The principle is only `stop` command here is a correct status for reading data. Other status might cause sth wrong. Thanks for waiting your respond.

mythbuster5, 2025-03-27:
I want to fix like this, means that we don't do read when i2c bus in a wrong state. Could you please try? If this can't solve your problem, please let me know.

[image]

mythbuster5, 2025-04-07:
Solved by https://github.com/espressif/esp-idf/commit/f8ecc4885d720c3a36b25097774691ca6d0e46b3

closed 2025-04-07 by mythbuster5