espressif/esp-idf #15626: ble50 examples all set scan DURATION and PERIOD to 0, which is invalid

hraftery, 2025-03-22:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.4

### Espressif SoC revision.

SP32-C3 (QFN32) (revision v0.4)

### Operating System used.

macOS

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom board

### Power Supply used.

USB

### What is the expected behavior?

That the BLE 5.0 examples projects successfully start a BLE scan.

### What is the actual behavior?

Bluetooth initialises and configures, but when `esp_ble_gap_start_ext_scan()` is called, the return value is 0x12. That error value maps to `HCI_ERR_ILLEGAL_PARAMETER_FMT`. The log looks like:

```
I (715) BLE_INIT: BT controller compile version [d752dea]
I (717) BLE_INIT: Bluetooth MAC: 9c:9e:6e:31:97:5a
I (718) phy_init: phy_version 1180,01f2a49,Jun  4 2024,16:34:25
I (767) BLE5: GATT client register, status 0
E (769) BT_BTM: LE ES En=1: cmd err=0x12
E (770) BT_HCI: CC evt: op=0x2042, status=0x12
E (770) BLE5: Extended scanning start failed, status 112
```

### Steps to reproduce.

1. Copy the contents of the example into a new project: https://github.com/espressif/esp-idf/tree/ee77c489e73993493e95bf765c0955cb87d59166/examples/bluetooth/bluedroid/ble_50/ble50_security_client
2. Build, Flash, Monitor
3. Observe the error noted above.

### Debug Logs.

```plain

```

### Diagnostic report archive.

[diag-39881643-4f12-4311-b162-b610775f7e12.zip](https://github.com/user-attachments/files/19403703/diag-39881643-4f12-4311-b162-b610775f7e12.zip)

### More Information.

The parameters to `esp_ble_gap_start_ext_scan()` are EXT_SCAN_DURATION and EXT_SCAN_PERIOD, which are universally defined to be `0`. The documentation for `esp_ble_gap_start_ext_scan()` says the arguments must be greater than zero. Sure enough, if these values are set to be non-zero, the function succeeds and the example proceeds as expected.

Is there some special meaning for 0 that applies in certain situations?

esp-zhp, 2025-03-27:
Thank you for reporting the issue.  

Setting **Duration** and **Period** to 0 has a special meaning.  

We will optimize the documentation accordingly.
[image]

esp-zhp, 2025-03-27:
```
/**
* @brief           Enables extended scanning.
*
* @param[in]       duration  Scan duration in units of 10 ms. 
*                            - Range: 0x0001 to 0xFFFF (Time = N * 10 ms).
*                            - 0x0000: Scan continuously until explicitly disabled.
*
* @param[in]       period    Time interval between the start of consecutive scan durations, in units of 1.28 seconds.  
*                            - Range: 0x0001 to 0xFFFF (Time = N * 1.28 sec).  
*                            - 0x0000: Scan continuously.
* @return            - ESP_OK : success
*                    - other  : failed
*/
esp_err_t esp_ble_gap_start_ext_scan(uint32_t duration, uint16_t period);
```

hraftery, 2025-03-28:
I figured that was probably the case, which is why I was surprised to get the `HCI_ERR_ILLEGAL_PARAMETER_FMT` error. There's something else afoot here.

hraftery, 2025-03-28:
Ah! I compared my project with the example line by line and noticed one difference:

The example sets `scan_duplicate` in `esp_ble_ext_scan_params_t` to `BLE_SCAN_DUPLICATE_DISABLE`. I've changed that to `BLE_SCAN_DUPLICATE_ENABLE_RESET`.

Sure enough, if I create a fresh project from the `ble50_security_client` example, set the target to `esp32c3` and build/run, it works fine. If change `scan_duplicate` to `BLE_SCAN_DUPLICATE_ENABLE_RESET`, then I get the same `BT_BTM: LE ES En=1: cmd err=0x12` / `BT_HCI: CC evt: op=0x2042, status=0x12` error as in the original report. If I then set DURATION/PERIOD to non-zero, everything is fine again.

esp-zhp, 2025-04-11:
@hraftery Do you have any other questions at the moment? I'm about to update the documentation.

hraftery, 2025-04-11:
Question? I'm just reporting an issue: `BLE_SCAN_DUPLICATE_DISABLE` is not compatible with a DURATION/PERIOD of zero. Documenting the meaning of zero helps, but doesn't resolve the issue.

esp-zhp, 2025-04-21:
Yes, we will update the documentation here.  
Besides updating the documentation, is there anything else you’d like me to take care of?

You also mentioned:  
> *If I create a fresh project from the `ble50_security_client` example, set the target to `esp32c3` and build/run, it works fine. If I change `scan_duplicate` to `BLE_SCAN_DUPLICATE_ENABLE_RESET`, then I get the same `BT_BTM: LE ES En=1: cmd err=0x12 / BT_HCI: CC evt: op=0x2042, status=0x12` error as in the original report. If I then set DURATION/PERIOD to non-zero, everything is fine again.*

According to the spec:  
> *If the Filter_Duplicates parameter is set to 0x02, duplicate advertisements in a single scan period should not be sent to the Host in advertising report events; this setting shall only be used if both Period and Duration are non-zero. If Filter_Duplicates is set to 0x02 and either Period or Duration is zero, the Controller shall return the error code Invalid HCI Command Parameters (0x12).*

So, I don’t think this is actually an issue.

---

see：
BLUETOOTH CORE SPECIFICATION Version 5.3 | Vol 4, Part E page 2483

hraftery, 2025-04-21:
Oh great, that makes sense. I don't think it will be obvious to anyone coming at this from ESP-IDF, rather than the Bluetooth spec. But hopefully these tidbits we've created will help.

esp-zhp, 2025-05-19:
Hello,

Thank you for your detailed report and patience.

We have optimized and clarified the documentation regarding the usage of esp_ble_gap_start_ext_scan(), especially the meaning and valid values of duration and period parameters.

This update is included in the commit ebb76933d01870c2d5f4e05da3816240fac432e6 on the latest ESP-IDF branch. Please update to the latest version to see the improved documentation.

If you have further questions or need additional assistance, feel free to reach out.

hraftery, 2025-05-19:
Much clearer.

closed 2025-05-20 by Alvin1Zhang