espressif/esp-idf #15627: ANCS example fails to pair

EdwardDeaver, 2025-03-22:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.5-dev-2488-gee77c489e7

### Operating System used.

macOS

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

The bluetooth example should successfully connect

### What is the actual behavior?

I (80509) BLE_ANCS: ESP_GATTC_OPEN_EVT
W (82129) BT_SMP: FOR LE SC LTK IS USED INSTEAD OF STK
W (82369) BT_APPL: gattc_conn_cb: if=3 st=0 id=3 rsn=0x3d
W (82369) BT_APPL: bta_dm_ble_smp_cback remove bond,rsn 102, BDA:0x7571XYZEC5C9
E (82369) BT_BTM: Device not found

W (82369) BT_HCI: hcif disc complete: hdl 0x1, rsn 0x3d
E (82379) BT_GATT: GATT_Write Illegal param: conn_id 3, type 02,
W (82389) BT_APPL: bta_gattc_mark_bg_conn unable to find the bg connection mask for: 75:71:XY:ZE:c5:c9
W (82389) BT_BTC: btc_dm_ble_auth_cmpl_evt, remove bond in flash bd_addr: 7571XYZEc5c9
I (82399) addr: 75 71 XY ZE c5 c9
I (82399) BLE_ANCS: pair status = fail

### Steps to reproduce.

1. Flash the script
2. Open serial monitor
3. Try connecting to it via an iPhone (13 Pro)
4. See error

### Build or installation Logs.

```plain

```

esp-zhp, 2025-03-27:
Which chip are you using? Could you provide me with the complete log, from startup to the point where the issue occurs?

EdwardDeaver, 2025-04-08:
Sorry about this, I figured it out. Issue on my end.

closed 2025-04-08 by EdwardDeaver