espressif/esp-idf #15628: napt not working on softap connecto to a wpa2 enterprise peap network

Preetam-Das, 2025-03-22:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I am trying to use softap_sta with a enterprise wpa2 PEAP network. I am modifying the existing exmaple from https://github.com/espressif/esp-idf/tree/v5.4/examples/wifi/softap_sta/main . Modified code is shown below.

- esp32c3
- espidf v5.4

`#include <string.h>
#include "esp_eap_client.h"
#include "esp_err.h"
#include "freertos/FreeRTOS.h"
#include "freertos/projdefs.h"
#include "freertos/task.h"
#include "freertos/event_groups.h"
#include "esp_mac.h"
#include "esp_wifi.h"
#include "esp_event.h"
#include "esp_log.h"
#include "esp_netif_net_stack.h"
#include "esp_netif.h"
#include "nvs_flash.h"
#include "lwip/inet.h"
#include "lwip/netdb.h"
#include "lwip/sockets.h"
#include "lwip/lwip_napt.h"
#include "lwip/err.h"
#include "lwip/sys.h"

/* STA Configuration */
#define ESP_WIFI_STA_SSID           "enterprise_ssid"
#define ESP_WIFI_STA_USERNAME       "enterprise_username"
#define ESP_WIFI_STA_PASSWD         "enterprise_passwd"
#define ESP_MAXIMUM_RETRY           3

#if CONFIG_ESP_WIFI_AUTH_OPEN
#define ESP_WIFI_SCAN_AUTH_MODE_THRESHOLD   WIFI_AUTH_OPEN
#elif CONFIG_ESP_WIFI_AUTH_WEP
#define ESP_WIFI_SCAN_AUTH_MODE_THRESHOLD   WIFI_AUTH_WEP
#elif CONFIG_ESP_WIFI_AUTH_WPA_PSK
#define ESP_WIFI_SCAN_AUTH_MODE_THRESHOLD   WIFI_AUTH_WPA_PSK
#elif CONFIG_ESP_WIFI_AUTH_WPA2_PSK
#define ESP_WIFI_SCAN_AUTH_MODE_THRESHOLD   WIFI_AUTH_WPA2_PSK
#elif CONFIG_ESP_WIFI_AUTH_WPA_WPA2_PSK
#define ESP_WIFI_SCAN_AUTH_MODE_THRESHOLD   WIFI_AUTH_WPA_WPA2_PSK
#elif CONFIG_ESP_WIFI_AUTH_WPA3_PSK
#define ESP_WIFI_SCAN_AUTH_MODE_THRESHOLD   WIFI_AUTH_WPA3_PSK
#elif CONFIG_ESP_WIFI_AUTH_WPA2_WPA3_PSK
#define ESP_WIFI_SCAN_AUTH_MODE_THRESHOLD   WIFI_AUTH_WPA2_WPA3_PSK
#elif CONFIG_ESP_WIFI_AUTH_WAPI_PSK
#define ESP_WIFI_SCAN_AUTH_MODE_THRESHOLD   WIFI_AUTH_WAPI_PSK
#endif

/* AP Configuration */
#define ESP_WIFI_AP_SSID            "espgateway"
#define ESP_WIFI_AP_PASSWD          "password"
#define ESP_WIFI_CHANNEL            1
#define ESP_MAX_STA_CONN            1

/* The event group allows multiple bits for each event, but we only care about two events:
 * - we are connected to the AP with an IP
 * - we failed to connect after the maximum amount of retries */
#define WIFI_CONNECTED_BIT BIT0
#define WIFI_FAIL_BIT      BIT1

[... 173 lines cut ...]
    } else {
        ESP_LOGE(TAG_STA, "UNEXPECTED EVENT");
        return;
    }

    /* Set sta as the default interface */
    esp_netif_set_default_netif(esp_netif_sta);

    vTaskDelay(pdMS_TO_TICKS(1000));

    /* Enable napt on the AP netif */
    if ((ret = esp_netif_napt_enable(esp_netif_ap)) != ESP_OK) {
        ESP_LOGE(TAG_STA, "NAPT not enabled on the netif: %p", esp_netif_ap);
        printf("%s\n",esp_err_to_name(ret));
    }

}
`

Everything works, except napt. So I can't connect to the internet or to any other device on the enterprise network through the access point. The error is:

`E (2385) WiFi Sta: NAPT not enabled on the netif: 0x3fca9dd0
ESP_ERR_NOT_SUPPORTED`

Note that the example works as it is for normal non-enterprise network. Not sure if I modified it the wrong way. Someone help.

kapilkedawat, 2025-03-24:
@Preetam-Das could you please share your sdkconfig?

Preetam-Das, 2025-03-24:
[sdkconfig.txt](https://github.com/user-attachments/files/19427970/sdkconfig.txt) @kapilkedawat

kapilkedawat, 2025-03-24:
```
CONFIG_LWIP_IP_FORWARD=y
CONFIG_LWIP_IPV4_NAPT=y
CONFIG_LWIP_IPV4_NAPT_PORTMAP=y
```

Are not set in your sdkconfig, please check if you have mistakenly removed `sdkconfig.defaults` from your project directory.