espressif/esp-idf #15630: [ESP32S3] Flush failed with 'A fatal error occurred: Failed to write to target RAM (result was 01070000: Operation timed out'

yuhangbin, 2025-03-23:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

## Device
MacBook Pro M1 Pro
ESP32 S3

## Context

I just copy the hello_world examples to build and flash.
The build was success and get trouble on flash stage.

` *  Executing task: /Users/yuhangbin/.espressif/python_env/idf5.3_py3.12_env/bin/python /Users/yuhangbin/esp/esp-idf/components/esptool_py/esptool/esptool.py -p /dev/tty.usbmodem5A450117361 -b 460800 --before default_reset --after hard_reset --chip esp32s3 write_flash --flash_mode dio --flash_freq 80m --flash_size 16MB 0x0 bootloader/bootloader.bin 0x100000 xiaozhi.bin 0x8000 partition_table/partition-table.bin 0xd000 ota_data_initial.bin 0x10000 srmodels/srmodels.bin 

esptool.py v4.8.1
Serial port /dev/tty.usbmodem5A450117361
Connecting....
Chip is ESP32-S3 (QFN56) (revision v0.2)
Features: WiFi, BLE, Embedded PSRAM 8MB (AP_3v3)
Crystal is 40MHz
MAC: 30:ed:a0:a5:86:24
Uploading stub...

A fatal error occurred: Failed to write to target RAM (result was 01070000: Operation timed out)

 *  The terminal process "/Users/yuhangbin/.espressif/python_env/idf5.3_py3.12_env/bin/python '/Users/yuhangbin/esp/esp-idf/components/esptool_py/esptool/esptool.py', '-p', '/dev/tty.usbmodem5A450117361', '-b', '460800', '--before', 'default_reset', '--after', 'hard_reset', '--chip', 'esp32s3', 'write_flash', '--flash_mode', 'dio', '--flash_freq', '80m', '--flash_size', '16MB', '0x0', 'bootloader/bootloader.bin', '0x100000', 'xiaozhi.bin', '0x8000', 'partition_table/partition-table.bin', '0xd000', 'ota_data_initial.bin', '0x10000', 'srmodels/srmodels.bin'" terminated with exit code: 2. 

 *  Executing task: /Users/yuhangbin/.espressif/python_env/idf5.3_py3.12_env/bin/python /Users/yuhangbin/esp/esp-idf/components/esptool_py/esptool/esptool.py -p /dev/tty.usbmodem5A450117361 -b 460800 --before default_reset --after hard_reset --chip esp32s3 write_flash --flash_mode dio --flash_freq 80m --flash_size 16MB 0x0 bootloader/bootloader.bin 0x100000 xiaozhi.bin 0x8000 partition_table/partition-table.bin 0xd000 ota_data_initial.bin 0x10000 srmodels/srmodels.bin 

esptool.py v4.8.1
Serial port /dev/tty.usbmodem5A450117361
Connecting...
Chip is ESP32-S3 (QFN56) (revision v0.2)
Features: WiFi, BLE, Embedded PSRAM 8MB (AP_3v3)
Crystal is 40MHz
MAC: 30:ed:a0:a5:86:24
Uploading stub...

A fatal error occurred: Failed to write to target RAM (result was 01070000: Operation timed out)

 *  The terminal process "/Users/yuhangbin/.espressif/python_env/idf5.3_py3.12_env/bin/python '/Users/yuhangbin/esp/esp-idf/components/esptool_py/esptool/esptool.py', '-p', '/dev/tty.usbmodem5A450117361', '-b', '460800', '--before', 'default_reset', '--after', 'hard_reset', '--chip', 'esp32s3', 'write_flash', '--flash_mode', 'dio', '--flash_freq', '80m', '--flash_size', '16MB', '0x0', 'bootloader/bootloader.bin', '0x100000', 'xiaozhi.bin', '0x8000', 'partition_table/partition-table.bin', '0xd000', 'ota_data_initial.bin', '0x10000', 'srmodels/srmodels.bin'" terminated with exit code: 2. 

 *  Executing task: ninja  

[1/4] cd /Users/yuhangbin/Code/Har...re/xiaozhi-esp32/build/xiaozhi.bin
xiaozhi.bin binary size 0x2a25a0 bytes. Smallest app partition is 0x600000 bytes. 0x35da60 bytes (56%) free.
[2/4] Performing build step for 'bootloader'
[1/1] cd /Users/yuhangbin/Code/Hardware/xiaozhi-esp32/build/bootloader/esp-idf/esptool_py && /Users/yuhangbin/.espressif/python_env/idf5.3_py3.12_env/bin/python /Users/yuhangbin/esp/esp-idf/components/partition_table/check_sizes.py --offset 0x8000 bootloader 0x0 /Users/yuhangbin/Code/Hardware/xiaozhi-esp32/build/bootloader/bootloader.bin
Bootloader binary size 0x3f20 bytes. 0x40e0 bytes (51%) free.
[4/4] Completed 'bootloader'

 *  Executing task: /Users/yuhangbin/.espressif/python_env/idf5.3_py3.12_env/bin/python /Users/yuhangbin/esp/esp-idf/components/esptool_py/esptool/esptool.py -p /dev/tty.usbmodem5A450117361 -b 460800 --before default_reset --after hard_reset --chip esp32s3 write_flash --flash_mode dio --flash_freq 80m --flash_size 16MB 0x0 bootloader/bootloader.bin 0x100000 xiaozhi.bin 0x8000 partition_table/partition-table.bin 0xd000 ota_data_initial.bin 0x10000 srmodels/srmodels.bin 

esptool.py v4.8.1
Serial port /dev/tty.usbmodem5A450117361
Connecting....
Chip is ESP32-S3 (QFN56) (revision v0.2)
Features: WiFi, BLE, Embedded PSRAM 8MB (AP_3v3)
Crystal is 40MHz
MAC: 30:ed:a0:a5:86:24
Uploading stub...

A fatal error occurred: Failed to write to target RAM (result was 01070000: Operation timed out)

 *  The terminal process "/Users/yuhangbin/.espressif/python_env/idf5.3_py3.12_env/bin/python '/Users/yuhangbin/esp/esp-idf/components/esptool_py/esptool/esptool.py', '-p', '/dev/tty.usbmodem5A450117361', '-b', '460800', '--before', 'default_reset', '--after', 'hard_reset', '--chip', 'esp32s3', 'write_flash', '--flash_mode', 'dio', '--flash_freq', '80m', '--flash_size', '16MB', '0x0', 'bootloader/bootloader.bin', '0x100000', 'xiaozhi.bin', '0x8000', 'partition_table/partition-table.bin', '0xd000', 'ota_data_initial.bin', '0x10000', 'srmodels/srmodels.bin'" terminated with exit code: 2. 
`

yuhangbin, 2025-03-27:
It's hardware problem, I just switch other esp32s3 it works. Maybe I should be close this issue, because I don't know what's problem with bad one device.

closed 2025-03-27 by yuhangbin

bbrungardt02, 2025-03-28:
Currently, I have two esp32-s3-wroom-1 boards and am experiencing the same issue with both.

radimkarnis, 2025-03-31:
Hello @yuhangbin @bbrungardt02, 

This issue is usually caused by delays or glitches in the serial communication. You can try:

- increasing timeouts, setting with [a configuration file](https://docs.espressif.com/projects/esptool/en/latest/esp32s3/esptool/configuration-file.html)
- deactivating the stub flasher (in the menuconfig, or by adding `--no-stub` to the esptool command)

This is hard to debug, as there are many parts in the chain that could cause this (drivers, USB hubs, nad USB cables, ...)

Since you are experiencing this only with one chip and the other works, this seems to point to a HW issue and I don't have much to suggest you.