espressif/esp-idf #15639: LWIP bridge w/ USB-NCM

mickeyl, 2025-03-24:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I tried adding USB-NCM to the LWIP bridge basing on code that has been developed by @kostaond as per https://github.com/espressif/esp-idf/issues/5697.

I've used @chegewara's usb-netif component here and slightly tweaked it (don't set IP and don't run DHCPS, since this will be happening on the bridge interface).

My example is @ https://github.com/mickeyl/esp-usb-wifi-eth-bridge -- but unfortunately it doesn't work.

The USB network interface appears, but despite adding it to the bridge interface (via `esp_netif_br_glue_add_port`, the USB interface is not coming up, i.e. there's no IP. It _should_ share 192.168.4.1 with the WiFI and the W5500, shouldn't it? Do I need to setup the MAC differently?

What am I missing?

chegewara, 2025-03-24:
I think you should study those examples
https://github.com/espressif/esp-idf/tree/master/examples/network/sta2eth
https://github.com/espressif/esp-idf/tree/master/examples/network/eth2ap

My component is based on the first example.

mickeyl, 2025-03-24:
I don't think so. As far as I can see these examples are doing the forwarding "on their own". The bridge example does this in a higher level way basing on LWIP. As it already works fine with WiFi and ETH, I think there can only be a small thing missing or wrong.

kostaond, 2025-03-24:
Hi @mickeyl, I'm terribly sorry but I currently don't have any time to have a look at it in details...

Check the followig:
* does USB sets its `netif` to UP? E.g. does it call `esp_netif_action_connected` anywhere? Port must be up to be forwarded.
* At least one of the interfaces either Ethernet or WiFi needs to be UP to set bridge netif UP. This is currently a limitation of [esp_netif_br_glue.c](https://github.com/espressif/esp-idf/blob/master/components/esp_netif/lwip/esp_netif_br_glue.c). It natively only monitors for ETH and WiFi actions to start the bridge but once started just by one interface, there is no problem.

  IP address e.g. `192.168.4.1` is IP address of bridge `netif` physical interfaces like ETH, WiFi doesn't have it's IP address. They are you to pass just Ethernet frames.

mickeyl, 2025-03-25:
I call `esp_netif_action_connected` right after starting the interface.

When adding log output to `netif_recv_callback` and `l2_free` I see that these functions are called:

```
D (136689) usb-netif: netif_recv_callback: 333
D (136689) usb-netif: l2_free
D (136689) usb-netif: netif_recv_callback: 110
D (136689) usb-netif: l2_free
D (136959) usb-netif: netif_recv_callback: 110
D (136959) usb-netif: l2_free
D (137309) usb-netif: netif_recv_callback: 86
D (137309) usb-netif: l2_free
D (138339) usb-netif: netif_recv_callback: 110
D (138339) usb-netif: l2_free
D (138359) usb-netif: netif_recv_callback: 62
D (138359) usb-netif: l2_free
```

Still though, the computer connected via USB_NCM interface does not seem to get an IP from the DHCPS whereas both the WiFi and the W5500 interfaces are getting one.

kostaond, 2025-03-26:
* Could you please enable debug logs for the lwIP bridge? `Component config->LWIP->Enable LWIP debug->Enable bridge *`
* Make sure your USB_NCM is set to promiscuous mode.
* Capture a traffic in computer connected to USB_NCM.

mickeyl, 2025-03-26:
```c
bool p = true;
esp_eth_ioctl(usb_netif, ETH_CMD_S_PROMISCUOUS, &p);
```

crashes. Perhaps I need a special way to do this on USB_NCM?

kostaond, 2025-03-28:
@mickeyl you can't use `esp_eth_ioctl` with `netif` handle, this must be used with Ethernet handle. Check if the current driver of USB_NCM support `esp_eth_ioctl`. Configuring the promiscuous mode is MAC specific feature.

chegewara, 2025-04-01:
Im not networking expert, actually i am pretty lame, but i spent few hours today trying to figure out if i can do it.
My conclusion is that USB netif is missing a glue code, which we have for wifi sta/ap and eth.

The idea i have is to make, but im not sure if that solution is satisfying (and not sure if/how it may works):
- usb-netif to eth
- bridge eth with wifi sta/ap

mickeyl, 2025-04-15:
@chegewara 

> Im not networking expert, actually i am pretty lame, but i spent few hours today trying to figure out if i can do it. My conclusion is that USB netif is missing a glue code, which we have for wifi sta/ap and eth.

But which part?
 
> The idea i have is to make, but im not sure if that solution is satisfying (and not sure if/how it may works):
> 
>     * usb-netif to eth
> 
>     * bridge eth with wifi sta/ap

Sounds like a plan B, but it would be a mixture of doing it ourselves and having LWIP bridge ETH and AP.
After all, the bridge code seems to be written with multiple ethernet interfaces in mind and from the viewpoint of LWIP, USB-NCM is ethernet. I'm pretty sure it's just a tiny thing we're overlooking to make it work with the builtin bridge.

kostaond, 2025-04-15:
Thank you @mickeyl for providing me your code. I could quickly go over it and I found the problem. The issue is the `br_glue` layer is waiting for `start` event to add the interface into a bridge.

It could be hacked as follows in `bridge_example_main.c` after Ethernet start.

```c
   void *handle = (void *)1; // see components/chegewara-usb-netif/usb_netif_ncm.c (line 67)
   esp_event_post(ETH_EVENT, ETHERNET_EVENT_START, &handle, sizeof(handle), 0);
```

Note that this is super ugly hack and proper solution will be provided later.

chegewara, 2025-04-15:
Hi @kostaond 
this is very useful info, and i added this code at the end, after init usb-netif, wifi-ap and bridge
```
	void *handle = (void *)1; // see components/chegewara-usb-netif/usb_netif_ncm.c (line 67)
	esp_event_post(ETH_EVENT, ETHERNET_EVENT_START, &handle, sizeof(handle), 0);
	esp_netif_action_connected(eth_netifs[0], 0, 0, 0);
```
It seems to work, at least partially. What i mean that after connecting smartphone to wifi AP USB on my PC is getting IP address, which is good, but i cant ping from PC to smartphone or smartphone to PC.

BTW i had to set flags in usb-netif to 0 (per comment in bride example)
```
    esp_netif_inherent_config_t base_cfg = {
        .flags =  0,
```

Next I will try to add ethernet on P4 and see how is it going

Thanks

chegewara, 2025-04-15:
@mickeyl You dont have to modify my usb-netif component to use it in this/your bridge code
All you have to do is to init it this way:
```
#ifdef BRIDGE_USB
    // Start USB netif
    const esp_netif_inherent_config_t base = {
        .flags =  0, // Run DHCP server; set the netif "ip" immediately
        .if_key = CONFIG_USB_NETIF_IF_KEY,                                  // Set mame, key, priority
        .if_desc = "usb ncm config device",
        .route_prio = 52,
    };
    ESP_ERROR_CHECK(init_tinyusb(NULL));
    ESP_ERROR_CHECK(usb_net_create(NULL));
    esp_netif_t *usb_netif = netif_create(&base, NULL, NULL);
    esp_netif_action_start(usb_netif, 0, 0, 0);
    
    uint8_t s_sta_mac[6] = {};
    ESP_ERROR_CHECK(esp_read_mac(s_sta_mac, ESP_MAC_ETH));
    ESP_ERROR_CHECK(esp_netif_set_mac(usb_netif, s_sta_mac));
    br_ports++;
#endif
```
And for wifi on esp32-p4 here is CMakeLists.txt
```
dependencies:
  ethernet_init:
    path: ${IDF_PATH}/examples/ethernet/basic/components/ethernet_init
  chegewara/usb-netif: '*'
  espressif/esp_wifi_remote: '*'
```
The only issue with above hack is that you have to connect ethernet cable first then USB cable to get IPs assigned.

Thank you both for this topic and letting us know how we can use usb-netif in bridge.

mickeyl, 2025-04-16:
**Thanks a lot** @kostaond  and also @chegewara ­– this is shaping up VERY nicely.

Apart from some minor timing issues (_when_ is _which_ interface being connected), this works just fine. I started `iperf3` as server on the computer with the attached USB interface and `iperf3` clients once on the computer connected via W5500 and once on the computer connected via WiFi.

Even with an unoptimized debug release (with lots of debug output), I already get 700-800KBit/sec, which is not bad considering that the S3 is only a Full-speed device. Will do more tests later. @chegewara perhaps you could test w/ your P4?

Let's keep this issue open for a short while to add more findings. Once again, great service by Espressif!

chegewara, 2025-04-16:
Actually I did a test on P4.
What I did is connect P4 to PC over USB and ethernet to router then I ran speedtest.
I believe that the result is not bad:
- downlink: 80mbit/s
- uplink: 50mbit/s
On my local network speedtest
- downlink: 89mbit/s
- uplink: I don't have setup

When using WiFi on my smartphone the results are not that good. If I remember correctly it was around 20mbit/s.

chegewara, 2025-04-26:
Hi @kostaond 
i am playing a bit with this example and i have questions. I am using example with DHCPS enabled.
I see some inconsistency when the init order is changed:
- when `esp_netif_br_glue_handle_t netif_br_glue = esp_netif_br_glue_new();` is called before `example_eth_init`, which mean before ethernet phy are init then i cant get IP  on my PC, when i call after eth phy init then all is fine
- when `esp_netif_br_glue_handle_t netif_br_glue = esp_netif_br_glue_new();`  either before or after wifi AP or usb-netif is init then all is ok

It would be nice, but is not some super important, if we could call `esp_netif_br_glue_handle_t netif_br_glue = esp_netif_br_glue_new();` before any netif config, including ethernet, or at least to get some explanation why is that

I am thinking about making some small netif bridge component to make it easier to init when more netifs is added, like thread or zigbee.

Thanks

kostaond, 2025-05-06:
@chegewara could you please send me exact code snipped of non-functional initialization?

kostaond, 2025-05-21:
@chegewara could you please share the example with `esp_netif_br_glue_new` order issue?

chegewara, 2025-05-21:
Hi @kostaond 
Sorry for my late reply
i am trying to reproduce it today, but i cant. I have to recollect the exact steps.

Thanks