espressif/esp-idf #15641: Crash in timer_remove() During BT PHY Disable on ESP32-H2 (esp_timer.c:327)

ahmetcobanoglu, 2025-03-24:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.4

### Espressif SoC revision.

ESP32-H2

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-H2-DevKitM-1

### Power Supply used.

External 5V

### What is the expected behavior?

1. The timer associated with BT PHY should be properly stopped and removed during the BT sleep/shutdown process
2. The timer_remove() function should safely handle the timer's removal without crashing
3. The BT controller should be able to enter sleep mode successfully with all associated timers properly cleaned up
4. The PHY modem should be disabled gracefully without causing system instability

### What is the actual behavior?

- The system crashes during the execution of timer_remove() function at line 327 in esp_timer.c
- The ESP32-H2 generates a core dump during the Bluetooth PHY disable sequence
- The crash occurs specifically when trying to stop a timer (0x4084ce04) as part of the BT controller sleep callback
- The program counter at the time of crash is at offset +58 bytes into the timer_remove function
- The BT controller fails to enter sleep mode properly
- The entire system becomes unstable and requires a restart

### Steps to reproduce.

1. Using Bluetooth functionality on ESP32-H2
2. When the Bluetooth controller attempts to go to sleep mode
3. During the PHY disable sequence for the Bluetooth modem

### Debug Logs.

```plain

```

### More Information.

### Configuration

- Light sleep with external XTAL enabled
- FreeRTOS tickless idle enabled
- **Reproduction frequency: Occurred only once**

### Stack Trace

```
#0  0x40809ab4 in timer_remove (timer=timer@entry=0x4084ce04) at ../esp/master/esp-idf/components/esp_timer/src/esp_timer.c:327
#1  0x40809cb0 in esp_timer_stop (timer=0x4084ce04) at ../esp/master/esp-idf/components/esp_timer/src/esp_timer.c:260
#2  0x4205f9e4 in phy_track_pll_deinit () at ../esp/master/esp-idf/components/esp_phy/src/phy_common.c:121
#3  0x4205f694 in esp_phy_disable (modem=modem@entry=PHY_MODEM_BT) at ../esp/master/esp-idf/components/esp_phy/src/phy_init_esp32hxx.c:136
#4  0x40802f12 in controller_sleep_cb (enable_tick=<optimized out>, arg=<optimized out>) at ../esp/master/esp-idf/components/bt/controller/esp32h2/bt.c:611
#5  0x4080409a in r_ble_lll_rfmgmt_disable ()
#6  0x4202939a in r_ble_ll_task ()
#7  0x408113a8 in vPortTaskWrapper (pxCode=<optimized out>, pvParameters=<optimized out>) at ../esp/master/esp-idf/components/freertos/FreeRTOS-Kernel/portable/riscv/port.c:255
```

### Register Values at Crash

```
ra             0x40809a90       0x40809a90 <timer_remove+22>
sp             0x4082ba60       0x4082ba60
gp             0x408156d4       0x408156d4 flash_constants::STORAGE_PARTITION_LABEL
tp             0x4082bb20       0x4082bb20
t0             0x4000af8c       1073786764
t1             0x40804596       1082148246
t2             0x0      0
fp             0x4084ce04       0x4084ce04
s1             0x0      0
a0             0xe      14
a1             0xff000000       -16777216
a2             0x4084fc0c       1082457100
a3             0x0      0
a4             0x4084ce0c       1082445324
a5             0x4081ad14       1082240276
a6             0x0      0
a7             0x0      0
s2             0x1      1
s3             0x668ddf 6720991
s4             0x4081b000       1082241024
s5             0x4208d000       1107873792
s6             0x0      0
s7             0x0      0
s8             0x0      0
s9             0x0      0
s10            0x0      0
s11            0x0      0
t3             0xde     222
t4             0x0      0
t5             0x19ce075f       432932703
t6             0x7e075f 8259423
pc             0x40809ab4       0x40809ab4 <timer_remove+58>
```

### Additional Information

The crashed task was not in an interrupt context, which suggests this occurred during normal task execution of the BLE controller task.

ChrysalisChenJ, 2025-03-25:
Hi @ahmetcobanoglu,

Can you provide more info about this crash:
- What is the circumstance when the crash happens, are you using BT alone or you are using some other modules as well. You can provide your demo code if that's ok with you.
- Can you provide the complete log files of the crash, is it a guru meditation error or an assertion? I can not see the reason of the crash from the info you provided.

ahmetcobanoglu, 2025-03-25:
Hi @ChrysalisChenJ 

Thanks for looking into this issue. Here's what you asked for:

- In my project I use ADC and BLE.
- I don't have log files since it is a sleepy device (light sleep configured). The information I shared comes from the core dump. My core dump doesn't show the panic reason.
- The crash occurred when the BLE tried to disable PHY for power management, as shown in the call stack.
- Since I cannot reproduce the issue, I can't provide more information about it.

Let me know if you need anything else.

ChrysalisChenJ, 2025-03-25:
Hi @ahmetcobanoglu,
Since we do not know how and why this crash is happening, we may need to know what your code does with BT so we might have a assumption on how this may happen. But it would be best if you can reproduce the issue and save the log at crash.

And while you run, please set these two configs `CONFIG_BT_LE_CONTROLLER_LOG_ENABLED`  and `CONFIG_BT_LE_CONTROLLER_LOG_WRAP_PANIC_HANDLER_ENABLE` to dump the BLE log files once the issue reproduce.

Alvin1Zhang, 2025-05-08:
@ahmetcobanoglu Thanks for reporting, would you please help share if any updates for the issue? Thanks.

ChrysalisChenJ, 2025-05-08:
Hi @ahmetcobanoglu,

Do you manage to reproduce the issue again? We would close the issue if not.