espressif/esp-idf #15645: esp_http_client_flush_response does not behave as expected

jdbaptista, 2025-03-24:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

ESP-IDF v5.4.0

### Espressif SoC revision.

ESP32-S3 (QFN56) (revision v0.2)

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

Custom Board

### Power Supply used.

USB

### What is the expected behavior?

I expect esp_http_client_flush_response to "process all remaining response data" such that esp_http_client_read does not return any of the previous response after the sequence: esp_http_client_close, esp_http_client_set_url, esp_http_client_open.

### What is the actual behavior?

I have a function that is meant to ignore any non-200 response codes and clear any response from it, with declaration `esp_err_t openServerFile(int64_t *contentLength, esp_http_client_handle_t client, const char *URL, int retryNum)`. This function works by calling `esp_http_client_set_url(client, URL)`, then `esp_http_client_open`, then `esp_http_client_fetch_headers`, then `esp_http_client_get_status_code`. If the status code is not 200, `esp_http_client_flush_response` is called, then `esp_http_client_close`, then returns. I expect that the next function call to `openServerFile` will open a file such that `esp_http_client_read` reads data from the start of the new file, however it actually returns the response from the previous non-200 query.

```
TEST_CASE("openServerFile_nonExistent", "[api_connect]")
{
    const char *URLNonexistent = CONFIG_DATA_SERVER CONFIG_TEST_DATA_BASE_URL "/DOES_NOT_EXIST";
    const char *URLTypical = CONFIG_DATA_SERVER CONFIG_TEST_DATA_BASE_URL "/openServerFile_typical.1";
    esp_err_t err;
    const int32_t bufLen = 30;
    char buffer[bufLen];
    char *expected;
    int64_t contentLength;

    err = openServerFile(&contentLength, client, URLNonexistent, RETRY_NUM);
    TEST_ASSERT_EQUAL(ESP_ERR_NOT_FOUND, err);

    err = openServerFile(&contentLength, client, URLTypical, RETRY_NUM);
    TEST_ASSERT_EQUAL(ESP_OK, err);
    TEST_ASSERT_EQUAL(55, contentLength); // size of file

    expected = "abcdefghijklmnopqrstuvwxyz\nhe";
    do {
        contentLength = esp_http_client_read(client, buffer, bufLen - 1);
    } while (contentLength == -ESP_ERR_HTTP_EAGAIN);
    TEST_ASSERT_EQUAL(bufLen - 1, contentLength);
    buffer[bufLen - 1] = '\0';
    TEST_ASSERT_EQUAL_STRING(expected, buffer);

    err = esp_http_client_close(client);
    TEST_ASSERT_EQUAL(ESP_OK, err);
[... 149 lines cut ...]
                return ESP_FAIL;
            }
            err = esp_http_client_close(client);
            if (err != ESP_OK) {
                ESP_LOGE(TAG, "failed to close client");
                return ESP_FAIL;
            }
            retryNum--;
            continue;
        }
        return ESP_OK;
    }
    return ESP_ERR_NOT_FOUND; // retried too many times
}
```

### Debug Logs.

```plain

```

### Diagnostic report archive.

[diag-58b293c7-8642-4f43-bfd9-fa106745d0be.zip](https://github.com/user-attachments/files/19437861/diag-58b293c7-8642-4f43-bfd9-fa106745d0be.zip)

jdbaptista, 2025-03-25:
Thank you for pointing that out, I've changed that now because its bad practice to make that cast. Unfortunately, changing it to a pointer to an actual int type stack variable or even just NULL produces the same results.

nileshkale123, 2025-04-11:
Hello @jdbaptista,

**First of all, API esp_http_client_flush_response() used to completely read and discard all remaining response data from an HTTP response. It calls esp_http_client_get_data() to read data until all data is received. It just discards the received data, and returns the total bytes discarded.** 

Now, coming to the issue:

I tried to reproduce the behavior, but I wasn’t able to observe the same results as you. In my case, the data was flushed successfully as expected. I’ve created a test scenario under the examples/esp_http_client directory.
I’m attaching a patch that adds a new API http_customized_request() to demonstrate the flow. You can replace the URLs in this API with your own and try it out on your setup.
Please let me know if I’m missing any steps or if you’re still seeing the issue as originally reported. You can revert with the patch with issue arised.

Here is the patch.
Add the following API call in **esp_http_client_example.c**
`static void http_customized_request()
{
    char output_buffer[MAX_HTTP_OUTPUT_BUFFER + 1] = {0};   // Buffer to store response of http request
    int content_length = 0;
    esp_http_client_config_t config = {
        .url = "NOT_EXISTED_FILE_URL",
        .event_handler = _http_event_handler,
        .crt_bundle_attach = esp_crt_bundle_attach,
    };
    esp_http_client_handle_t client = esp_http_client_init(&config);

    // GET Request
    esp_http_client_set_method(client, HTTP_METHOD_GET);
    esp_err_t err = esp_http_client_open(client, 0);
    if (err != ESP_OK) {
        ESP_LOGE(TAG, "Failed to open HTTP connection: %s", esp_err_to_name(err));
    } else {
        content_length = esp_http_client_fetch_headers(client);
        if (content_length < 0) {
            ESP_LOGE(TAG, "HTTP client fetch headers failed");
        } else {
            int data_read = esp_http_client_read_response(client, output_buffer, MAX_HTTP_OUTPUT_BUFFER);
            if (data_read >= 0) {
                int status_code = esp_http_client_get_status_code(client);
                if(status_code != 200) {
                    ESP_LOGE(TAG, "HTTP GET Response Status Code = %d", status_code);
                    int flushed_length = 0;
                    esp_http_client_flush_response(client, &content_length);
                    ESP_LOGI(TAG, "HTTP GET Response Flushed Length = %d bytes", flushed_length);
                }
                ESP_LOGI(TAG, "HTTP GET Response = %s", output_buffer);
                ESP_LOGI(TAG, "HTTP GET Response Length = %d bytes", data_read);
            } else {
                ESP_LOGE(TAG, "Failed to read response");
            }
        }
    }
    esp_http_client_close(client);

    esp_http_client_set_url(client, "CORRECT_EXISTING_URL");
    // GET Request
    esp_http_client_set_method(client, HTTP_METHOD_GET);
    err = esp_http_client_open(client, 0);
    if (err != ESP_OK) {
        ESP_LOGE(TAG, "Failed to open HTTP connection: %s", esp_err_to_name(err));
    } else {
        content_length = esp_http_client_fetch_headers(client);
        if (content_length < 0) {
            ESP_LOGE(TAG, "HTTP client fetch headers failed");
        } else {
            int data_read = esp_http_client_read_response(client, output_buffer, MAX_HTTP_OUTPUT_BUFFER);
            if (data_read >= 0) {
                ESP_LOGI(TAG, "HTTP GET Response = %s", output_buffer);
                ESP_LOGI(TAG, "HTTP GET Response Length = %d bytes", data_read);
            } else {
                ESP_LOGE(TAG, "Failed to read response");
            }
        }
    }
    esp_http_client_close(client);

}`,

and call this API from task API http_test_task*().

nileshkale123, 2025-05-06:
Hello @jdbaptista ,

I am closing this issue. Please feel free to reopen it if needed.

closed 2025-05-06 by nileshkale123