espressif/esp-idf #15646: [esp_http_client]: HTTP Request Pipelining/Multiplexing Support

bryghtlabs-richard, 2025-03-24:
### Describe the solution you'd like.

I'd like the ability to submit multiple outstanding HTTP(s) requests through `esp_http_client` and tell apart the responses.

With HTTP/1.1, this is known as [HTTP pipelining](https://en.wikipedia.org/wiki/HTTP_pipelining).
With HTTP/2, this is handled by HTTP multiplexing.

### Additional context.

Some notes about what would need addressed:

- pipelining should probably be off by default for compatibility with broken proxies, so need a flag to enable multiple requests
- `esp_http_state_t` would need to be split up into  a request-side and response-side state variables.
- Would need a way to tell which response went with which request.
  - Currently, getters like `esp_http_client_get_url()` work within the event-handler. If we want these to work with pipelining/multiplexing, the client would need to capture the necessary data to rejoin with the response.
  - Alternatively, for most uses it would be enough to capture `event_handler` and `user_data` per request, so that the event handler's `esp_http_client_event_t` can be populated correctly
-  As long as there are outstanding requests, `esp_http_client_perform()` would need to check for responses, as well as submit new requests - might be better to split into separate functions.

AshUK, 2025-04-10:
@bryghtlabs-richard have you looked at using https://components.espressif.com/components/espressif/nghttp ?

bryghtlabs-richard, 2025-04-10:
Thanks @AshUK , I had not seen that - looks like it should cover the needs I have.

closed 2025-04-10 by bryghtlabs-richard