espressif/esp-idf #15647: [esp_http_client]: HTTP readline support

bryghtlabs-richard, 2025-03-24:
### Is your feature request related to a problem?

No, just an idea to make line-based server-push HTTP parsing simpler.

### Describe the solution you'd like.

I'd like a delimited equivalent of `int esp_http_client_read_response()`, where I could specify a delimiter(or just use newline). Upon finding the delimiter or length-limit, the function would return.

This would make implementing HTTP chunked push-based streaming protocols like [NDJSON](https://github.com/ndjson/ndjson-spec)/[LDJSON](https://github.com/kupodigital/ldjson)/[JSONL](https://jsonlines.org/), stock-ticker type APIs, and other line-based formats simpler.

### Describe alternatives you've considered.

It isn't a difficult thing to write, but I find myself repeating the implementation. Here are some approaches without changing `esp_http_client`

Option1: `esp_http_client_read(len=1)` and appending to line-buffer until a newline is found, then parsing the line, then resetting the buffer-index.
Option2: `esp_http_client_read(large fixed buffer)`, then finding the newline, parsing the line, shifting the buffer over to erase the previous line, repeating.

nileshkale123, 2025-06-19:
Hello @bryghtlabs-richard ,

Apologies for the delayed response.

I’ve implemented the API and attached the patch here. Please take a look and let me know if this aligns with what you had in mind. I haven’t tested it end-to-end yet — I’ll proceed with full testing once you confirm that the API meets your requirnment.
[patch_issue_15647.txt](https://github.com/user-attachments/files/20815359/patch_issue_15647.txt)

Since this API uses esp_http_client->response.raw_data to store the remaining data after the delimiter, and no changes were made to the IDF itself, do you think it would be more appropriate for users to implement and use such an API on their own side?

bryghtlabs-richard, 2025-06-23:
Thanks @nileshkale123 , I can see the patch now!

Supporting multi-byte delimeters is definitely a useful feature! It may complicate the parsing though - I worry about what may happen to this parser under either of the following edge-cases:

- 1st byte of delimeter is in raw_data buffer, but additional bytes of delimeter are fetched from esp_http_client_read()
- 1st byte of delimeter is fetched from esp_http_client_read(), but additional bytes of delimeter is fetched from a different esp_http_client_read() call.

Maybe it would be simpler to keep appending to `raw_data` until the delimeter is located? Roughly like:

```
while (total_read < len && no-delim-in-raw-data) {
        // Read data using the existing esp_http_client_read function
}

//at this point, we must have a delim, or hit the max line len requested
```

nileshkale123, 2025-06-26:
Hello @bryghtlabs-richard ,

Thank you for bringing these points to our attention.
I have updated the API as per the changes outlined in the following patch:
[patch_issue_15647.txt](https://github.com/user-attachments/files/20923279/patch_issue_15647.txt)

That said, considering this API and the fact that users can achieve similar functionality using the existing members of esp_http_client_handle_t, I would like to ask — do you think it might be more appropriate to leave this implementation to the user side?

bryghtlabs-richard, 2025-06-26:
> I would like to ask — do you think it might be more appropriate to leave this implementation to the user side?

It would be nice support this in the `esp_http_client`, as many other http clients do, rather than requiring each line-based client to reimplement it themselves. But I can see the argument for keeping the `esp_http_client` as simple as possible.