espressif/esp-idf #15648: Allocation failures from calling i2s_alloc_dma_desc are not handled properly

kasperl, 2025-03-25:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

If calling `i2s_alloc_dma_desc` returns an error, the functions that call it correctly check for that. However, they do this after updating state that indicates that the buffer was successfully allocated. This makes any retries after freeing memory forget to re-allocate the structures (`i2s_alloc_dma_desc` is not called again), which eventually will lead to segmentation faults.

This is one example of where the state (`handle->dma.buf_size`) is updated to eagerly:

https://github.com/espressif/esp-idf/blob/23c73cdc37d9b8e6c840ea99c5ae12b9a0ce7032/components/esp_driver_i2s/i2s_std.c#L116

As far as I can tell, the `handle->dma.buf_size` field should be updated after the last error check in the block:

```
     /* The DMA buffer need to re-allocate if the buffer size changed */
     if (handle->dma.buf_size != buf_size) {
-        handle->dma.buf_size = buf_size;
         ESP_RETURN_ON_ERROR(i2s_free_dma_desc(handle), TAG, "failed to free the old dma descriptor");
         ESP_RETURN_ON_ERROR(i2s_alloc_dma_desc(handle, handle->dma.desc_num, buf_size),
                             TAG, "allocate memory for dma descriptor failed");
+        handle->dma.buf_size = buf_size;
     }
```

I believe there are 4 such calls to `i2s_alloc_dma_desc` that are mishandled, but you may want to grep for them.

kasperl, 2025-03-25:
Hmm. It actually looks like `buf_size` is set by `i2s_alloc_dma_desc`. I think we need to fix that too and just drop all the updates to `buf_size` outside of that function.

kasperl, 2025-03-25:
I suggest something like this to make `i2s_free_dma_desc` more robust. You don't really need to guard the call to `free` with a null check but that seems to be the style.

```
esp_err_t i2s_free_dma_desc(i2s_chan_handle_t handle)
{
    I2S_NULL_POINTER_CHECK(TAG, handle);
    uint32_t desc_num = handle->dma.desc_num;
    handle->dma.desc_num = 0;
    handle->dma.buf_size = 0;

    if (handle->dma.desc) {
        for (int i = 0; i < desc_num; i++) {
            if (handle->dma.desc[i]) {
                free(handle->dma.desc[i]);
                handle->dma.desc[i] = NULL;
            }
        }
        free(handle->dma.desc);
        handle->dma.desc = NULL;
    }

    if (handle->dma.bufs) {
        for (int i = 0; i < desc_num; i++) {
            if (handle->dma.bufs[i]) {
                free(handle->dma.bufs[i]);
                handle->dma.bufs[i] = NULL;
            }
        }
        free(handle->dma.bufs);
        handle->dma.bufs = NULL;
    }

    return ESP_OK;
}
```

kasperl, 2025-03-25:
Sorry @L-KAYA about the multiple updates, but I just realized that the handling of the `buf_size` field was wrong in more places - and the issue is always that it isn't reset correctly on allocation failures.

L-KAYA, 2025-03-25:
Yeah, I got what you mean, it's important to reset these values while the allocation failed, I'll take a further look into this issue

L-KAYA, 2025-03-25:
Hi @kasperl, could you take a look at this fix patch? Do you think this fix is proper?

[0001-fix-i2s-reset-the-dma-buf_size-while-allocation-fail.patch](https://github.com/user-attachments/files/19445320/0001-fix-i2s-reset-the-dma-buf_size-while-allocation-fail.patch)

kasperl, 2025-03-25:
Hi @L-KAYA: I think this is a bit over the top. The `i2s_alloc_dma_desc` function will call `i2s_free_dma_desc` if allocations fail within it, so as long as free is robust and clears `buf_size` - and ideally `desc_num` - there should be no need for the extra checks at every call of `i2s_alloc_dma_desc`.

This is what I ended up with on our fork of esp-idf: https://github.com/toitware/esp-idf/pull/103/files.

You know this code better than I do, so I could definitely be wrong about some of this :)

kasperl, 2025-03-25:
@L-KAYA - quick extra point: I think it is important the `buf_size` is always reset in `i2s_free_dma_desc` so make sure to do it before any returns in the function. Even if `handle->dma.desc` is null, you still need to clear it.

https://github.com/espressif/esp-idf/blob/23c73cdc37d9b8e6c840ea99c5ae12b9a0ce7032/components/esp_driver_i2s/i2s_common.c#L460

kasperl, 2025-03-25:
For what it is worth, I dislike the early exit if `handle->dma.desc` is null, because it assumes that `dma.desc` is allocated before `dma.bufs` which is incredibly brittle.

kasperl, 2025-03-25:
Turns out I was wrong to set `handle->dma.desc_num` to zero when freeing. I also think it is wrong that it is being set in `i2s_alloc_dma_desc`. The `num` passed to `i2s_alloc_dma_desc` must match `handle->dma.desc_num`, so maybe it shouldn't be passed separately. For now, I've done this instead:

```
diff --git a/components/esp_driver_i2s/i2s_common.c b/components/esp_driver_i2s/i2s_common.c
index ca7332e32a..f4cd73126b 100644
--- a/components/esp_driver_i2s/i2s_common.c
+++ b/components/esp_driver_i2s/i2s_common.c
@@ -380,27 +380,29 @@ uint32_t i2s_get_buf_size(i2s_chan_handle_t handle, uint32_t data_bit_width, uin
 esp_err_t i2s_free_dma_desc(i2s_chan_handle_t handle)
 {
     I2S_NULL_POINTER_CHECK(TAG, handle);
-    if (!handle->dma.desc) {
-        return ESP_OK;
-    }
-    for (int i = 0; i < handle->dma.desc_num; i++) {
-        if (handle->dma.bufs[i]) {
-            free(handle->dma.bufs[i]);
-            handle->dma.bufs[i] = NULL;
-        }
-        if (handle->dma.desc[i]) {
-            free(handle->dma.desc[i]);
-            handle->dma.desc[i] = NULL;
+    handle->dma.buf_size = 0;
+
+    if (handle->dma.desc) {
+        for (int i = 0; i < handle->dma.desc_num; i++) {
+            if (handle->dma.desc[i]) {
+                free(handle->dma.desc[i]);
+                handle->dma.desc[i] = NULL;
+            }
         }
+        free(handle->dma.desc);
+        handle->dma.desc = NULL;
     }
+
     if (handle->dma.bufs) {
+        for (int i = 0; i < handle->dma.desc_num; i++) {
+            if (handle->dma.bufs[i]) {
+                free(handle->dma.bufs[i]);
+                handle->dma.bufs[i] = NULL;
+            }
+        }
         free(handle->dma.bufs);
         handle->dma.bufs = NULL;
     }
-    if (handle->dma.desc) {
-        free(handle->dma.desc);
-        handle->dma.desc = NULL;
-    }
 
     return ESP_OK;
 }
@@ -409,8 +411,8 @@ esp_err_t i2s_alloc_dma_desc(i2s_chan_handle_t handle, uint32_t num, uint32_t bu
 {
     I2S_NULL_POINTER_CHECK(TAG, handle);
     esp_err_t ret = ESP_OK;
+    ESP_RETURN_ON_FALSE(num == handle->dma.desc_num, ESP_ERR_INVALID_ARG, TAG, "num doesn't match dma.desc_num");
     ESP_RETURN_ON_FALSE(bufsize <= I2S_DMA_BUFFER_MAX_SIZE, ESP_ERR_INVALID_ARG, TAG, "dma buffer can't be bigger than %d", I2S_DMA_BUFFER_MAX_SIZE);
-    handle->dma.desc_num = num;
     handle->dma.buf_size = bufsize;
 
     /* Descriptors must be in the internal RAM */
diff --git a/components/esp_driver_i2s/i2s_pdm.c b/components/esp_driver_i2s/i2s_pdm.c
index 9d10058af4..26082d0895 100644
--- a/components/esp_driver_i2s/i2s_pdm.c
+++ b/components/esp_driver_i2s/i2s_pdm.c
@@ -95,7 +95,6 @@ static esp_err_t i2s_pdm_tx_set_slot(i2s_chan_handle_t handle, const i2s_pdm_tx_
     uint32_t buf_size = i2s_get_buf_size(handle, slot_cfg->data_bit_width, handle->dma.frame_num);
     /* The DMA buffer need to re-allocate if the buffer size changed */
     if (handle->dma.buf_size != buf_size) {
-        handle->dma.buf_size = buf_size;
         ESP_RETURN_ON_ERROR(i2s_free_dma_desc(handle), TAG, "failed to free the old dma descriptor");
         ESP_RETURN_ON_ERROR(i2s_alloc_dma_desc(handle, handle->dma.desc_num, buf_size),
                             TAG, "allocate memory for dma descriptor failed");
@@ -385,7 +384,6 @@ static esp_err_t i2s_pdm_rx_set_slot(i2s_chan_handle_t handle, const i2s_pdm_rx_
     uint32_t buf_size = i2s_get_buf_size(handle, slot_cfg->data_bit_width, handle->dma.frame_num);
     /* The DMA buffer need to re-allocate if the buffer size changed */
     if (handle->dma.buf_size != buf_size) {
-        handle->dma.buf_size = buf_size;
         ESP_RETURN_ON_ERROR(i2s_free_dma_desc(handle), TAG, "failed to free the old dma descriptor");
         ESP_RETURN_ON_ERROR(i2s_alloc_dma_desc(handle, handle->dma.desc_num, buf_size),
                             TAG, "allocate memory for dma descriptor failed");
diff --git a/components/esp_driver_i2s/i2s_std.c b/components/esp_driver_i2s/i2s_std.c
index c3ef0c0469..489ac73313 100644
--- a/components/esp_driver_i2s/i2s_std.c
+++ b/components/esp_driver_i2s/i2s_std.c
@@ -101,7 +101,6 @@ static esp_err_t i2s_std_set_slot(i2s_chan_handle_t handle, const i2s_std_slot_c
     uint32_t buf_size = i2s_get_buf_size(handle, slot_cfg->data_bit_width, handle->dma.frame_num);
     /* The DMA buffer need to re-allocate if the buffer size changed */
     if (handle->dma.buf_size != buf_size) {
-        handle->dma.buf_size = buf_size;
         ESP_RETURN_ON_ERROR(i2s_free_dma_desc(handle), TAG, "failed to free the old dma descriptor");
         ESP_RETURN_ON_ERROR(i2s_alloc_dma_desc(handle, handle->dma.desc_num, buf_size),
                             TAG, "allocate memory for dma descriptor failed");
diff --git a/components/esp_driver_i2s/i2s_tdm.c b/components/esp_driver_i2s/i2s_tdm.c
index 584e75881c..46ef013571 100644
--- a/components/esp_driver_i2s/i2s_tdm.c
+++ b/components/esp_driver_i2s/i2s_tdm.c
@@ -106,7 +106,6 @@ static esp_err_t i2s_tdm_set_slot(i2s_chan_handle_t handle, const i2s_tdm_slot_c
     uint32_t buf_size = i2s_get_buf_size(handle, slot_cfg->data_bit_width, handle->dma.frame_num);
     /* The DMA buffer need to re-allocate if the buffer size changed */
     if (handle->dma.buf_size != buf_size) {
-        handle->dma.buf_size = buf_size;
         ESP_RETURN_ON_ERROR(i2s_free_dma_desc(handle), TAG, "failed to free the old dma descriptor");
         ESP_RETURN_ON_ERROR(i2s_alloc_dma_desc(handle, handle->dma.desc_num, buf_size),
                             TAG, "allocate memory for dma descriptor failed");
```

Also as a patch file here: 

[patch.txt](https://github.com/user-attachments/files/19450284/patch.txt)

kasperl, 2025-03-25:
All calls to `i2s_alloc_dma_desc` just pass `handle->dma.desc_num` as `num` as far as I can tell:

```
$ git grep 'i2s_alloc_dma_desc'
components/esp_driver_i2s/i2s_common.c:esp_err_t i2s_alloc_dma_desc(i2s_chan_handle_t handle, uint32_t num, uint32_t bufsize)
components/esp_driver_i2s/i2s_pdm.c:        ESP_RETURN_ON_ERROR(i2s_alloc_dma_desc(handle, handle->dma.desc_num, buf_size),
components/esp_driver_i2s/i2s_pdm.c:        ESP_RETURN_ON_ERROR(i2s_alloc_dma_desc(handle, handle->dma.desc_num, buf_size),
components/esp_driver_i2s/i2s_private.h:esp_err_t i2s_alloc_dma_desc(i2s_chan_handle_t handle, uint32_t num, uint32_t bufsize);
components/esp_driver_i2s/i2s_std.c:        ESP_RETURN_ON_ERROR(i2s_alloc_dma_desc(handle, handle->dma.desc_num, buf_size),
components/esp_driver_i2s/i2s_tdm.c:        ESP_RETURN_ON_ERROR(i2s_alloc_dma_desc(handle, handle->dma.desc_num, buf_size),
```

L-KAYA, 2025-03-28:
You're right! Then maybe `handle->dma.desc_num` is not necessary to pass in, just move it from the parameter list

kasperl, 2025-03-28:
> You're right! Then maybe `handle->dma.desc_num` is not necessary to pass in, just move it from the parameter list

That would be nice and make it clearer that `handle->dma.desc_num` is a configured field that isn't supposed to be changed after initialization. Thanks for looking into this! :)

closed 2025-04-02 by espressif-bot