espressif/esp-idf #15649: [Provisioning via NimBLE] ESP32S3 has been reset after de-init provisioning and start NimBle ESP-IDF V5.4.0

tranthanh1699, 2025-03-25:
Hello everyone,

While developing the feature for provisioning WiFi credentials via BLE using NimBLE, I encountered an issue. After successfully completing the provisioning process and connecting the ESP32 to WiFi, I de-initialize the provisioning service using the wifi_prov_mgr_deinit() API.

However, when I attempt to start a new BLE service (NOTIFY) with notifications for logging via BLE using the API below, the ESP32 resets unexpectedly

void ble_init(void)
{
    bleQueue = xQueueCreate(20, sizeof(char *));
    nvs_flash_init();					  // 1 - Initialize NVS flash using
	nimble_port_init();					  // 3 - Initialize the host stack

	ble_svc_gap_device_name_set("BLE_Logger"); // 4 - Initialize NimBLE configuration - server name
	ble_svc_gap_init();						   // 4 - Initialize NimBLE configuration - gap service
	ble_svc_gatt_init();					   // 4 - Initialize NimBLE configuration - gatt service

	ble_gatts_count_cfg(gatt_svcs);		  // 4 - Initialize NimBLE configuration - config gatt services
	ble_gatts_add_svcs(gatt_svcs);		  // 4 - Initialize NimBLE configuration - queues gatt services.
	ble_hs_cfg.sync_cb = ble_app_on_sync; // 5 - Initialize application
	xTaskCreate(ble_notify_task, "ble_notify_task", 4096, NULL, 5, &bleNotifyTask);
	nimble_port_freertos_init(host_task); // 6 - Run the thread
}

rahult-github, 2025-03-26:
Hi @tranthanh1699 ,  the default behaviour is that once provisioning is completed, since there is no further use of BLE, so the memory used by stack is released back.  

I suggest to please enable [CONFIG_WIFI_PROV_KEEP_BLE_ON_AFTER_PROV](https://github.com/espressif/esp-idf/blob/master/components/wifi_provisioning/Kconfig#L47)  which will ensure BLE memory is not released and check your use case 

If the above suggestion still doesn't work for you, request you to please share your application code changes and sdkconfig so that we can reproduce the same at our end and debug more .

rahult-github, 2025-04-25:
Closing this issue. Feel free to reopen in case of any further queries.

closed 2025-04-25 by rahult-github