espressif/esp-idf #15650: Issues Configuring HID and CDC Composite Device on ESP32-S3

Talha-Babar, 2025-03-25:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hello everyone,​

I'm currently working on a project involving the ESP32-S3. I aim to configure it as a composite USB device supporting both HID and CDC functionalities using the TinyUSB stack. I want to use Gamepad and CDC at the same time.

Development Environment:

Hardware: ESP32-S3​
Framework: ESP-IDF v5.0​
USB Stack: TinyUSB integrated with ESP-IDF​

Here is the code:
`
#define CONFIG_TINYUSB_CDC_COUNT 1

static const char *TAG = "CDC+HID";
static uint8_t buf[CONFIG_TINYUSB_CDC_RX_BUFSIZE + 1];

#define TUSB_DESC_TOTAL_LEN (TUD_CONFIG_DESC_LEN + CFG_TUD_CDC * TUD_CDC_DESC_LEN + CFG_TUD_HID * TUD_HID_DESC_LEN)

const uint8_t hid_report_descriptor[] = {
TUD_HID_REPORT_DESC_GAMEPAD(HID_REPORT_ID(3))
};

/**

@brief String descriptor
*/
const char *hid_string_descriptor[5] = {
// array of pointer to string descriptors
(char[]){0x09, 0x04}, // 0: is supported language is English (0x0409)
"TinyUSB", // 1: Manufacturer
"CDC+HID", // 2: Product
"123456", // 3: Serials, should use chip ID
"HID interface", // 4: HID
};
static const uint8_t hid_configuration_descriptor[] = {
// Configuration Descriptor: 1 configuration, 1 HID interface, 1 CDC interface
TUD_CONFIG_DESCRIPTOR(2, 2, 0, TUSB_DESC_TOTAL_LEN, TUSB_DESC_CONFIG_ATT_REMOTE_WAKEUP, 100),

// CDC Interface Descriptor
TUD_CDC_DESCRIPTOR(2, 0, 0x81, 16, 0x82, 0x83, 64),

// HID Interface Descriptor
TUD_HID_DESCRIPTOR(1, 4, false, sizeof(hid_report_descriptor), 0x81, 16, 10), // HID interface with interrupt IN endpoint
};

uint8_t const *tud_hid_descriptor_report_cb(uint8_t instance)
{
// We use only one interface and one HID report descriptor, so we can ignore parameter 'instance'
return hid_report_descriptor;
}

// Invoked when received GET_REPORT control request
// Application must fill buffer report's content and return its length.
// Return zero will cause the stack to STALL request
uint16_t tud_hid_get_report_cb(uint8_t instance, uint8_t report_id, hid_report_type_t report_type, uint8_t *buffer, uint16_t reqlen)
{
(void)instance;
(void)report_id;
(void)report_type;
(void)buffer;
(void)reqlen;

return 0;
}

// Invoked when received SET_REPORT control request or
// received data on OUT endpoint ( Report ID = 0, Type = 0 )
void tud_hid_set_report_cb(uint8_t instance, uint8_t report_id, hid_report_type_t report_type, uint8_t const *buffer, uint16_t bufsize)
{
}

void tinyusb_cdc_rx_callback(int itf, cdcacm_event_t *event)
{
size_t rx_size = 0;
esp_err_t ret = tinyusb_cdcacm_read(itf, buf, CONFIG_TINYUSB_CDC_RX_BUFSIZE, &rx_size);
if (ret == ESP_OK)
{
ESP_LOGI(TAG, "Data from channel %d:", itf);
ESP_LOG_BUFFER_HEXDUMP(TAG, buf, rx_size, ESP_LOG_INFO);
}
else
{
ESP_LOGE(TAG, "Read error");
}
tinyusb_cdcacm_write_queue(itf, buf, rx_size);
tinyusb_cdcacm_write_flush(itf, 0);
}

void tinyusb_cdc_line_state_changed_callback(int itf, cdcacm_event_t *event)
{
int dtr = event->line_state_changed_data.dtr;
int rts = event->line_state_changed_data.rts;
ESP_LOGI(TAG, "Line state changed on channel %d: DTR:%d, RTS:%d", itf, dtr, rts);
}

void app_main(void)
{
ESP_LOGI(TAG, "USB initialization");
const tinyusb_config_t tusb_cfg = {
.device_descriptor = NULL,
.string_descriptor = hid_string_descriptor,
.external_phy = false,
.configuration_descriptor = hid_configuration_descriptor,
};
ESP_ERROR_CHECK(tinyusb_driver_install(&tusb_cfg));

tinyusb_config_cdcacm_t acm_cfg = {
    .usb_dev = TINYUSB_USBDEV_0,
    .cdc_port = TINYUSB_CDC_ACM_0,
    .rx_unread_buf_sz = 64,
    .callback_rx = &tinyusb_cdc_rx_callback,
    .callback_rx_wanted_char = NULL,
    .callback_line_state_changed = NULL,
    .callback_line_coding_changed = NULL
};
`
ESP_ERROR_CHECK(tusb_cdc_acm_init(&acm_cfg));

ESP_ERROR_CHECK(tinyusb_cdcacm_register_callback(
    TINYUSB_CDC_ACM_0,
    CDC_EVENT_LINE_STATE_CHANGED,
    &tinyusb_cdc_line_state_changed_callback));

ESP_LOGI(TAG, "USB initialization DONE");

const char *msg = "Hello, World!\n";

while (1)
{
    if (tud_mounted())
    {
        static bool send_hid_data = true;

        tud_hid_n_gamepad_report(0, 3, 0, 0, 0, 0, 0, 0, 0, 5);
    }
    vTaskDelay(pdMS_TO_TICKS(100));
}

// while (1)
// {
// vTaskDelay(10000 / portTICK_PERIOD_MS);
// tinyusb_cdcacm_write_queue(TINYUSB_CDC_ACM_0, (uint8_t *)msg, strlen(msg));
// tinyusb_cdcacm_write_flush(TINYUSB_CDC_ACM_0, 0);
// }

// tinyusb_cdcacm_write_queue_char(TINYUSB_CDC_ACM_1, "T");
}
`
Upon connecting the ESP32-S3 to a host system, the device is not recognized.
Thank you for your time and assistance.

Best regards,

igi540, 2025-03-26:
Hi @Talha-Babar 

Thanks for sharing the detailed context and code!

I reviewed your composite USB (CDC + HID Gamepad) setup with TinyUSB on ESP32-S3 and here are key issues & suggestions to get it working:

---

###  1. Interface count mismatch

You're using `TUD_CONFIG_DESCRIPTOR(2, 2, ...)`, but CDC actually uses **2 interfaces** (control + data), and HID adds 1 → **total = 3 interfaces**.

#### Fix:
```c
#define ITF_NUM_CDC       0
#define ITF_NUM_CDC_DATA  1
#define ITF_NUM_HID       2
#define ITF_NUM_TOTAL     3
```
Then in your config descriptor:
```c
TUD_CONFIG_DESCRIPTOR(1, ITF_NUM_TOTAL, 0, TUSB_DESC_TOTAL_LEN, TUSB_DESC_CONFIG_ATT_REMOTE_WAKEUP, 100),
TUD_CDC_DESCRIPTOR(ITF_NUM_CDC, 0, 0x81, 8, 0x82, 0x83, 64),
TUD_HID_DESCRIPTOR(ITF_NUM_HID, 4, false, sizeof(hid_report_descriptor), 0x84, 16, 10),
```
Make sure endpoint addresses are unique (`0x81`, `0x82`, `0x83` for CDC, and `0x84` for HID IN).

### 2. Device & config descriptor callbacks missing
If `tinyusb_config_t.device_descriptor = NULL`, TinyUSB expects these callbacks:
```c
uint8_t const *tud_descriptor_device_cb(void) {
  static tusb_desc_device_t desc = {
    .bLength = sizeof(desc),
    .bDescriptorType = TUSB_DESC_DEVICE,
    .bcdUSB = 0x0200,
    .bDeviceClass = TUSB_CLASS_MISC,
    .bDeviceSubClass = MISC_SUBCLASS_COMMON,
    .bDeviceProtocol = MISC_PROTOCOL_IAD,
    .bMaxPacketSize0 = 64,
    .idVendor = 0xCafe,
    .idProduct = 0x4000,
    .bcdDevice = 0x0100,
    .iManufacturer = 0x01,
    .iProduct = 0x02,
    .iSerialNumber = 0x03,
    .bNumConfigurations = 1
  };
  return (uint8_t const *) &desc;
}

uint8_t const *tud_descriptor_configuration_cb(uint8_t index) {
  (void) index;
  return hid_configuration_descriptor;
}
```
### 3. Incorrect Language ID string descriptor:
Your original language ID descriptor `(char[]){0x09, 0x04}` is incorrect. USB spec requires it to have length and type explicitly set:
```c
const uint8_t desc_str_langid[] = {
  0x04, TUSB_DESC_STRING, 0x09, 0x04 // English (US)
};

const char* string_desc_arr[] = {
  (const char*)desc_str_langid,
  "TinyUSB",
  "CDC+HID",
  "123456",
  "HID interface",
};
```

After applying these corrections, your ESP32-S3 should correctly enumerate as a composite CDC + HID USB device.

Hope this helps you resolve your issue! Let me know if you need anything else.

Talha-Babar, 2025-03-27:
Hello @igi540 ,

Thank you for your help, i have made changes according to your suggestions but still the devices is not recognise. I am again providing my complete code if there is any issue in other portion of the code. 

I am also providing the tinyusb portion of sdkconfig file.

# code:
`
#include "class/hid/hid_device.h"
#include "driver/gpio.h"
#include "esp_log.h"
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "sdkconfig.h"
#include "tinyusb.h"
#include "tusb_cdc_acm.h"
#include <stdint.h>

#define CONFIG_TINYUSB_CDC_COUNT 1

static const char *TAG = "CDC+HID";
static uint8_t buf[CONFIG_TINYUSB_CDC_RX_BUFSIZE + 1];

// Interface definitions
#define ITF_NUM_CDC 0
#define ITF_NUM_CDC_DATA 1
#define ITF_NUM_HID 2
#define ITF_NUM_TOTAL 3

#define TUSB_DESC_TOTAL_LEN                                                    \
  (TUD_CONFIG_DESC_LEN + CFG_TUD_CDC * TUD_CDC_DESC_LEN +                      \
   CFG_TUD_HID * TUD_HID_DESC_LEN)

const uint8_t hid_report_descriptor[] = {
    TUD_HID_REPORT_DESC_GAMEPAD(HID_REPORT_ID(3))};

// Corrected string descriptor
const uint8_t desc_str_langid[] = {
    0x04, TUSB_DESC_STRING, 0x09, 0x04 // English (US)
};

const char *string_desc_arr[] = {
    (const char *)desc_str_langid,
    "TinyUSB",
    "CDC+HID",
    "123456",
    "HID interface",
};

static const uint8_t hid_configuration_descriptor[] = {
    TUD_CONFIG_DESCRIPTOR(1, ITF_NUM_TOTAL, 0, TUSB_DESC_TOTAL_LEN,
                          TUSB_DESC_CONFIG_ATT_REMOTE_WAKEUP, 100),
    TUD_CDC_DESCRIPTOR(ITF_NUM_CDC, 0, 0x81, 8, 0x82, 0x83, 64),
    TUD_HID_DESCRIPTOR(ITF_NUM_HID, 4, false, sizeof(hid_report_descriptor),
                       0x84, 16, 10),
};

static tusb_desc_device_t my_custom_device_desc = {
    .bLength = sizeof(tusb_desc_device_t),
    .bDescriptorType = TUSB_DESC_DEVICE,
    .bcdUSB = 0x0200,
    .bDeviceClass = TUSB_CLASS_MISC,
    .bDeviceSubClass = MISC_SUBCLASS_COMMON,
    .bDeviceProtocol = MISC_PROTOCOL_IAD,
    .bMaxPacketSize0 = 64,
    .idVendor = 0xCafe,  // Change this to your vendor ID
    .idProduct = 0x4000, // Change this to your product ID
    .bcdDevice = 0x0100,
    .iManufacturer = 1,
    .iProduct = 2,
    .iSerialNumber = 3,
    .bNumConfigurations = 1};

uint8_t const *tud_hid_descriptor_report_cb(uint8_t instance) {
  return hid_report_descriptor;
}

uint16_t tud_hid_get_report_cb(uint8_t instance, uint8_t report_id,
                               hid_report_type_t report_type, uint8_t *buffer,
                               uint16_t reqlen) {
  (void)instance;
  (void)report_id;
  (void)report_type;
  (void)buffer;
  (void)reqlen;
  return 0;
}

void tud_hid_set_report_cb(uint8_t instance, uint8_t report_id,
                           hid_report_type_t report_type, uint8_t const *buffer,
                           uint16_t bufsize) {}

void tinyusb_cdc_rx_callback(int itf, cdcacm_event_t *event) {
  size_t rx_size = 0;
  esp_err_t ret =
      tinyusb_cdcacm_read(itf, buf, CONFIG_TINYUSB_CDC_RX_BUFSIZE, &rx_size);
  if (ret == ESP_OK) {
    ESP_LOGI(TAG, "Data from channel %d:", itf);
    ESP_LOG_BUFFER_HEXDUMP(TAG, buf, rx_size, ESP_LOG_INFO);
  } else {
    ESP_LOGE(TAG, "Read error");
  }
  tinyusb_cdcacm_write_queue(itf, buf, rx_size);
  tinyusb_cdcacm_write_flush(itf, 0);
}

void tinyusb_cdc_line_state_changed_callback(int itf, cdcacm_event_t *event) {
  int dtr = event->line_state_changed_data.dtr;
  int rts = event->line_state_changed_data.rts;
  ESP_LOGI(TAG, "Line state changed on channel %d: DTR:%d, RTS:%d", itf, dtr,
           rts);
}

void app_main(void) {
  ESP_LOGI(TAG, "USB initialization");
  const tinyusb_config_t tusb_cfg = {
      .device_descriptor = &my_custom_device_desc,
      .string_descriptor = string_desc_arr,
      .external_phy = false,
      .configuration_descriptor = hid_configuration_descriptor,
  };

  ESP_ERROR_CHECK(tinyusb_driver_install(&tusb_cfg));

  tinyusb_config_cdcacm_t acm_cfg = {.usb_dev = TINYUSB_USBDEV_0,
                                     .cdc_port = TINYUSB_CDC_ACM_0,
                                     .rx_unread_buf_sz = 64,
                                     .callback_rx = &tinyusb_cdc_rx_callback,
                                     .callback_rx_wanted_char = NULL,
                                     .callback_line_state_changed = NULL,
                                     .callback_line_coding_changed = NULL};

  ESP_ERROR_CHECK(tusb_cdc_acm_init(&acm_cfg));
  ESP_ERROR_CHECK(tinyusb_cdcacm_register_callback(
      TINYUSB_CDC_ACM_0, CDC_EVENT_LINE_STATE_CHANGED,
      &tinyusb_cdc_line_state_changed_callback));

  ESP_LOGI(TAG, "USB initialization DONE");

  while (1) {
    if (tud_mounted()) {
      tud_hid_n_gamepad_report(0, 3, 0, 0, 0, 0, 0, 0, 0, 5);
    }
    vTaskDelay(pdMS_TO_TICKS(100));
  }
}
`

# sdkconfig file

`

CONFIG_TINYUSB_DEBUG_LEVEL=1
CONFIG_TINYUSB_RHPORT_FS=y

CONFIG_TINYUSB_MODE_DMA=y

CONFIG_TINYUSB_TASK_PRIORITY=5
CONFIG_TINYUSB_TASK_STACK_SIZE=4096

CONFIG_TINYUSB_TASK_AFFINITY_CPU1=y
CONFIG_TINYUSB_TASK_AFFINITY=0x1

CONFIG_TINYUSB_DESC_CUSTOM_VID=0xCafe

CONFIG_TINYUSB_DESC_CUSTOM_PID=0x4000
CONFIG_TINYUSB_DESC_BCD_DEVICE=0x0100
CONFIG_TINYUSB_DESC_MANUFACTURER_STRING="Espressif Systems"
CONFIG_TINYUSB_DESC_PRODUCT_STRING="Espressif Device"
CONFIG_TINYUSB_DESC_SERIAL_STRING="123456"
CONFIG_TINYUSB_DESC_CDC_STRING="Espressif CDC Device"

CONFIG_TINYUSB_CDC_ENABLED=y
CONFIG_TINYUSB_CDC_COUNT=2
CONFIG_TINYUSB_CDC_RX_BUFSIZE=512
CONFIG_TINYUSB_CDC_TX_BUFSIZE=512

CONFIG_TINYUSB_MIDI_COUNT=0

CONFIG_TINYUSB_HID_COUNT=3

Talha-Babar, 2025-03-29:
Hello @igi540,

I wanted to share an update regarding the issue we discussed earlier. It turned out that the problem was related to the menuconfig settings. After updating the configuration as follows:
`
CDC Channel Count = 2
TinyUSB HID Interfaces Count = 1
`

Windows now correctly detects the HID device (a gamepad), and I'm able to see gamepad input working properly.
However, there are still a couple of issues:

- The HID device does not appear under Device Manager, although it works fine in the "Game Controllers" panel.

- The CDC Virtual COM port is listed in Device Manager, but it does not open for read/write operations.

So it seems we’re very close, but this remaining issue with the CDC port is holding things back.
Would appreciate any guidance on how to resolve this last part.
Thanks again for your help! 😊

igi540, 2025-03-31:
Hi @Talha-Babar,

You're very close — great job getting the HID Gamepad working! The remaining CDC issue likely could be fixed by these steps:

CDC uses two interfaces (Control + Data), and they must be grouped with an IAD for Windows to recognize them as one functional unit.
Fix: Update your CDC descriptor line to this:
`TUD_CDC_DESCRIPTOR(ITF_NUM_CDC, ITF_NUM_CDC_DATA, 0x81, 8, 0x82, 0x83, 64),
`
`.idVendor = 0xCafe
.idProduct = 0x4000
`
These are non-standard. Windows may not associate any driver by default.

Fix: Try using Espressif's registered VID/PID for CDC:
`.idVendor = 0x303A
.idProduct = 0x4001
`
tip:
Increase TinyUSB debug level in sdkconfig. This will help you inspect enumeration logs and spot if any descriptors are rejected by the host.

Talha-Babar, 2025-04-01:
Hello @igi540,

Thank you for your help! We’ve finally resolved the issue. I used the following configuration, and now both are working fine:
`
TUD_CDC_DESCRIPTOR(ITF_NUM_CDC, 0, 0x81, 8, 0x82, 0x04, 64)  
`
I have one more request—do you know of any documentation where I can learn more about configuring devices, understanding the parameters, and their specific purposes?

Thanks again!

igi540, 2025-04-01:
Hi @Talha-Babar,

That’s awesome to hear — I’m really glad everything is working now! 
Thanks for the update and for sharing the final working configuration. It's always rewarding to see these composite USB setups come together.

As for learning resources and documentation on USB descriptors, TinyUSB, and general USB device configuration, here are a few recommendations:
[Espressif documentation — details on configuring descriptors with TinyUSB](https://docs.espressif.com/projects/esp-idf/en/v5.4.1/esp32s3/api-reference/peripherals/usb_device.html#descriptor-configuration)
[Official API and descriptor macro reference](https://docs.tinyusb.org/en/stable/reference/)
[Great for studying how composite devices are structured](https://github.com/hathach/tinyusb/tree/master/examples)
[Classic explanation of USB protocol and descriptor structure](http://www.beyondlogic.org/usbnutshell/usb1.shtml)
[Very helpful for visualizing and validating descriptor bytes](https://eleccelerator.com/usbdescreqparser/)

closed 2025-04-01 by Indastri