espressif/esp-idf #15652: Add UART example where it is setup using raw registers and is controlled by interrupts and ISRs

arasan90, 2025-03-26:
### Is your feature request related to a problem?

Yes, I am trying to write a UART abstraction layer that only uses the hardware instead of hardware + OS for speed reasons.

### Describe the solution you'd like.

I would like to have an example where the UART setup and usage are performed via raw registers and interrupts

zekageri, 2025-03-27:
Iam about to write my own driver because the idf driver has a lot of flaws. I need dma and proper interrupt handling. With the queue based approach the esp either loses the first two or three bytes or entire packets. Also the write function is unusable because it needs a lot of time to toggle the rts pin after all data sent out.

LYNHQQ, 2025-04-01:
Here are some implementations, it may be useful to you for a long time。https://github.com/[EternityFOR/Personal-Uart_DMA](https://github.com/EternityFOR/Personal-Uart_DMA)

arasan90, 2025-04-03:
Thank you @LYNHQQ  but this is still using the driver_install API