espressif/esp-idf #15653: OFF-PACKAGE QUAD PSRAM DOES NOT WORK WITH ESP32S3

MR-VENOM, 2025-03-26:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.2.3

### Espressif SoC revision.

esp32s3

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom Board

### Power Supply used.

USB

### What is the expected behavior?

I've designed a custom PCB with an 8MB flash chip in-package and an external 8MB quad ESP PSRAM. When i try to enable the PSRAM i am getting the bellow error.  I switched the mode from OCTAL to QUAD in menuconfig, but the error still persists:  

```
I (271) cpu_start: Multicore app  
E (271) octal_psram: PSRAM ID read error: 0x00000000, PSRAM chip not found or not supported, or wrong PSRAM line mode  
E (275) cpu_start: Failed to init external RAM!  
```

i am attaching the sdkconfig and the schematics of my PCB.

[PCB](https://github.com/user-attachments/files/19463289/Screenshot.2025-03-26.at.12.51.32.PM.pdf)

[sdkconfig](https://github.com/user-attachments/files/19463315/sdkconfig.txt)

### What is the actual behavior?

The PSRAM should be configured.

### Steps to reproduce.

1. Change menuconfig to enbale QUAD PSRAM.
2. Build
3. Flash
4. Monitor

### Debug Logs.

```plain
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3820,len:0x568
load:0x403c9700,len:0x4
load:0x403c9704,len:0xb5c
load:0x403cc700,len:0x2c34
entry 0x403c98f0
I (271) cpu_start: Multicore app
E (271) octal_psram: PSRAM ID read error: 0x00000000, PSRAM chip not found or not supported, or wrong PSRAM line mode
E (275) cpu_start: Failed to init external RAM!
```

### More Information.

[diag-837718fc-8703-4e8b-a816-440dad8b2510.zip](https://github.com/user-attachments/files/19463397/diag-837718fc-8703-4e8b-a816-440dad8b2510.zip)

suda-morris, 2025-03-31:
Hope this doc can give you some help: https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-guides/flash_psram_config.html#how-to-configure-flash-and-psram

MR-VENOM, 2025-03-31:
@suda-morris  Thank you for the doc we have gone through it and have not any solution from it. And all it talks about is DTR mode for octal . No STR mode for quad spi.

Icarus113, 2025-04-07:
Could you help check if this commit 6f8da01e388543487ff025f051f108574ee66a1b is included in your branch? @MR-VENOM

MR-VENOM, 2025-04-07:
@Icarus113  Yes i have that commit on my branch here is the file : 

[esp_psram_impl_quad.txt](https://github.com/user-attachments/files/19629616/esp_psram_impl_quad.txt)

Icarus113, 2025-04-14:
> octal_psram: PSRAM ID read error: 0x00000000

this log means the PSRAM ID that the ESP reads from the on-chip PSRAM chip is all 0. 

> 8MB quad ESP PSRAM.

Would you help check if it's the ID calculation in the driver doesn't match with your PSRAM chip?

Alvin1Zhang, 2025-05-07:
@MR-VENOM Thanks for reporting, would you please help share if any further updates for the issue? Thanks.

ginkgm, 2025-05-23:
Hi @MR-VENOM ,

Quality of your schematic is too low so I'm not sure whether you are using module or SoC. But I guess it's SoC.

The only model matches your description is ESP32-S3FN8, is this correct?  You can check it by:

https://docs.espressif.com/projects/esp-packaging/en/latest/esp32s3/01-marking/index_chip.html
https://www.espressif.com/sites/default/files/documentation/esp32-s3_datasheet_en.pdf (section 1.2)

And If I'm correct, the PSRAM you used should be supplied with 3.3V voltage. Is this satisfied? And what's the model of the PSRAM?

ginkgm, 2025-05-23:
There are two different 8M PSRAM from Espressif: ESP-PSRAM64 and ESP-PSRAM64H.

ESP-PSRAM64 requires an 1.8V power supply while ESP-PSRAM64H works under 3.3V. 

You can distinguish between them from the laser mark of the package. (`ESP-PSRAM64H` is printed on the package)

closed 2025-06-30 by ginkgm