espressif/esp-idf #15654: Problems with spi_lcd_touch_example

ManelCS, 2025-03-26:
I'm trying to follow this example
[https://github.com/espressif/esp-idf/tree/master/examples/peripherals/lcd/spi_lcd_touch](url)
with a ESP32-s3 and the display NHD-2.4-240320AF-CSXP but I only see the backlight on the screen. The connection is:

[image]

`#define EXAMPLE_LCD_PIXEL_CLOCK_HZ     (15 * 1000 * 1000)`
`#define EXAMPLE_LCD_BK_LIGHT_ON_LEVEL  1`
`#define EXAMPLE_LCD_BK_LIGHT_OFF_LEVEL !EXAMPLE_LCD_BK_LIGHT_ON_LEVEL`
`#define EXAMPLE_PIN_NUM_SCLK           14       // tft dcx `
`#define EXAMPLE_PIN_NUM_MOSI           47       // tft sda`
`#define EXAMPLE_PIN_NUM_MISO           38       // tft sdo`
`#define EXAMPLE_PIN_NUM_LCD_DC         11`
`#define EXAMPLE_PIN_NUM_LCD_RST        12       // tft resx`
`#define EXAMPLE_PIN_NUM_LCD_CS         45       // tft csx`
`#define EXAMPLE_PIN_NUM_BK_LIGHT       4`
`#define EXAMPLE_PIN_NUM_TOUCH_CS       -1`

also i change the line
`ESP_ERROR_CHECK(esp_lcd_new_panel_ili9341(io_handle, &panel_config, &panel_handle));`
for
`ESP_ERROR_CHECK(esp_lcd_new_panel_st7789(io_handle, &panel_config, &panel_handle));`

suda-morris, 2025-03-27:
I'm not familar with NHD-2.4-240320AF-CSXP, but from the picture, I don't know why you connect the "DCX" to the "SCL"? "SCL" is the SPI clock signal.

closed 2025-03-27 by suda-morris

ManelCS, 2025-04-03:
> I'm not familar with NHD-2.4-240320AF-CSXP, but from the picture, I don't know why you connect the "DCX" to the "SCL"? "SCL" is the SPI clock signal. "DCX" is the "Data/Command selection".
> 
> If you post such questions in the [ESP32 Forum](https://esp32.com/), you will get more help :)

The photo is from the datasheet of the display I use:
[https://newhavendisplay.com/content/specs/NHD-2.4-240320AF-CSXP.pdf](url)