espressif/esp-idf #15655: ADC buffer size

crivi30, 2025-03-26:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

@Bruce297 @Icarus113 Hello, I have a question on the ADC buffer size calculation, in continuous mode. 

Based on the official doc, it seems that in `.conv_frame_size`, we need to consider how many channels we are using:

```c
#define FRAME_SIZE ((CHANNEL_NUMS) * (SAMPLES_PER_CHANNEL) * sizeof(adc_digi_output_data_t))
```

However, in the ADC library (specifically in `adc_continuous.c`), I noticed that memory is allocated as follows (conv_frame_size*num channels):

```c
adc_ctx->rx_dma_buf = heap_caps_calloc(INTERNAL_BUF_NUM, hdl_config->conv_frame_size, MALLOC_CAP_INTERNAL | MALLOC_CAP_DMA | MALLOC_CAP_8BIT);
```

This implies that the correct `FRAME_SIZE` definition should be:

```c
#define FRAME_SIZE ((SAMPLES_PER_CHANNEL) * sizeof(adc_digi_output_data_t))
```

without multiplying by the number of channels in advance. Is this correct?

### Additional Observation:

If I set POOL_SIZE = FRAME_SIZE*2 for example and I'm using 5 channels,
the memory allocated for the conversion frame appears to be larger than the max buffer size because the last, it is multiplied by the number of channels:

```c
adc_ctx->ringbuf_storage = heap_caps_calloc(1, hdl_config->max_store_buf_size, MALLOC_CAP_INTERNAL | MALLOC_CAP_8BIT);
adc_ctx->rx_dma_buf = heap_caps_calloc(INTERNAL_BUF_NUM, hdl_config->conv_frame_size, MALLOC_CAP_INTERNAL | MALLOC_CAP_DMA | MALLOC_CAP_8BIT);
```

How is the correct way to set the size of conversion frame and the max buffer size?
Thank you in advance!

Bruce297, 2025-03-27:
Hello @crivi30 You only need to define 
`#define FRAME_SIZE ((SAMPLES_PER_CHANNEL) * sizeof(adc_digi_output_data_t))`

 rather than multiplying by the number of channels in advance.
ADC continuous-mode driver internally processes multi-channel sampling, interlacing the sampling data from each channel into a conversion frame. 

And the max_store_buf_size is the maximum number of bytes for the entire internal circular buffer. There is no need to multiply it by the number of channels because each conversion frame already contains the data of all channels. 
After setting it this way, the driver will automatically allocate the required memory according to the number of internal buffers (INTERNAL_BUF_NUM) when allocating the DMA buffer, and unexpected additional memory allocation issues will not occur.

crivi30, 2025-03-27:
Let's say I want to read 128 samples for 5 channels (128 x 5 = 640).

```c
#define NUM_CHANNELS 5
#define SAMPLES 128
#define FRAME (SAMPLES * sizeof(adc_digi_output_data_t))
```

If I want to store only one conversion frame, should I set `MAX_BYTE_RNG` as:

```c
#define MAX_BYTE_RNG FRAME    
```

Then, configuring the ADC handle:

```c
adc_continuous_handle_cfg_t adc_config;
adc_config.max_store_buf_size = MAX_BYTE_RNG;
adc_config.conv_frame_size = FRAME;
```

When reading the returned buffer:

```c
int max_len = FRAME * NUM_CHANNELS;
esp_err_t ret;
uint8_t result[max_len] = {0};

ret = adc_continuous_read(stream_handle, result, max_len, len_out, 0);
```

In this example, with these buffer size settings, I receive a random `len_out` buffer size, that can be 128, 256, or sometimes 640 as expected. 

Thank you so much for the clarifications!

crivi30, 2025-03-30:
Hello @Bruce297, any thoughts on this?
Thanks

crivi30, 2025-04-03:
Hello @Bruce297 and @Icarus113, any thoughts on this?
Thanks

crivi30, 2025-04-03:
Just checking if anyone has any suggestions or insights on this?

crivi30, 2025-04-04:
Hello! I think the confusion stems from this line in adc_continuous.c:

`adc_ctx->rx_dma_buf = heap_caps_calloc(INTERNAL_BUF_NUM, hdl_config->conv_frame_size, MALLOC_CAP_INTERNAL | MALLOC_CAP_DMA | MALLOC_CAP_8BIT);`

Initially, I assumed `INTERNAL_BUF_NUM` represented the number of ADC channels I wanted to read, but it appears to be a hardcoded value — currently set to 5, no matter how many channel i'm using.

So my question is:

What exactly is `INTERNAL_BUF_NUM`, and why is the DMA buffer size allocated as 5 × `conv_frame_size`?

Thanks in advance!

Bruce297, 2025-05-05:
Hi, for now, the num of ADC DMA descriptor is always set to 5, it's the least num to avoid the frames lost  due to the delay between DMA writes and application reads.

I think what you concerned should be conv_frame_size.

crivi30, 2025-05-05:
Ok, thank you!