espressif/esp-idf #15658: ninja error "libesp_zb_api.ed.a" needed by "light_bulb.elf" missing and no known rule to make it

boemekeld, 2025-03-26:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.4

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

CMD

### What is the expected behavior?

That the build completed without errors.

### What is the actual behavior?

```C:\Espressif\frameworks\esp-idf-v5.4\examples\zigbee\light_sample\HA_on_off_light>idf.py build
Executing action: all (aliases: build)
Running ninja in directory C:\Espressif\frameworks\esp-idf-v5.4\examples\zigbee\light_sample\HA_on_off_light\build
Executing "ninja all"...
ninja: error: 'C:/Espressif/frameworks/esp-idf-v5.4/examples/zigbee/light_sample/HA_on_off_light/managed_components/espressif__esp-zigbee-lib/lib/esp32/libesp_zb_api.ed.a', needed by 'light_bulb.elf', missing and no known rule to make it
ninja failed with exit code 1, output of the command is in the C:\Espressif\frameworks\esp-idf-v5.4\examples\zigbee\light_sample\HA_on_off_light\build\log\idf_py_stderr_output_26140 and C:\Espressif\frameworks\esp-idf-v5.4\examples\zigbee\light_sample\HA_on_off_light\build\log\idf_py_stdout_output_26140

### Steps to reproduce.

1. Install ESP-IDF through Windows Installer;
2. Run the shortcut "ESP-IDF 5.4 CMD" created on the Desktop;
3. Run:
 - cd C:\Espressif\frameworks\esp-idf-v5.4\examples\zigbee\light_sample\HA_on_off_light
 - idf.py build

### Build or installation Logs.

```plain
ninja: error: 'C:/Espressif/frameworks/esp-idf-v5.4/examples/zigbee/light_sample/HA_on_off_light/managed_components/espressif__esp-zigbee-lib/lib/esp32/libesp_zb_api.ed.a', needed by 'light_bulb.elf', missing and no known rule to make it
```

xieqinan, 2025-03-27:
Hi,

Since the Zigbee light can only run on an 802.15.4-enabled chip, could you try running `idf.py set-target esp32h2` before `idf.py build`?

boemekeld, 2025-03-27:
@xieqinan Hi, yes, that works! Thanks, man.

closed 2025-03-27 by boemekeld