espressif/esp-idf #15660: Why NimBLE throughput demo much slower than Bluedroid ble50_throughput demo?

dukelec, 2025-03-26:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Both on 2M PHY

The speed test provided in the README.md of the relevant demo is as follows:

nimble throughput_app max speed: ~500 kbps

ble50_throughput max speed: ~1401 kbps

dukelec, 2025-03-26:
I ultimately need to use L2CAP CoC for communication since CoC is theoretically faster than GATT.
However, since ESP only supports L2CAP CoC with NimBLE, it becomes meaningless because NimBLE is too slow.

Shreeyash-17, 2025-03-27:
@dukelec According to the information you provided, you are comparing the **throughput_app** NimBLE throughput example with the **ble50_throughput** Bluedroid example on **2M PHY** for write operation. However, the NimBLE example currently does not support 2M PHY and is running on 1M PHY. This means you are comparing the NimBLE throughput example on 1M PHY with the Bluedroid example on 2M PHY, which naturally results in a significant throughput difference.

I am currently working on adding 2M PHY support for the NimBLE example.

You can try comparing the **throughput_app** NimBLE throughput example with the **ble_throughput** (not ble50_throughput) Bluedroid example on 1M PHY.

Additionally, I am sharing a patch with you. Apply the patch and make the following changes in menuconfig:
Navigate to Component config → Bluetooth → NimBLE Options → NimBLE Host log verbosity, and select No logs on the
 blecent_throughput side. Then, build and run the example.

[image]

 
path to apply patch : esp-idf/
[blecent_throughput.txt](https://github.com/user-attachments/files/19480837/blecent_throughput.txt)

dukelec, 2025-03-31:
Thank you for your reply.

I previously tested the write speed of the NimBLE throughput demo with 2M PHY enabled. At that time, the speed was around ~500 kbps. Manually commenting out the relevant print statements only resulted in a slight improvement. However, after disabling logging via menuconfig as you suggested, the speed increased to ~794 kbps. Further, by adjusting the connection interval from `6` to `100`, as you recommended, the speed increased to ~1100 kbps.

First, why does increasing the connection interval result in higher speed instead of lower?

Second, why is the speed still not reaching the ~1401 kbps of ble50_throughput?

I enabled 2M PHY by adding the following in the `gatts_on_sync` function of `nimble_bleprph_throughput`:  
`rc = ble_gap_set_prefered_default_le_phy(BLE_HCI_LE_PHY_2M_PREF_MASK, BLE_HCI_LE_PHY_2M_PREF_MASK);`

dukelec, 2025-03-31:
I found the reason for the second question myself.

By changing the WRITE_THROUGHPUT_PAYLOAD from 500 to 495, the data size of each fragment is maximized. The speed can reach ~1404 kbps at conn_itvl=100 (and ~1063 kbps at conn_itvl=6).

There's only this issue left:  
Why does increasing the connection interval result in higher speed instead of lower?

Shreeyash-17, 2025-04-01:
@dukelec In ESP-IDF, Data Length Extension (DLE : 251) allows sending larger packets per connection event. When the connection interval is increased, more packets can be transmitted in each event, making data transfer more efficient.
 
With a higher MTU (Maximum Transmission Unit), a longer connection interval allows more data to be sent at once, reducing the number of small, frequent transmissions. This helps lower communication overhead and makes data transfer more efficient, leading to faster speeds even with a longer interval.
 
Could you kindly share the patch with all the changes you made to achieve this throughput on ESP32-C3?

closed 2025-04-02 by dukelec