espressif/esp-idf #15666: LEDC does not release GPIO reservation, causing "GPIO is not usable" warning on reuse

zhaoleicpp, 2025-03-27:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

ESP-IDF 5.4 PowerShell

### Espressif SoC revision.

P32 s3

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

QEMU

### Power Supply used.

Battery

### What is the expected behavior?

No warning printed, anyway

### What is the actual behavior?

When using the LEDC module, the first call to `ledc_channel_config()` successfully configures a GPIO (e.g., GPIO 5). However, after releasing resources with `ledc_stop()` and `ledc_timer_config(.deconfigure = true)`, attempting to configure the same GPIO again triggers the following warning:

W (xxx) ledc: GPIO 5 is not usable, maybe conflict with others

### Steps to reproduce.

Once run the code, it comes 100%.

#include "driver/ledc.h"
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"

#define PWM_CHANNEL LEDC_CHANNEL_0
#define PWM_FREQ 5000
#define WAVE_OUT_0_PIN GPIO_NUM_5

void setupPWM() {
    ledc_timer_config_t timer = {
        .speed_mode = LEDC_LOW_SPEED_MODE,
        .duty_resolution = LEDC_TIMER_10_BIT,
        .timer_num = LEDC_TIMER_0,
        .freq_hz = PWM_FREQ,
        .clk_cfg = LEDC_AUTO_CLK,
    };
    ledc_timer_config(&timer);

    ledc_channel_config_t channel = {
        .gpio_num = WAVE_OUT_0_PIN,
        .speed_mode = LEDC_LOW_SPEED_MODE,
        .channel = PWM_CHANNEL,
        .timer_sel = LEDC_TIMER_0,
        .duty = 0,
    };
    ledc_channel_config(&channel);
}

void deletePWM() {
    ledc_stop(LEDC_LOW_SPEED_MODE, PWM_CHANNEL, 0);
    ledc_timer_config_t timer = {
        .speed_mode = LEDC_LOW_SPEED_MODE,
        .timer_num = LEDC_TIMER_0,
        .deconfigure = true,
    };
    ledc_timer_config(&timer);
}

void app_main() {
    setupPWM();
    vTaskDelay(5000 / portTICK_PERIOD_MS);
    deletePWM();
    vTaskDelay(5000 / portTICK_PERIOD_MS);
    setupPWM(); // Triggers warning on second call
}

### Debug Logs.

```plain
I (xxx) gpio: GPIO[5] configured for PWM
I (xxx) gpio: PWM stopped
I (xxx) gpio: PWM timer deconfigured
W (xxx) ledc: GPIO 5 is not usable, maybe conflict with others
```

### More Information.

Add a call to esp_gpio_unreserve() in ledc_stop() or ledc_timer_del() to release the GPIO reservation when the LEDC resources are freed.

zhaoleicpp, 2025-03-27:
### Additional Information
After further investigation, I found that the correct function to release GPIO reservations in ESP-IDF v5.4 is `esp_gpio_revoke` (defined in `esp_private/esp_gpio_reserve.h`), not `esp_gpio_unreserve` as I initially assumed. However, searching through `ledc.c`, I confirmed that neither `esp_gpio_revoke` nor `esp_gpio_unreserve` is called when releasing LEDC resources (e.g., in `ledc_stop()` or `ledc_timer_config(.deconfigure = true)`). This means the GPIO reservation set by `esp_gpio_reserve` in `_ledc_set_pin()` persists, causing the "GPIO 5 is not usable" warning on subsequent `ledc_channel_config()` calls. Could you verify if this is an intended design or a bug in the LEDC module?

songruo, 2025-03-28:
We are not likely to add `esp_gpio_revoke` in LEDC driver, but we have provided a workaround, please call `gpio_reset_pin`, it will do the revoke. Referring the same issue https://github.com/espressif/esp-idf/issues/15598#issuecomment-2747081623.

closed 2025-03-28 by songruo

zhaoleicpp, 2025-03-28:
Thanks for the response and workaround suggestion. I understand that calling `gpio_reset_pin` resolves the issue, but from a design perspective, it feels asymmetric since `esp_gpio_reserve` is called in `_ledc_set_pin()` without a corresponding `esp_gpio_revoke` in the LEDC driver. Ideally, the driver should handle resource allocation and release symmetrically (e.g., in `ledc_stop` or `ledc_timer_config(.deconfigure = true)`). Could you reconsider adding `esp_gpio_revoke` to maintain consistency and elegance in the LEDC module? This would align with the "whoever allocates releases" principle and reduce user-side workarounds.

songruo, 2025-03-28:
This is because LEDC driver is an old driver, the driver did not consider the channel release from the beginning. We don't want to take too much efforts to add the channel release logic to the driver, since there won't be too much benefits it can bring. If there will be a new driver for LEDC in the future, we would definitely consider the symmetry.

mickeyl, 2025-10-21:
@macksal Please open a dedicated issue about that, I'm sure Espressif will take care of that eventually.