espressif/esp-idf #15670: release/v5.4 branch does not emit `disconnect` event for NimBLE stack for all remotes

finger563, 2025-03-28:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.4-727-g5cbd2a3877

### Espressif SoC revision.

ESP32-S3 (QFN56) (revision v0.1)

### Operating System used.

macOS

### How did you build your project?

Command line with CMake

### If you are using Windows, please specify command line type.

None

### Development Kit.

QtPy ESP32S3

### Power Supply used.

USB

### What is the expected behavior?

BLE stack emits disconnect event each time a disconnect occurs.

### What is the actual behavior?

The disconnect event is not emitted for certain remotes. For other remotes, it is always emitted.

This works as expected on `v5.4.0` tag, but not on `release/v5.4` branch.

### Steps to reproduce.

1. Pair esp32s3 (peripheral) to BLE host
2. press button (for example) to tell the esp32 to disconnect
3. See that the unsubscribe events are generated, but not the disconnect event

### Debug Logs.

```plain
Failing (USB BLE dongle):

[Hid Service Example/I][266.326]: Button pressed!
I (266548) NimBLE: GAP procedure initiated: terminate connection; conn_handle=1 hci_reason=19

D (266549) NimBLE: ble_hs_hci_cmd_send: ogf=0x01 ocf=0x0006 len=3

D (266549) NimBLE: 0x06
D (266550) NimBLE: 0x04
D (266550) NimBLE: 0x03
D (266550) NimBLE: 0x01
D (266550) NimBLE: 0x00
D (266550) NimBLE: 0x13
D (266550) NimBLE:

D (266568) NimBLE: ble_hs_event_rx_hci_ev; opcode=0x5
D (266568) NimBLE:

D (266569) NimBLEServer: >> handleGapEvent: BLE_GAP_EVENT_SUBSCRIBE
I (266569) NimBLEServer: subscribe event; attr_handle=8, subscribed: false
D (266569) NimBLEServer: << handleGapEvent
D (266569) NimBLEServer: >> handleGapEvent: BLE_GAP_EVENT_SUBSCRIBE
I (266570) NimBLEServer: subscribe event; attr_handle=24, subscribed: false
D (266570) NimBLECharacteristicCallbacks: onSubscribe: default
D (266570) NimBLEServer: << handleGapEvent
D (266570) NimBLEServer: >> handleGapEvent: BLE_GAP_EVENT_SUBSCRIBE
I (266571) NimBLEServer: subscribe event; attr_handle=28, subscribed: false
D (266571) NimBLECharacteristicCallbacks: onSubscribe: default
D (266571) NimBLEServer: << handleGapEvent
D (266571) NimBLEServer: >> handleGapEvent: BLE_GAP_EVENT_SUBSCRIBE
I (266572) NimBLEServer: subscribe event; attr_handle=54, subscribed: false
D (266572) NimBLECharacteristicCallbacks: onSubscribe: default
D (266572) NimBLEServer: << handleGapEvent
[Hid Service Example/I][266.495]: Button released!
I (267136) NimBLE: GATT procedure initiated: notify;
I (267137) NimBLE: att_handle=28

E (267137) NimBLECharacteristic: << sendValue: failed to send value, rc=7 No open connection with handle
I (267138) NimBLE: GATT procedure initiated: notify;
I (267138) NimBLE: att_handle=24

![Image](https://github.com/user-attachments/assets/8bd6f225-2541-4580-80a7-87e27fdaf84b)

Succeeding (phone is BLE remote):

[Hid Service Example/I][16.589]: Button pressed!
I (16810) NimBLE: GAP procedure initiated: terminate connection; conn_handle=1 hci_reason=19

D (16811) NimBLE: ble_hs_hci_cmd_send: ogf=0x01 ocf=0x0006 len=3

D (16811) NimBLE: 0x06
D (16812) NimBLE: 0x04
D (16812) NimBLE: 0x03
D (16812) NimBLE: 0x01
D (16812) NimBLE: 0x00
D (16812) NimBLE: 0x13
D (16812) NimBLE:

D (16830) NimBLE: ble_hs_event_rx_hci_ev; opcode=0x5
D (16830) NimBLE:

D (16831) NimBLEServer: >> handleGapEvent: BLE_GAP_EVENT_SUBSCRIBE
I (16831) NimBLEServer: subscribe event; attr_handle=8, subscribed: false
D (16831) NimBLEServer: << handleGapEvent
D (16832) NimBLEServer: >> handleGapEvent: BLE_GAP_EVENT_SUBSCRIBE
I (16832) NimBLEServer: subscribe event; attr_handle=24, subscribed: false
D (16832) NimBLECharacteristicCallbacks: onSubscribe: default
D (16832) NimBLEServer: << handleGapEvent
D (16832) NimBLEServer: >> handleGapEvent: BLE_GAP_EVENT_SUBSCRIBE
I (16833) NimBLEServer: subscribe event; attr_handle=28, subscribed: false
D (16833) NimBLECharacteristicCallbacks: onSubscribe: default
D (16833) NimBLEServer: << handleGapEvent
D (16833) NimBLEServer: >> handleGapEvent: BLE_GAP_EVENT_SUBSCRIBE
I (16834) NimBLEServer: subscribe event; attr_handle=54, subscribed: false
D (16834) NimBLECharacteristicCallbacks: onSubscribe: default
D (16834) NimBLEServer: << handleGapEvent
D (16834) NimBLEServer: >> handleGapEvent: BLE_GAP_EVENT_DISCONNECT
[Hid Service Example/I][16.614]: Device disconnected: LOCAL_USER_TERMINATED
D (16836) NimBLEAdvertising: >> Advertising start: duration=0, dirAddr=NULL
I (16836) NimBLE: GAP procedure initiated: advertise;
I (16836) NimBLE: disc_mode=2
I (16837) NimBLE:  adv_channel_map=0 own_addr_type=0 adv_filter_policy=0 adv_itvl_min=0 adv_itvl_max=0
I (16837) NimBLE:

![Image](https://github.com/user-attachments/assets/ddfdf95a-5ec6-4573-a649-2db44ea7041d)
```

rahult-github, 2025-03-29:
Hi @finger563 , 

Can you kindly provide one log with below change:  

1. Enable BT_HCI_LOG_DEBUG_EN  from menuconfig.  It is under "(Top) → Component config → Bluetooth"
2. Apply given patch . It need to be applied in $IDF_PATH.
3. Run your test . Once issue is observed, please share the console log. You should observe some hex values on console after this patch is applied. Please share the complete log. 

[debug_print.txt](https://github.com/user-attachments/files/19520981/debug_print.txt)

finger563, 2025-03-29:
@rahult-github 

Thanks, after applying the patch and enabling the logging here are the logs when disconnecting from the device for which the ESP32s3 does not generate the disconnect callback:

[image]

```console
D (47056) NimBLEServer: >> handleGapEvent: BLE_GAP_EVENT_NOTIFY_TX
D (47056) NimBLECharacteristicCallbacks: onStatus: default
D (47057) NimBLEServer: << handleGapEvent
90 H:01 00 08 00 04 00 04 00 1b 36 00 07
91 H:01 00 08 00 04 00 04 00 1b 1c 00 05
92 H:01 00 17 00 13 00 04 00 1b 18 00 c3 34 71 18 71 18 c3 34 3b 03 59 02 09 00 01 a4
93 E:13 05 01 01 00 01 00
94 E:13 05 01 01 00 01 00
95 E:13 05 01 01 00 01 00
[Hid Service Example/I][47.032]: Button pressed!
I (47254) NimBLE: GAP procedure initiated: terminate connection; conn_handle=1 hci_reason=19

D (47255) NimBLE: ble_hs_hci_cmd_send: ogf=0x01 ocf=0x0006 len=3

D (47255) NimBLE: 0x06
D (47255) NimBLE: 0x04
D (47256) NimBLE: 0x03
D (47256) NimBLE: 0x01
D (47256) NimBLE: 0x00
D (47256) NimBLE: 0x13
D (47256) NimBLE:

96 C:06 04 03 01 00 13
97 E:0f 04 00 05 06 04
98 E:05 04 00 01 00 16
D (47265) NimBLE: ble_hs_event_rx_hci_ev; opcode=0x5
D (47265) NimBLE:

D (47265) NimBLEServer: >> handleGapEvent: BLE_GAP_EVENT_SUBSCRIBE
I (47266) NimBLEServer: subscribe event; attr_handle=8, subscribed: false
D (47266) NimBLEServer: << handleGapEvent
D (47266) NimBLEServer: >> handleGapEvent: BLE_GAP_EVENT_SUBSCRIBE
I (47266) NimBLEServer: subscribe event; attr_handle=24, subscribed: false
D (47267) NimBLECharacteristicCallbacks: onSubscribe: default
D (47267) NimBLEServer: << handleGapEvent
D (47267) NimBLEServer: >> handleGapEvent: BLE_GAP_EVENT_SUBSCRIBE
I (47267) NimBLEServer: subscribe event; attr_handle=28, subscribed: false
D (47267) NimBLECharacteristicCallbacks: onSubscribe: default
D (47268) NimBLEServer: << handleGapEvent
D (47268) NimBLEServer: >> handleGapEvent: BLE_GAP_EVENT_SUBSCRIBE
I (47268) NimBLEServer: subscribe event; attr_handle=54, subscribed: false
D (47268) NimBLECharacteristicCallbacks: onSubscribe: default
D (47268) NimBLEServer: << handleGapEvent
[Hid Service Example/I][47.251]: Button released!
I (48044) NimBLE: GATT procedure initiated: notify;
I (48044) NimBLE: att_handle=28

E (48045) NimBLECharacteristic: << sendValue: failed to send value, rc=7 No open connection with handle
I (48046) NimBLE: GATT procedure initiated: notify;
I (48046) NimBLE: att_handle=24
```

rahult-github, 2025-03-31:
Hi @finger563 , 

Thanks for the logs. In log i observe that remote is responding with error for Read remote used features, but also providing features list. This seems wrong. This is causing one state for stack to handling events go wrong and hence the behaviour you observed. 

[image]

 
What is the remote device being used ? ( The central role device ).  Can you change it to some other device and check? 

Additionally i have made one patch with some debug prints and a change to confirm if the above is the reason. can you kindly check if this patch helps and share console log ( if issue is seen or not seen ).  The patch is to be applied in $IDF_PATH/components/bt/host/nimble/nimble path. 

[debug_prints.txt](https://github.com/user-attachments/files/19529995/debug_prints.txt)

finger563, 2025-03-31:
@rahult-github 

I do not believe it is fully an issue with my remote device (a USB BLE 5.0 dongle connected to a Windows 10 desktop) because this remote device worked perfectly fine with v5.4.0 esp-idf and previous versions.

Thank you for that patch - with it, the disconnect callback is properly called. 

I've copied the logs below for reference:

[image]

```console
D (61768) NimBLEServer: >> handleGapEvent: BLE_GAP_EVENT_NOTIFY_TX
D (61768) NimBLECharacteristicCallbacks: onStatus: default
D (61769) NimBLEServer: << handleGapEvent
4a H:01 00 17 00 13 00 04 00 1b 18 00 9b 9a cc 02 cc 02 9b 9a e8 03 d4 00 07 40 00 a4
4b E:13 05 01 01 00 01 00
[Hid Service Example/I][62.010]: Button pressed!
I (62234) NimBLE: GAP procedure initiated: terminate connection; conn_handle=1 hci_reason=19

D (62234) NimBLE: ble_hs_hci_cmd_send: ogf=0x01 ocf=0x0006 len=3

D (62235) NimBLE: 0x06
D (62235) NimBLE: 0x04
D (62235) NimBLE: 0x03
D (62235) NimBLE: 0x01
D (62235) NimBLE: 0x00
D (62235) NimBLE: 0x13
D (62235) NimBLE:

4c C:06 04 03 01 00 13
4d E:0f 04 00 05 06 04
4e E:05 04 00 01 00 16
D (62243) NimBLE: ble_hs_event_rx_hci_ev; opcode=0x5
D (62243) NimBLE:

D (62244) NimBLEServer: >> handleGapEvent: BLE_GAP_EVENT_SUBSCRIBE
I (62244) NimBLEServer: subscribe event; attr_handle=8, subscribed: false
D (62244) NimBLEServer: << handleGapEvent
D (62245) NimBLEServer: >> handleGapEvent: BLE_GAP_EVENT_SUBSCRIBE
I (62245) NimBLEServer: subscribe event; attr_handle=24, subscribed: false
D (62245) NimBLECharacteristicCallbacks: onSubscribe: default
D (62245) NimBLEServer: << handleGapEvent
D (62245) NimBLEServer: >> handleGapEvent: BLE_GAP_EVENT_SUBSCRIBE
I (62246) NimBLEServer: subscribe event; attr_handle=28, subscribed: false
D (62246) NimBLECharacteristicCallbacks: onSubscribe: default
D (62246) NimBLEServer: << handleGapEvent
D (62246) NimBLEServer: >> handleGapEvent: BLE_GAP_EVENT_DISCONNECT
[Hid Service Example/I][62.025]: Device disconnected: LOCAL_USER_TERMINATED
D (62248) NimBLEAdvertising: >> Advertising start: duration=0, dirAddr=NULL
I (62248) NimBLE: GAP procedure initiated: advertise;
I (62248) NimBLE: disc_mode=2
I (62248) NimBLE:  adv_channel_map=0 own_addr_type=0 adv_filter_policy=0 adv_itvl_min=0 adv_itvl_max=0
I (62249) NimBLE:

D (62249) NimBLE: ble_hs_hci_cmd_send: ogf=0x08 ocf=0x0006 len=15

D (62249) NimBLE: 0x06
D (62249) NimBLE: 0x20
D (62250) NimBLE: 0x0f
D (62250) NimBLE: 0x30
D (62250) NimBLE: 0x00
D (62250) NimBLE: 0x60
D (62250) NimBLE: 0x00
D (62250) NimBLE: 0x00
D (62250) NimBLE: 0x00
D (62250) NimBLE: 0x00
D (62250) NimBLE: 0x00
D (62250) NimBLE: 0x00
D (62251) NimBLE: 0x00
D (62251) NimBLE: 0x00
D (62251) NimBLE: 0x00
D (62251) NimBLE: 0x00
D (62251) NimBLE: 0x07
D (62251) NimBLE: 0x00
D (62251) NimBLE:

4f C:06 20 0f 30 00 60 00 00 00 00 00 00 00 00 00 00 07 00
50 E:0e 04 05 06 20 00
D (62253) NimBLE: ble_hs_hci_cmd_send: ogf=0x08 ocf=0x000a len=1

D (62253) NimBLE: 0x0a
D (62253) NimBLE: 0x20
D (62254) NimBLE: 0x01
D (62254) NimBLE: 0x01
D (62254) NimBLE:

51 C:0a 20 01 01
52 E:0e 04 05 0a 20 00
D (62255) NimBLEAdvertising: << Advertising start
D (62255) NimBLEServer: << handleGapEvent
```

finger563, 2025-03-31:
@rahult-github do you think this will come in as a patch for `esp-idf v5.4.1` or what do you think the timeline for an official fix for this will be?

Similarly, should we use this `ev->status == 0x1a` check as the patch until an official implementation or is there a better patch to use for this fix?

rahult-github, 2025-04-01:
Hello @finger563  ,

> I do not believe it is fully an issue with my remote device (a USB BLE 5.0 dongle connected to a Windows 10 desktop) 

As per spec, any value apart from 0x0 is considered to be not success. So this is remote device behaviour .

[image]

> because this remote device worked perfectly fine with v5.4.0 esp-idf and previous versions

That is because the status of this HCI event was not considered previously for decision making. We have started considering the status now and take actions accordingly.  

> do you think this will come in as a patch for esp-idf v5.4.1 

No, Actually this tag got released today 

> should we use this ev->status == 0x1a check as the patch 

It is incorrect to add such conditions when spec clearly mandates 0 as success case whereas other values to be failures.  I had given the change just to confirm it works based on my suspicion from the logs. 

Give me sometime , i am thinking  of making a configurable option to handle this. That way you can toggle its value as per your use case and it should work in your case too without breaking any other functionality.

rahult-github, 2025-04-01:
Hi @finger563   , 

Please check if this patch helps now . This should be a simpler solution that doesn't depend on any status code for internal operations..  Please revert the previous patch, before applying this patch. 

[fix.txt](https://github.com/user-attachments/files/19548233/fix.txt)

finger563, 2025-04-01:
@rahult-github Yes, that patch works as well, thanks!

Do you think this will make it into an upcoming patch release soon, or should we plan to use this patch for now?

rahult-github, 2025-04-02:
Hi @finger563 , thanks for confirming.  It will be part of next official release tag for 5.4 ( in 5.4.2 ) , but will merge the change , so should be available on github .

rahult-github, 2025-04-29:
Change is [merged](https://github.com/espressif/esp-nimble/commit/5a1882e4b97ab29e1a800144f576e6ad05fd4913). Closing issue.