espressif/esp-idf #15672: STA Wifi Error 15 (4 way handshake timeout) whenever connecting to wifi

JensenMcKenzie, 2025-03-29:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.5-dev-2715-ga45d713b03

### Espressif SoC revision.

Chip is ESP32-C5 (revision v0.1)

### Operating System used.

macOS

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-C5-DevKitC-1

### Power Supply used.

USB

### What is the expected behavior?

Device should connect to wifi without issue

### What is the actual behavior?

Device always shows error code 15 or 2 (mostly 15). This issue happens across flashes, across projects, and on the example fast_start project. After I got this error once in my main project, it was never able to successfully connect again. I have rebooted my router, same result. I have triple checked the ssid and password are 100% correct.

### Steps to reproduce.

Not quite sure how it was originally triggered. I was trying to reinitialize STA wifi via this code:

ESP_ERROR_CHECK(esp_netif_init());
    ESP_ERROR_CHECK(esp_event_loop_create_default());

    wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();
    ESP_ERROR_CHECK(esp_wifi_init(&cfg));

    ESP_ERROR_CHECK(esp_event_handler_instance_register(WIFI_EVENT, ESP_EVENT_ANY_ID, &event_handler, NULL, NULL));
    ESP_ERROR_CHECK(esp_event_handler_instance_register(IP_EVENT, IP_EVENT_STA_GOT_IP, &event_handler, NULL, NULL));

    // Initialize default station as network interface instance (esp-netif)
    esp_netif_t *sta_netif = esp_netif_create_default_wifi_sta();
    assert(sta_netif);

    // Initialize and start WiFi
    wifi_config_t wifi_config = {
        .sta = {
            .scan_method = WIFI_FAST_SCAN,
            .sort_method = WIFI_CONNECT_AP_BY_SIGNAL,
            .threshold.rssi = -127,
            .threshold.authmode = WIFI_AUTH_OPEN,
            .threshold.rssi_5g_adjustment = 0,
        },
    };

[... 238 lines cut ...]
I (29665) wifi:(phy)ppe_thresholds_present:1, nominal_packet_padding:3
I (29675) wifi:(phy)dcm tx(constellation:0, nss:0), dcm rx(constellation:0, nss:0)
I (29675) wifi:(phy)rx_mcs_map:0xffaa, tx_mcs_map:0xffaa, stbc_tx:0, bfmer(su:1, mu:1), ldpc:1, max_rate:172
I (29685) wifi:(phy)nsts:4, ru_index_bitmap:0x3(242:1, 484:1, 996:0, 2*996:0)
I (29695) wifi:(phy)threshold_bits:48, nsts_num:4, ru_num:2, he_cap->len:29, ppe_threshold_len:7(6,11,4)
I (29705) wifi:(ppe)RU242, NSTS0, PPE16:0, PPE8:7, nominal_packet_padding:2
W (29715) wifi:(hal)co_hosted_bss:1, max_indicator:3, bitmask:0xf8, mBSSIDsEnable:0
I (29715) wifi:(opr)len:8, TWT Required:0, VHT Opr Info Present:0, 6GHz Opr Info Present:0, Co-Hosted BSS:1(max_indicator:3), Basic MCS and NSS:0xfffc
I (29735) wifi:(opr)len:8, Default PE Duration:4, TXOP RTS Threshold:0(0 us), ER-SU-Disable:1
I (29745) wifi:(opr)len:8, BSS Color:7, disabled:0, Partial BSS Color:0
I (29745) wifi:state: assoc -> run (0x10)
I (29755) wifi:(he)ppe_thresholds_present:1, nominal_packet_padding(rx:3, cfg:2)
I (29755) wifi:(trc)phytype:CBW20-SGI, snr:67, maxRate:172, highestRateIdx:0
W (29765) wifi:(trc)band:2G, phymode:3, highestRateIdx:0, lowestRateIdx:11, dataSchedTableSize:14
I (29775) wifi:(trc)band:2G, rate(S-MCS7, rateIdx:0), ampdu(rate:S-MCS7, schedIdx(0, stop:8)), snr:67, ampduState:wait operational
I (29785) wifi:ifidx:0, rssi:-28, nf:-95, phytype(0x3, CBW20-SGI), phymode(0x7, 11ax), max_rate:172, he:1, vht:0, ht:1
I (29795) wifi:(ht)max.RxAMPDULenExponent:3(65535 bytes), MMSS:5(4 us)
I (31765) wifi:state: run -> init (0xf00)
I (31775) wifi:new:<6,0>, old:<6,0>, ap:<255,255>, sta:<6,0>, prof:1, snd_ch_cfg:0x0
I (31775) scan: event: 15
```

### Diagnostic report archive.

[idf-diag-ab2b7127-f044-4497-9920-d994239f930f.zip](https://github.com/user-attachments/files/19514820/idf-diag-ab2b7127-f044-4497-9920-d994239f930f.zip)

QingzhaoYin, 2025-04-02:
@JensenMcKenzie Hi～，this issue has already been fixed in commit a7d99f92046c3d424ff0f31a6424a6569167800d. Please update your branch.

AxelLin, 2025-04-02:
> [@JensenMcKenzie](https://github.com/JensenMcKenzie) Hi～，this issue has already been fixed in commit a7d99f92046c3d424ff0f31a6424a6569167800d. Please update your branch.

Wrong commit id or not yet sync to github? a7d99f92046c3d424ff0f31a6424a6569167800d does not exist.

JensenMcKenzie, 2025-04-02:
@QingzhaoYin can't find this commit either

QingzhaoYin, 2025-04-02:
@AxelLin @JensenMcKenzie  Sorry, this commit not yet sync to github, we will sync soon

Xiehanxin, 2025-04-02:
hi @AxelLin  @JensenMcKenzie  please refer this commit :https://github.com/espressif/esp-idf/commit/00293d7a40c18c84b5fc8201f0264acb885127d9

Alvin1Zhang, 2025-05-07:
Thanks for reporting, feel free to reopen. Thanks.

closed 2025-05-07 by Alvin1Zhang