espressif/esp-idf #15673: BT_AV : Target device not found

Admsher, 2025-03-29:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

ESP IDF v5.3

### Espressif SoC revision.

ESP32 Chip

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32 One

### Power Supply used.

USB

### What is the expected behavior?

When trying to run this code from this code: 
```
#define HELIX_LOGGING_ACTIVE false
#include "AudioTools.h"
#include "AudioTools/AudioLibs/A2DPStream.h"
#include "AudioTools/AudioLibs/AudioSourceSDFAT.h"
#include "AudioTools/AudioCodecs/CodecMP3Helix.h"

const char *startFilePath="/music/";
const char* ext="mp3";
AudioSourceSDFAT source(startFilePath, ext, 15);
A2DPStream out;
MP3DecoderHelix decoder;
AudioPlayer player(source, out, decoder);

void setup() {
  Serial.begin(115200);
  AudioToolsLogger.begin(Serial, AudioToolsLogLevel::Warning);
  // setup player
  // Setting up SPI if necessary with the right SD pins by calling 
  SPI.begin(14, 12, 13, 15);
  player.setVolume(0.8);
  player.begin();

  // setup output - We send the test signal via A2DP - so we conect to a Bluetooth Speaker
  auto cfg = out.defaultConfig(TX_MODE);
  cfg.silence_on_nodata = true; // prevent disconnect when there is no audio data
  cfg.name = "CLIP 5";  // set the device here. Otherwise the first available device is used for output
  //cfg.auto_reconnect = true;  // if this is use we just quickly connect to the last device ignoring [cfg.name](http://cfg.name/)
  out.begin(cfg);
}

void loop() {
  player.copy();
[... 213 lines cut ...]
# CONFIG_BT_ALLOCATION_FROM_SPIRAM_FIRST is not set
# CONFIG_BT_BLE_DYNAMIC_ENV_MEMORY is not set
CONFIG_BT_SMP_ENABLE=y
CONFIG_BT_SMP_MAX_BONDS=15
CONFIG_BT_MAX_DEVICE_NAME_LEN=32
# CONFIG_BT_BLE_RPA_SUPPORTED is not set
# end of Bluedroid Options

```

also th pr for arduio-esp32 should help : https://github.com/espressif/arduino-esp32/issues/11181 to configure arduino esp-32

### Debug Logs.

```plain
None
```

### Diagnostic report archive.

[diag-9f7a3c32-577e-49ae-a90d-31993d5936dc.zip](https://github.com/user-attachments/files/19516059/diag-9f7a3c32-577e-49ae-a90d-31993d5936dc.zip)

### More Information.

Works on Arduino IDE and connects without an issue

ESP-YTGerd, 2025-05-12:
Hi, @Admsher 
Could you please confirm whether the target device (CLIP5) was in pairing mode when the discovery failed? Additionally, at that time, were other devices—such as a phone—able to scan and connect to CLIP5?

Admsher, 2025-05-12:
Hi, thanks for the reply the CLIP 5 was indeed in pair mode with no other device connected.

Admsher, 2025-05-12:
> Hi, thanks for the reply the CLIP 5 was indeed in pair mode with no other device connected.
To confirm I have successfully connected with phone and also the embedded target when I flash using Arduino IDE