espressif/esp-idf #15675: 35us time gaps in i2c_master_transmit_receive() signal

DushkaF, 2025-03-29:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.4.0

### Espressif SoC revision.

ESP32-WROOM

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

esp32 devkit v1

### Power Supply used.

USB

### What is the expected behavior?

I am working with a robotics platform and I use mpu6050 IMU for navigation. I need to exchange packets as fast as possible. This is what a packet looked like on the previous version of the I2C driver. With the new version, when using the `i2c_master_write_read_device()` function, I get good delays after a read request and after receiving a response. They are about 9 µs.

[image]

### What is the actual behavior?

However, when migrating to the new ESPIDF 5.4.0, I noticed that the transaction time with the sensor has increased significantly. This is due to the delays after the request byte and the receive bytes. They are now somewhere around 30 µs!

[image]

### Steps to reproduce.

I am upgrading from an old driver to a new one, which resulted in the sensor processing cycle frequency dropping, causing the robot to behave inappropriately. I attach the codes to the **Additional Information** block.

### Debug Logs.

```plain

```

### More Information.

<details><summary>Code with new driver implementation</summary>
<p>

``` cpp
// Configure I2C master
i2c_master_bus_config_t i2c_master_config;
i2c_master_config.clk_source = I2C_CLK_SRC_DEFAULT; // Clock source for the bus
i2c_master_config.i2c_port = I2C_MASTER_NUM;        // I2C port number (I2C_NUM_0 or I2C_NUM_1)
i2c_master_config.scl_io_num = I2C_MASTER_SCL_IO;   // GPIO number for SCL line
i2c_master_config.sda_io_num = I2C_MASTER_SDA_IO;   // GPIO number for SDA line
i2c_master_config.flags.enable_internal_pullup = 1; // Use internal GPIO pull-up
i2c_master_config.glitch_ignore_cnt = 7;            // Glitch filter count, default is 7
i2c_master_config.intr_priority = 0;                // Interrupt priority: auto
i2c_master_config.trans_queue_depth = 0;            // Internal queue depth, valid only for async transfers

i2c_master_bus_handle_t bus_handle;
ESP_ERROR_CHECK(i2c_new_master_bus(&i2c_master_config, &bus_handle));
ESP_LOGI(logTAG, "I2C bus is configured");

// ==========================================================
// Configure the slave device
// ==========================================================
// Set parameters for the slave device
i2c_device_config_t i2c_device_config;
i2c_device_config.dev_addr_length = I2C_ADDR_BIT_LEN_7; // Standard 7-bit addressing
i2c_device_config.device_address = 0x68;               // Device address
i2c_device_config.scl_speed_hz = I2C_MASTER_FREQ_HZ;   // I2C clock frequency
i2c_device_config.scl_wait_us = 0;                     // Default wait time
i2c_device_config.flags.disable_ack_check = 0;         // Enable ACK check

i2c_master_dev_handle_t dev_handle;
ESP_ERROR_CHECK(i2c_master_bus_add_device(bus_handle, &i2c_device_config, &dev_handle));
ESP_LOGI(logTAG, "MPU6050 is initialized");

uint8_t bufData[14];

// Wake up MPU-6050
uint8_t bufCmd[2] = {0x6B, 0x00}; // Initialization command
ESP_ERROR_CHECK(i2c_master_transmit(dev_handle, &bufCmd[0], sizeof(bufCmd), -1));
ESP_LOGI(logTAG, "MPU-6050 is configured");
vTaskDelay(pdMS_TO_TICKS(100));

while (1)
{
    // Send command to start measurement
    uint8_t bufCmd[1] = {0x3B}; // Start measurement
    ESP_ERROR_CHECK(i2c_master_transmit_receive(dev_handle, &bufCmd[0], sizeof(bufCmd), &bufData[0], sizeof(bufData), 1000));
    vTaskDelay(pdMS_TO_TICKS(100));
}
```

</p>
</details> 

<details><summary>Code with old driver implementation</summary>
<p>

``` cpp
    i2c_config_t conf = {};
    conf.clk_flags = 0;
    conf.mode = I2C_MODE_MASTER;
    conf.sda_io_num = I2C_MASTER_SDA_IO;
    conf.sda_pullup_en = GPIO_PULLUP_ENABLE;
    conf.scl_io_num = I2C_MASTER_SCL_IO;
    conf.scl_pullup_en = GPIO_PULLUP_ENABLE;
    conf.master.clk_speed = I2C_MASTER_FREQ_HZ;
    ESP_ERROR_CHECK(i2c_param_config(I2C_MASTER_PORT, &conf));
    ESP_ERROR_CHECK(i2c_driver_install(I2C_MASTER_PORT, I2C_MODE_MASTER, 0, 0, 0));

    // Wake up MPU-6050
    uint8_t wake_up_cmd[] = {0x6B, 0x00};
    ESP_ERROR_CHECK(i2c_master_write_to_device(I2C_MASTER_PORT, MPU_ADDR, wake_up_cmd, sizeof(wake_up_cmd), 1000 / portTICK_PERIOD_MS));

    while (1)
    {
        // Read data from MPU-6050
        uint8_t start_reg = 0x3B;
        uint8_t rawData[14];
        ESP_ERROR_CHECK(i2c_master_write_read_device(I2C_MASTER_PORT, MPU_ADDR, &start_reg, 1, rawData, sizeof(rawData), 1000 / portTICK_PERIOD_MS));
        vTaskDelay(pdMS_TO_TICKS(200));
    }
```

</p>
</details>

mythbuster5, 2025-04-01:
@DushkaF That is because esp32 is pretty special compared with other chip. But the cause there is an interrupt inserted between write and read. What you can do is to remove this interrupt. Go i2c_master.c and find function `s_i2c_write_command`, do this patch:

[image]

This might can solve your problem according to my testing. Waiting for your respond, thanks.

DushkaF, 2025-04-02:
@mythbuster5 Thank you! It is really works!

[image]

closed 2025-04-02 by DushkaF