espressif/esp-idf #15682: Need UART_RX_IDLE_THRHD usage

ToJIka4, 2025-03-30:
### Is your feature request related to a problem?

> **UART_RX_IDLE_THRHD** When the receiver takes more time to receive Byte data than what this register indicates, it will produce a frame-end signal. (R/W)

As i understand this flag should help to organize things like Modbus 3.5 character time and helps to see silence on UART for some period of time. 

### Describe the solution you'd like.

I'd like to see any description of behaviour of UART with this flag and see ESP-IDF integration

BitsForPeople, 2025-03-30:
The workings of the threshold registers are described in the TRMs, and "ESP-IDF integration"  is present via e.g. [`uart_set_rx_timeout()`](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/peripherals/uart.html#_CPPv419uart_set_rx_timeout11uart_port_tK7uint8_t).

songruo, 2025-04-02:
> **UART_RX_IDLE_THRHD** When the receiver takes more time to receive Byte data than what this register indicates, it will produce a frame-end signal. (R/W)

Have confirmed that it can produce a frame-end signal, but no interrupt can be triggered from it. Therefore, I think it is not possible if you want to utilize this bit to distinguish where one message ends and another begins for Modbus protocol. You have to try with [uart_set_rx_timeout()](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/peripherals/uart.html#_CPPv419uart_set_rx_timeout11uart_port_tK7uint8_t), this will produce the interrupt, and give out `UART_DATA` event with the `timeout_flag` set.

ToJIka4, 2025-04-02:
@BitsForPeople @songruo thanks a lot for the answers!

closed 2025-04-03 by songruo