espressif/esp-idf #15683: Connected Low-speed device, not supported

alex-k8, 2025-03-30:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.4.0

### Espressif SoC revision.

ESP32-S3

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

Custom Board

### Power Supply used.

USB

### What is the expected behavior?

I expect the enumeration of the device to be correct so I can communicate with the connected keyboard

### What is the actual behavior?

The enumeration does not happen and I get an error 

### Steps to reproduce.

1. Build and upload the HID Host or the USB Host Lib Example code (must have HUB support turned on, as this keyboard has a HUB internally)
2. Connect the Apple A1243 MB110LL/A Wired Aluminum Keyboard
3. Check the logs

### Debug Logs.

```plain
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x2b (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce2820,len:0x15d0
load:0x403c8700,len:0xd20
load:0x403cb700,len:0x2f60
entry 0x403c8924
I (21) boot: ESP-IDF v5.5-dev-1942-g1160a86ba0 2nd stage bootloader
I (22) boot: compile time Feb 24 2025 01:01:29
I (22) boot: Multicore bootloader
I (24) boot: chip revision: v0.2
I (26) boot: efuse block revision: v1.3
I (30) boot.esp32s3: Boot SPI Speed : 80MHz
I (34) boot.esp32s3: SPI Mode       : DIO
I (37) boot.esp32s3: SPI Flash Size : 2MB
I (41) boot: Enabling RNG early entropy source...
I (46) boot: Partition Table:
[... 93 lines cut ...]
[0;32mI (225) heap_init: At 3FCE9710 len 00005724 (21 KiB): RAM[0m
[0;32mI (231) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM[0m
[0;32mI (238) heap_init: At 600FE01C len 00001FCC (7 KiB): RTCRAM[0m
[0;32mI (245) spi_flash: detected chip: generic[0m
[0;32mI (248) spi_flash: flash io: dio[0m
[0;33mW (252) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.[0m
[0;32mI (266) sleep_gpio: Configure to isolate all GPIO pins in sleep state[0m
[0;32mI (273) sleep_gpio: Enable automatic switching of GPIO sleep configuration[0m
[0;32mI (281) main_task: Started on CPU0[0m
[0;32mI (291) main_task: Calling app_main()[0m
[0;32mI (291) example: HID Host example[0m
[0;32mI (291) gpio: GPIO[0]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:2 [0m
[0;32mI (301) gpio: GPIO[13]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 [0m
[0;32mI (341) example: Waiting for HID Device to be connected[0m
[0;33mW (1261) hid-host: No HID device at USB port 1[0m
[0;31mE (1491) HUB: Connected Low-speed device, not supported[0m
[0;31mE (1491) EXT_PORT: [1:2] Port disabled, reset attempts=1[0m
[0;33mW (1491) HUB: Device tree node (parent_port=2): not found[0m
```

### More Information.

I'm using the ESP32-S3 on a custom board with the USB Host capabilities. It generally works with most keyboards, and recently I've been trying to make it work with HUBs as well. It seems to work properly in most devices, but on the Apple Aluminum Wired Keyboard it does not work. The keyboard internally has a HUB, and I assume the actual keyboard is connected to the first slot of the hub. I've tried with other HUB keyboards and they worked.

On the Apple keyboard I get the logs that were shared above `Connected Low-speed device, not supported`. Is it just not supported? Any menuconfig settings I can try? Thank you!

roma-jam, 2025-03-30:
Hi @alex-k8, 

To enable Hub support, you can use the menuconfig settings: `Component config` → `USB-OTG` → `Hub Driver Configuration` → Enable `Support Hubs` 

Support of Low-speed device, connected via external Full-speed Hub in v5.4 still is an experimental feature  as there is still a problem, which is partially fixed in master and other fix will be publish soon.

I linked this issue so it could be closed when the fix will be published.

alex-k8, 2025-03-30:
Thanks @roma-jam , so I will just wait for the fixes to be published, much appreciated!

closed 2025-04-14 by espressif-bot

alex-k8, 2025-04-26:
Hey @roma-jam ! It hasn't been released yet, right? Is it in the master branch? I tried compiling my code using the master branch ESP-IDF, but I'm still getting the same error.

roma-jam, 2025-04-27:
Hi @alex-k8, 

the fix was introduced in here: https://github.com/espressif/esp-idf/commit/dbf9d78763e63f3e7d23df860dd0200613ae6d8e

Could you please share the log? 
If the output is the same: `HUB: Connected Low-speed device, not supported`, please check that you  used the right version during the build (just in case, don't forget to `git pull`). 

Because this exactly line was removed from the master in here: https://github.com/espressif/esp-idf/commit/858947e461a0f81411e65115147342ffeef05dd8 and should no longer present.

roma-jam, 2025-05-06:
Hi @alex-k8, 

any updates?

alex-k8, 2025-05-06:
Hey @roma-jam sorry for the late reply.

I revisited now, I noticed that I can install ESP-IDF in VS Code directly from the master branch, so I tried that to make sure it's up to date. Unfortunately it didn't work in my code because I'm using Arduino ESP32, which apparently only supports up to version 5.4.99 (master is 5.5), but that's probably a whole different issue.

I tried with the example HID example and it works properly, however.

```
W (15634) HCD DWC: Low-speed, extra delay will be applied in ISR
```

I see data correctly, thanks for this.

Any idea when this will be pushed into a stable release, and do you have any insight on the Arduino ESP32 (Arduino as a component)?

gperit, 2026-01-31:
Hi @roma-jam,
thanks for your answer. I just tried this morning with the settings you suggested (and PweOn2PwcGood delay at 500) and absolutely *nothing* changed.

At the moment I'm stuck in this situation:
1) An ASUS and a "no logo" tv remote shaped keyboard + mouse are working without any issue
2) A Samsung and a Logitech keyboard are *not* working, with the same issue

I switched to verbose logging and I got (Logitech keyboard):

I (11995) main.cpp: Waiting for USB keyboard...
D (16245) HUB: Root port reset
D (16245) HUB: Device tree node (uid=1): new
D (16245) ENUM: Start processing device with address 0
E (16265) USBH: Dev 0 EP 0 Error
(16265) ENUM: GET_SHORT_DEV_DESC OK
D (16275) USBH: Processing actions 0xe
D (16275) USBH: EP0 halt flush
D (16285) USBH: Default pipe device 0
V (16285) cache: addr_end: 0x4ff3d180
V (16285) cache: M2C DIR
D (16295) USBH: EP0 clear
E (16295) ENUM: Bad transfer status 1: CHECK_SHORT_DEV_DESC
E (16295) ENUM: CHECK_SHORT_DEV_DESC FAILED
D (16305) ENUM: CANCEL OK
D (16305) HUB: Disabling root port

and the usual log (no verbose enabled) with Samsung keyboard:

W (365106) HCD DWC: Low-speed, extra delay will be applied in ISR
E (365106) USBH: Dev 0 EP 0 Error
E (365106) ENUM: Bad transfer status 1: CHECK_SHORT_DEV_DESC
E (365106) ENUM: CHECK_SHORT_DEV_DESC FAILED
E (365106) EXT_PORT: [1:4] Port disabled, reset attempts=1

Please note that on ESP32-P4 I'm using the 2.0 port (I did not understand how I can use the 1.1 port) since I see not any option in the USB system calls.

I also tried to increase OTG timeouts with no success...

gperit, 2026-02-01:
It seems that this mechanism of enum.c is not working with those keyboards:
ENUM_STAGE_SECOND_RESET,                /**< Reset the device again (Workaround for old USB devices that get confused by the previous short dev desc request). */

gperit, 2026-02-01:
Here I am @roma-jam,
it seems you're right, because this keyboard has a small joystick (but the Lenovo keyboard does not, but I've not it here) and there are sign of your hypotesys after stage 33. It seems there are problems only after the usual "Low-speed, extra delay will be applied in ISR".
 (I've put a *** comment on a custom enum stages log):

I (9208) usb.c: USB driver init...
V (9218) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): checking args
V (9218) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): Args okay. Resulting flags 0x802
D (9228) intr_alloc: Connected src 93 to int 4 (cpu 0)
D (9228) EXT_PORT: Driver installed
D (9238) EXT_HUB: Driver installed
D (9238) HCD DWC: FIFO config lines: RX=512, PTX=128, NPTX=256
D (9278) hid-host: USB HID handling start
D (9278) hid-host: USB HID handling
D (9278) USB_MSC: USB MSC handling start
V (9278) USB_MSC: USB MSC handling
I (9278) usb.c: USB ready
I (9278) main.cpp: Waiting for USB keyboard...

V (9588) cache: addr_end: 0x4ff3c880
V (9588) cache: C2M DIR
D (9588) HUB: Root port reset
V (9588) cache: addr_end: 0x4ff3c880
V (9588) cache: C2M DIR
D (9588) HUB: Device tree node (uid=1): new
D (9588) ENUM: Start processing device with address 0
E (9598) ENUM: Enum stage: 1 ***
E (9598) ENUM: control_request_general(1)
V (9598) cache: addr_end: 0x4ff3cb80
V (9608) cache: C2M DIR
V (9608) cache: addr_end: 0x4ff3f240
V (9608) cache: C2M DIR
V (9608) cache: addr_end: 0x4ff3f240
V (9608) cache: M2C DIR
E (9608) ENUM: set_next_stage(1), last_stage(1)
D (9618) ENUM: GET_SHORT_DEV_DESC OK
E (9628) ENUM: stage_need_process(2)
D (9628) USBH: Processing actions 0x4
D (9628) USBH: Default pipe device 0
V (9638) cache: addr_end: 0x4ff3cb80
V (9638) cache: M2C DIR
E (9638) ENUM: Enum stage: 2 ***
E (9638) ENUM: control_response_handling(2) expected_num_bytes: 16(16), ctrl_xfer_status: 0
E (9648) ENUM: control_xfer_status: 0, stage: 2
V (9658) ENUM: 0x4ff3ca40   80 06 00 01 00 00 08 00  12 01 10 01 09 00 00 08  |................|
E (9668) ENUM: Stage: 2
E (9668) ENUM: set_next_stage(1), last_stage(2)
D (9668) ENUM: CHECK_SHORT_DEV_DESC OK
E (9678) ENUM: stage_need_process(3)
E (9678) ENUM: Enum stage: 3 ***
V (9738) cache: addr_end: 0x4ff3c880
V (9738) cache: C2M DIR
E (9738) ENUM: set_next_stage(1), last_stage(3)
D (9738) ENUM: SECOND_RESET OK
E (9738) ENUM: stage_need_process(4)
E (9738) ENUM: Enum stage: 4 ***
E (9738) ENUM: set_next_stage(1), last_stage(4)
D (9748) ENUM: SECOND_RESET_COMPLETE OK
E (9748) ENUM: stage_need_process(5)
E (9748) ENUM: Enum stage: 5 ***
E (9758) ENUM: control_request_general(5)
V (9758) cache: addr_end: 0x4ff3cb80
V (9758) cache: C2M DIR
V (9768) cache: addr_end: 0x4ff3f640
V (9768) cache: C2M DIR
V (9768) cache: addr_end: 0x4ff3f640
V (9768) cache: M2C DIR
E (9768) ENUM: set_next_stage(1), last_stage(5)
D (9778) ENUM: SET_ADDR OK
E (9778) ENUM: stage_need_process(6)
D (9788) USBH: Processing actions 0x4
[... 866 lines cut ...]
D (15928) EXT_HUB: [1] Processing actions 0x2
V (15928) EXT_HUB: 0x4ff3f440   23 01 01 00 04 00 00 00                           |#.......|
D (15938) EXT_HUB: Stage PORT_FEATURE OK
D (15938) EXT_PORT: [1:4] Processing actions 0x10
V (15948) cache: addr_end: 0x4ff3f580
V (15948) cache: C2M DIR
V (15948) cache: addr_end: 0x4ff3f640
V (15948) cache: C2M DIR
V (15958) cache: addr_end: 0x4ff3f640
V (15958) cache: M2C DIR
D (15968) USBH: Processing actions 0x4
D (15968) USBH: Default pipe device 1
V (15968) cache: addr_end: 0x4ff3f580
V (15968) cache: M2C DIR
D (15978) EXT_HUB: [1] Processing actions 0x2
V (15978) EXT_HUB: 0x4ff3f440   a3 00 00 00 04 00 04 00  01 03 00 00              |............|
D (15988) EXT_HUB: Stage PORT_STATUS_REQUEST OK
D (15998) EXT_PORT: [1:4] Processing actions 0x1
D (15998) EXT_PORT: [1:4] change=0x0000, status=0x0301, state=3
D (16008) EXT_PORT: [1:4] Port is idle
D (16008) EXT_HUB: [1] Device ports handle complete
D (16008) EXT_HUB: [1] Enable IN EP
V (16018) cache: addr_end: 0x4ff25100
V (16018) cache: C2M DIR
and so on...