espressif/esp-idf #15685: I2C crashes with the new way of doing things on the master branch

DaveKnell, 2025-03-30:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.5-dev-2715-ga45d713b03

### Espressif SoC revision.

ESP-S3

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

Custom board

### Power Supply used.

External 3.3V

### What is the expected behavior?

I've been working with the "new" I2C API.  I've the following routine to read data from a slave:

`esp_err_t` i2c_slave_read (uint8_t addr, const int reg, uint8_t *data, uint32_t len)
{
    uint8_t addr_read = (addr << 1) | I2C_MASTER_READ;
    uint8_t addr_write = (addr << 1) | I2C_MASTER_WRITE;
    esp_err_t err = ESP_OK;
    i2c_operation_job_t i2c_ops[8] = { 0 };
    int i2c_op_count = 0;
    if (reg >= 0) {
        ESP_LOGI(TAG, "i2s_slave_read 0x%02x:0x%02x", addr, reg);
        uint8_t cmd[2] = { 0 };
        cmd[0] = addr_write;
        cmd[1] = reg & 0xFF;
        i2c_ops[i2c_op_count++] = (i2c_operation_job_t) {
            .command = I2C_MASTER_CMD_START
        };
        i2c_ops[i2c_op_count++] = (i2c_operation_job_t) {
            .command = I2C_MASTER_CMD_WRITE,
            .write = {
                .ack_check = true,
                .data = cmd,
                .total_bytes = 2
            }
        };
    } 
    if (data) {
        ESP_LOGI(TAG, "Reading %lu bytes from 0x%02x", len, addr);
        i2c_ops[i2c_op_count++] = (i2c_operation_job_t) {
            .command = I2C_MASTER_CMD_START
        };
        i2c_ops[i2c_op_count++] = (i2c_operation_job_t) {
            .command = I2C_MASTER_CMD_WRITE,
            .write = {
                .ack_check = true,
                .data = (uint8_t *)&addr_read,
                .total_bytes = 1
            }
        };
        if (len > 1) {
            ESP_LOGE(TAG, "Reading %lu bytes into 0x%08lx", len, (int32_t) data);
            i2c_ops[i2c_op_count++] = (i2c_operation_job_t) {
                .command = I2C_MASTER_CMD_READ,
                .read = {
                    .ack_value = I2C_ACK_VAL,
                    .data = data,
                    .total_bytes = len - 1
                }
            };
        } 
        ESP_LOGE(TAG, "2nd part reading %lu bytes into 0x%08lx", 1, (uint32_t) (data + (len - 1)));
        i2c_ops[i2c_op_count++] = (i2c_operation_job_t) {
            .command = I2C_MASTER_CMD_READ,
            .read = {
                .ack_value = I2C_NACK_VAL,
                .data = data + (len - 1),
                .total_bytes = 1
            }
        };
    }
    i2c_ops[i2c_op_count++] = (i2c_operation_job_t) {
        .command = I2C_MASTER_CMD_STOP
    };  
    err = i2c_master_execute_defined_operations(state.dev, i2c_ops, i2c_op_count, 100);
    if (err != ESP_OK) {
        ESP_LOGI(TAG, "Error: %d from i2c on reading from 0x%02x:0x%02x", err, addr, reg);
    }
    return err;
}`

which causes a hard crash (see below) when reading one byte, but works fine when reading more than one.

Taking out the "if (len > 1)" test so that, in the case where it's being asked to read one byte, it still has two reads with the one with ack_value as I2C_ACK_VALUE having total_bytes of 0 makes it work perfectly.

### What is the actual behavior?

Without the seemingly redundant read, we see a hard crash:

I (6505) AQM: Sensor ZSCS2016 found
I (6505) AQM: i2s_slave_read 0x42:0x80
I (6505) AQM: Reading 4 bytes from 0x42
E (6505) AQM: Reading 4 bytes into 0x3fca2654
E (6515) AQM: 2nd part reading 1 bytes into 0x3fca2657
I (6515) AQM: i2s_slave_read 0x42:0x82
I (6525) AQM: Reading 1 bytes from 0x42
E (6525) AQM: 2nd part reading 1 bytes into 0x3fcad12e
Guru Meditation Error: Core  0 panic'ed (StoreProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x40379290  PS      : 0x00060033  A0      : 0x8037af3d  A1      : 0x3fc9e1c0
--- 0x40379290: i2c_ll_read_rxfifo at C:/Users/david/esp/master/esp-idf/components/hal/esp32s3/include/hal/i2c_ll.h:703
---  (inlined by) i2c_isr_receive_handler at C:/Users/david/esp/master/esp-idf/components/esp_driver_i2c/i2c_master.c:676

A2      : 0x3fcbeafc  A3      : 0x3fcbeb0c  A4      : 0x3fc9e1fc  A5      : 0x00000000
A6      : 0x3fcbeb60  A7      : 0x3fcbefc8  A8      : 0x00000000  A9      : 0x00000000
A10     : 0x00000024  A11     : 0x00000000  A12     : 0x60013000  A13     : 0x3fcad12e
A14     : 0x60013000  A15     : 0x0000cdcd  SAR     : 0x00000000  EXCCAUSE: 0x0000001d
EXCVADDR: 0x00000000  LBEG    : 0x40056f5c  LEND    : 0x40056f72  LCOUNT  : 0xffffffff
--- 0x40056f5c: memcpy in ROM
--- 0x40056f72: memcpy in ROM

Backtrace: 0x4037928d:0x3fc9e1c0 0x4037af3a:0x3fc9e1f0 0x40376921:0x3fc9e230 0x403778b1:0x3fc9e250 0x4037b22f:0x3fcada00 0x42004c1e:0x3fcada20 0x4037f165:0x3fcada40 0x4037df09:0x3fcada60
--- 0x4037928d: i2c_ll_read_rxfifo at C:/Users/david/esp/master/esp-idf/components/hal/esp32s3/include/hal/i2c_ll.h:703
---  (inlined by) i2c_isr_receive_handler at C:/Users/david/esp/master/esp-idf/components/esp_driver_i2c/i2c_master.c:676
--- 0x4037af3a: i2c_master_isr_handler_default at C:/Users/david/esp/master/esp-idf/components/esp_driver_i2c/i2c_master.c:714
--- 0x40376921: shared_intr_isr at C:/Users/david/esp/master/esp-idf/components/esp_hw_support/intr_alloc.c:471
--- 0x403778b1: _xt_lowint1 at C:/Users/david/esp/master/esp-idf/components/xtensa/xtensa_vectors.S:1240
--- 0x4037b22f: xt_utils_wait_for_intr at C:/Users/david/esp/master/esp-idf/components/xtensa/include/xt_utils.h:82
---  (inlined by) esp_cpu_wait_for_intr at C:/Users/david/esp/master/esp-idf/components/esp_hw_support/cpu.c:55
--- 0x42004c1e: esp_vApplicationIdleHook at C:/Users/david/esp/master/esp-idf/components/esp_system/freertos_hooks.c:58
--- 0x4037f165: prvIdleTask at C:/Users/david/esp/master/esp-idf/components/freertos/FreeRTOS-Kernel/tasks.c:4362 (discriminator 1)
--- 0x4037df09: vPortTaskWrapper at C:/Users/david/esp/master/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:139

ELF file SHA256: 819305bd9

### Steps to reproduce.

The other bits of code needed are:
`void i2c_free(void)
{
    if (state.dev) {
        i2c_master_bus_rm_device(state.dev);
        state.dev = NULL;
    }
    if (state.bus) {
        i2c_del_master_bus(state.bus);
        state.bus = NULL;
    }
}

esp_err_t i2c_begin(int sda, int scl, int speed)
{
    i2c_free();
    i2c_master_bus_config_t i2c_mst_config = {
        .clk_source = I2C_CLK_SRC_DEFAULT,
        .i2c_port = I2C_NUM_0,
        .scl_io_num = scl,
        .sda_io_num = sda,
        .glitch_ignore_cnt = 7,
        .flags.enable_internal_pullup = true,
    };

    ESP_ERROR_CHECK(i2c_new_master_bus(&i2c_mst_config, &(state.bus)));
    
    i2c_device_config_t dev_cfg = {
        .dev_addr_length = I2C_ADDR_BIT_LEN_7,
        .device_address = I2C_DEVICE_ADDRESS_NOT_USED,
        .scl_speed_hz = speed,
    };
    
    ESP_ERROR_CHECK(i2c_master_bus_add_device(state.bus, &dev_cfg, &(state.dev)));
    return ESP_OK;
}

esp_err_t i2c_end()
{
    i2c_free();
    return ESP_OK;
}

int i2c_readu8(int addr, int reg)
{
    uint8_t data[2];
    int res = i2c_slave_read(addr, reg, data, 1);
    if (res == ESP_OK) {
        return data[0];
    } else {
        return -1;
    }
}`

Calling:
i2c_begin(SDA, SCL, speed);
i2c_readu8(I2C_DEV_ADDR, REGISTER);

is sufficient to cause the bug to occur.

### Debug Logs.

```plain

```

mythbuster5, 2025-03-31:
Already fixed several days ago, please rebase.

mythbuster5, 2025-04-07:
Solved by https://github.com/espressif/esp-idf/commit/f8ecc4885d720c3a36b25097774691ca6d0e46b3

closed 2025-04-07 by mythbuster5

DaveKnell, 2025-04-07:
Pulled latest master, can confirm that this now works as expected.  Thank you,