espressif/esp-idf #15687: Commit v5.4 to v5.5 with renamed wdt functions causes build errors in Micropython

electronicsguy, 2025-03-31:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.5

### Operating System used.

Linux

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

Micropython port esp32 should build correctly

### What is the actual behavior?

build fails with the linker error 'undefined reference' (see log below)

Ref. to this commit: https://github.com/espressif/esp-idf/commit/b562afa08e8abafdf52151f8a37794de741257b0

the function ```esp_panic_handler_reconfigure_wdts()``` has been renamed to ```esp_panic_handler_feed_wdts```. 

Could we keep the original function name for compatibility reasons (with a deprecation notice) ?

### Steps to reproduce.

1. Setup esp-idf v5.5
2. clone micropython from: https://github.com/micropython/micropython
3. 
```
cd <MICROPYTHON_GIT_ROOT>/ports/esp32
get_idf
make submodules
make
```

### Build or installation Logs.

```plain
/home/electronicsguy/.espressif/tools/xtensa-esp-elf/esp-14.2.0_20241119/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/14.2.0/../../../../xtensa-esp-elf/bin/ld: esp-idf/main/libmain.a(panichandler.c.obj):(.literal.__wrap_esp_panic_handler+0x4): undefined reference to `esp_panic_handler_reconfigure_wdts'
/home/electronicsguy/.espressif/tools/xtensa-esp-elf/esp-14.2.0_20241119/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/14.2.0/../../../../xtensa-esp-elf/bin/ld: esp-idf/main/libmain.a(panichandler.c.obj): in function `__wrap_esp_panic_handler':
/home/electronicsguy/programming/micropython/micropython/ports/esp32/panichandler.c:50:(.text.__wrap_esp_panic_handler+0x6): undefined reference to `esp_panic_handler_reconfigure_wdts'
```

### Diagnostic report archive.

[diag.log](https://github.com/user-attachments/files/19528558/diag.log)

ESP-Marius, 2025-03-31:
`esp_panic_handler_reconfigure_wdts` was never intended to be a public API, so it was changed without concern for compatibility.

We could add a wrapper for the old function name, but since the behavior is different I'm leaning towards that it would be better to just fix this in micropython. What do you think?

electronicsguy, 2025-03-31:
Understandable. A wrapper would be a good idea, with some kind of deprecation notice. 

You may not have intended to make it public, but once any API is present, 3rd parties may use it liberally 😃. Generally, any 3rd party library will be 1 or 2 generations behind, in supporting your latest upstream release. I will raise a PR on the Micropython side as well, but IMO, it'd not be desirable if a popular 3rd party library completely fails to build between esp-idf dot-version changes.

cnadler86, 2025-04-05:
> `esp_panic_handler_reconfigure_wdts` was never intended to be a public API, so it was changed without concern for compatibility.
>
> We could add a wrapper for the old function name, but since the [...]

I would also vote for the  wrapper around the old function name, in order to make it backwards compatible (wit the deprecation notice)