espressif/esp-idf #15688: Build linux target meet Failed to resolve component problem.

joyhope, 2025-03-31:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

 v5.4.1

### Operating System used.

Linux

### How did you build your project?

Command line with CMake

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

config is ok.

### What is the actual behavior?

-- Configuring incomplete, errors occurred!
HINT: The component 'driver' could not be found. This could be because: component name was misspelled, the component was not added to the build, the component has been moved to the IDF component manager, the component has been removed and refactored into some other component or the component may not be supported by the selected target.

### Steps to reproduce.

esp-idf/examples/system/console/basic$ idf.py --preview set-target linux

...
-- Configuring incomplete, errors occurred!
HINT: The component 'driver' could not be found. This could be because: component name was misspelled, the component was not added to the build, the component has been moved to the IDF component manager, the component has been removed and refactored into some other component or the component may not be supported by the selected target.
Please look out for the component in 'https://components.espressif.com' and add using 'idf.py add-dependency' command.
Refer to the migration guide for more details about moved components.
Refer to the build-system guide for more details about how components are found and included in the build.

It seems tools/cmake script adjusted impact this.  it may be  caused by  linux target returns "empty", but in v5.3.1, this is accept, but recent changed may deny the module.  I think for the easy use simulator it should be supported, without use changed too much dependend module. 

### Build or installation Logs.

```plain

```

fhrbata, 2025-03-31:
Hello @joyhope , thank you for your report. You are right that the `console/basic` example cmake configuration, compared to `v5.3.1`, is not passing in `v5.4.1`. This is related to [fix: properly resolve component dependencies](https://github.com/espressif/esp-idf/commit/fbe8bf89ee4431c72631bc227bf7008cb3aa024e) which was introduced in `v5.4`.

I may be missing something, but it seems that the `console/basic` is not supported on the Linux target. I don't see it listed here https://github.com/espressif/esp-idf/tree/master/examples/system/console/basic. Also trying it build with `v5.3.1`  results in error, because `esp_wifi` is not available and `esp_wifi.h` cannot be included in `cmd_wifi.c` from the `cmd_wifi` component,which requires the `esp_wifi` component. 

So it looks that the error about the missing `esp_wifi` component is caught early, during the configuration phase in `5.4`, compared to failed compilation in `5.3`, which I believe is correct.

Thank you

joyhope, 2025-04-01:
>>I don't see it listed here https://github.com/espressif/esp-idf/tree/master/examples/system/console/basic.

The fact is the console/basic/ are supported by Linux in v5.3.1, but all the later versions do not work.  (I reported issues: https://github.com/espressif/esp-idf/issues/15056#issuecomment-2553028823 )  

Linux target for v5.3.1 is working, but still needs to fix some small compile problems.  I send my fixed project via email.  but it seems not updated.  The later version has more problems, even the config is failed.

I hope,  first solve the config and compile problem. The function problem may be fixed by the user later. 

Currently too many problems for Linux target. I think Linux target is an important feature for idf-esp, it is easy for developer to run and quick develop logic business in PC platform.

joyhope, 2025-04-01:
If you wanted, I could send a workable console/basic project on v5.3.1 idf-esp.  I hope the idf-esp can keep the linux target working, it is just a small effects, but help great for developers.

fhrbata, 2025-04-01:
Hello @joyhope ,

> The fact is the console/basic/ are supported by Linux in v5.3.1, but all the later versions do not work. (I reported issues: #15056 (comment) )

> Linux target for v5.3.1 is working, but still needs to fix some small compile problems. I send my fixed project via email. but it seems not updated. The later version has more problems, even the config is failed.

I believe that even if you can use the `console/basic/` example with some adjustments on Linux, it doesn't imply that it's officially supported. My current understanding is that it is not supported on Linux target.

> Currently too many problems for Linux target. I think Linux target is an important feature for idf-esp, it is easy for developer to run and quick develop logic business in PC platform.

I agree and understand that the unsuccessful CMake configuration reported here for your modified example is unhelpful and adds to your problems. However, I believe this is how it should function, rather than failing during compilation. I also believe there is ongoing work to improve the usability of the Linux target.

For reference, I attempted quickly and possibly incorrectly  to build the example on master branch.
```diff
diff --git a/examples/system/console/advanced/components/cmd_nvs/cmd_nvs.c b/examples/system/console/advanced/components/cmd_nvs/cmd_nvs.c
index 5451008b18d6..815f83e0fd2c 100644
--- a/examples/system/console/advanced/components/cmd_nvs/cmd_nvs.c
+++ b/examples/system/console/advanced/components/cmd_nvs/cmd_nvs.c
@@ -302,12 +302,12 @@ static esp_err_t get_value_from_nvs(const char *key, const char *str_type)
     } else if (type == NVS_TYPE_I64) {
         int64_t value;
         if ((err = nvs_get_i64(nvs, key, &value)) == ESP_OK) {
-            printf("%lld\n", value);
+            printf("%"PRId64"\n", value);
         }
     } else if (type == NVS_TYPE_U64) {
         uint64_t value;
         if ( (err = nvs_get_u64(nvs, key, &value)) == ESP_OK) {
-            printf("%llu\n", value);
+            printf("%"PRIu64"\n", value);
         }
     } else if (type == NVS_TYPE_STR) {
         size_t len;
diff --git a/examples/system/console/advanced/components/cmd_system/cmd_system_common.c b/examples/system/console/advanced/components/cmd_system/cmd_system_common.c
index ca1fb17a728e..205a1de04bdf 100644
--- a/examples/system/console/advanced/components/cmd_system/cmd_system_common.c
+++ b/examples/system/console/advanced/components/cmd_system/cmd_system_common.c
@@ -25,6 +25,8 @@
 #include "freertos/task.h"
 #include "cmd_system.h"
 #include "sdkconfig.h"
+#include "esp_idf_version.h"
+#include "esp_system.h"
 
 #ifdef CONFIG_FREERTOS_USE_STATS_FORMATTING_FUNCTIONS
 #define WITH_TASKS_INFO 1
diff --git a/examples/system/console/advanced/components/cmd_wifi/CMakeLists.txt b/examples/system/console/advanced/components/cmd_wifi/CMakeLists.txt
index fd42b546b6a8..277d0ebecea6 100644
--- a/examples/system/console/advanced/components/cmd_wifi/CMakeLists.txt
+++ b/examples/system/console/advanced/components/cmd_wifi/CMakeLists.txt
@@ -1,9 +1,17 @@
+idf_build_get_property(target IDF_TARGET)
+
 if(CONFIG_ESP_WIFI_ENABLED OR CONFIG_ESP_HOST_WIFI_ENABLED)
     set(srcs "cmd_wifi.c")
 else()
     set(srcs)
 endif()
 
+if(${target} STREQUAL "linux")
+    set(reqs console)
+else()
+    set(reqs console esp_wifi)
+endif()
+
 idf_component_register(SRCS "${srcs}"
                     INCLUDE_DIRS .
-                    REQUIRES console esp_wifi)
+                    REQUIRES ${reqs})
diff --git a/examples/system/console/basic/main/console_example_main.c b/examples/system/console/basic/main/console_example_main.c
index 3ab666bff501..c8abe3d2fdf6 100644
--- a/examples/system/console/basic/main/console_example_main.c
+++ b/examples/system/console/basic/main/console_example_main.c
@@ -12,12 +12,16 @@
 #include "esp_system.h"
 #include "esp_log.h"
 #include "esp_console.h"
+#if CONFIG_CONSOLE_STORE_HISTORY
 #include "esp_vfs_dev.h"
 #include "esp_vfs_fat.h"
+#endif
 #include "nvs.h"
 #include "nvs_flash.h"
 #include "cmd_system.h"
+#if CONFIG_ESP_WIFI_ENABLED
 #include "cmd_wifi.h"
+#endif
 #include "cmd_nvs.h"
 
 /*

```
With `CONFIG_CONSOLE_STORE_HISTORY` and esp_wifi disabled (since vfs and wifi components aren't supported on Linux), I'm able to build the example. However, I noticed that the `read` syscall in linenoise (`linenoiseEdit`) is constantly interrupted (EINTR), as mentioned by @ESP-Marius in his [comment](https://github.com/espressif/esp-idf/issues/15056#issuecomment-2553028823).

Regarding the issue reported here, you're correct. The change I mentioned earlier has caused the cmake configuration to fail if a required component isn't registered, such as when it doesn't support the specified target, as is the case here. The reason the configuration works in version 5.3 is due to a bug in the build system that has since been fixed. You should be able to bypass this by conditionally set the `esp_wifi` dependency in the `cmd_wifi` component as in the patch above, but as you already pointed out there will be other problems.

I think unless `console/basic` is truly supported on the Linux target, you might encounter various issues while attempting to get it to work.

Thank you

closed 2025-04-07 by dobairoland