espressif/esp-idf #15690: esp_https_ota for download firmware for other chip esp_ota_verify_chip_id workaround

MacWyznawca, 2025-03-31:
### Is your feature request related to a problem?

I use `esp_https_ota` to download the (encrypted) software to the Base from the ESP32-S3. I then send it via ESP-NOW to devices with ESP32-C3. Unfortunately, I currently have to modify the chip function `esp_ota_verify_chip_id` for the download to the Base to succeed.

Translated with www.DeepL.com/Translator (free version)

### Describe the solution you'd like.

Is there an elegant way around the `esp_ota_verify_chip_id` verification? If not, can you please add a setting so that verification can be bypassed, or a list of supported processors can be provided?

mahavirj, 2025-04-04:
@MacWyznawca 

Chip ID verification happens at 2 places, once during the HTTPS OTA init phase (in `esp_ota_verify_chip_id`) and then again during the image verification step (`esp_image_verify) at the end of OTA process. In addition, during image verification there are some more checks related to chip revision, efuse block revision etc. I am not sure if we could make those specific to list of targets. 

Could you please share your patch, what all checks you had to skip?

Ideally, this should be treated as a custom partition OTA update and not an application update.

MacWyznawca, 2025-04-04:
Hi @mahavirj!
Yes, but the second check is when app set new partition as working.
I download do Base (bridge), and send to other devices with different chips.
Only upgrade for Base or in devices after sent via ESP-NOW is checked by `esp_image_verify` in my system.

You use a similar method in your esp-now solution, but there you use all devices with the same chip by default, so there are no conflicts.

```
static esp_err_t esp_ota_verify_chip_id(const void *arg)
{
    esp_image_header_t *data = (esp_image_header_t *)(arg);
    esp_https_ota_dispatch_event(ESP_HTTPS_OTA_VERIFY_CHIP_ID, (void *)(&data->chip_id), sizeof(esp_chip_id_t));

    if (data->chip_id != CONFIG_IDF_FIRMWARE_CHIP_ID) {
        ESP_LOGE(TAG, "Mismatch chip id, expected %d, found %d", CONFIG_IDF_FIRMWARE_CHIP_ID, data->chip_id);
       // return ESP_ERR_INVALID_VERSION;
    }
    return ESP_OK;
}
```

In my case, it was enough to comment out the error return in esp_ota_verify_chip_id. And everything works fine for months also “in production”. However, this is not an elegant solution 😉

AdityaHPatwardhan, 2025-04-08:
Hi @MacWyznawca,

I assume the use case is as follows `Cloud Endpoint -> ESP32-S3 based bridge -> end ESP32-C3 device`.
I assume the `esp32s3` which is acting as a bridge doesn't use any of the data and it just acts as a medium to host an image for the `esp32c3` based device.

As asked already by @mahavirj 
Can you let me know what is your exact sequence of APIs that you execute of esp32-s3?
I think if you perform OTA with an esp32-c3 image on esp32s3 then it would fail when you call esp_ota_end().
Do you observe that? or do you abort the OTA before that since the esp32-s3 only wants to download image and not actually perform the OTA.

The `esp_https_ota` layer is to be used when the device itself wants to perform the OTA hence that layer would expect that the chip id present in the downloaded OTA image matches with the chip id in the eFuse.

May I ask what is the reason to use the esp_https_ota layer for downloading the image on esp32-s3?

1) I guess you could just download the image through `esp_http_client` APIs on the esp32-s3 first and then the esp32-c3 will download this image through the esp_https_ota layer. Note that the image will still be in encrypted format at this point

2) If you want to download the unencrypted image on the esp32s3 then the ideal way would be  to download encrypted image on esp32s3 and then perform the in-place decryption.

MacWyznawca, 2025-04-08:
> I assume the use case is as follows `Cloud Endpoint -> ESP32-S3 based bridge -> end ESP32-C3 device`. I assume the `esp32s3` which is acting as a bridge doesn't use any of the data and it just acts as a medium to host an image for the `esp32c3` based device.

@AdityaHPatwardhan Yes, s3 only download, decrypt and store in “free to use” app partition, and send from this partition firmware data to c3 devices. All process is similar to [esp-now OTA](https://github.com/espressif/esp-now/blob/master/examples/ota/main/app_main.c#L191), but I decrypt firmware in s3 (base) device. All firmwares are encrypted with `esp_encrypted_img`.

> As asked already by [@mahavirj](https://github.com/mahavirj) Can you let me know what is your exact sequence of APIs that you execute of esp32-s3? I think if you perform OTA with an esp32-c3 image [...]

@mahavirj 

I use `esp_encrypted_img` and for this reason I download software to s3 not as ordinary data via http(s), but using libraries that decrypt the software.

`esp_ota_end() `in this case return `ESP_ERR_OTA_VALIDATE_FAILED`, but if 'esp_encrypted_img_decrypt_end()' return ESP_OK, and 'esp_ota_get_partition_description()'  give correct parameters (saved in compilation device model, firmware version), I use this data to send do c3 devices.

> The `esp_https_ota` layer is to be used when the device itself wants to perform the OTA hence that layer would expect that the chip id present in the downloaded OTA image matches with the chip id in [...]

For easy and save decryption with 'esp_encrypted_img' API.

> 1. I guess you could just download the image through `esp_http_client` APIs on the esp32-s3 first and then the esp32-c3 will download this image through the esp_https_ota layer. Note that the image [...]

In c3 devices, I check firmware sign (all firmware are signed). Transfer from s3 to c3 via ESP-NOW is fully encrypted, by our method.

AdityaHPatwardhan, 2025-04-08:
I understand your use case now. Thanks for the detailed explanation.

In your case could you use the esp_http_client layer instead of the `esp_https_ota`?,
In your case, if you use the `esp_http_client` layer to download the image and then use `esp_encrypted_img` APIs to decrypt the content, the same result can be obtained.

Since the chip id check is essential for the OTA to proceed, it shall always fail in the `esp_https_ota` layer.

MacWyznawca, 2025-04-08:
But @AdityaHPatwardhan I asked only of 'esp_ota_verify_chip_id()' used in 'ESP_HTTPS_OTA_BEGIN'.

With 'esp_ota_end()' I don't have problem. I described above how I handle 'ESP_ERR_OTA_VALIDATE_FAILED'

AdityaHPatwardhan, 2025-04-08:
@MacWyznawca Yes, I agree, but the same logic applies here.
The `esp_https_ota` layer is to be used when you want to perform the OTA for a device. Technically, when a device attempts for OTA then it is considered that the OTA image is for the same device. If we introduce a workaround here only for the `BEGIN` part then it would introduce an assymetric behaviour for the OTA process, since it would anyways fail at the `END` part. It will be difficult for us to introduce such a workaround which shall make the `OTA` process as a whole to be inconsistent.

In this particular case, the use of `esp_https_ota` layer is not being done for an actual OTA update but just to download an image, which is not the purpose of `esp_https_ota`. As mentioned earlier, more appropriate method here would be to use `esp_http_client` to download the image and then use `esp_encrypted_img` APIs to decrypt the data and write it to respective partition with help of the `esp_partition_write` API.

MacWyznawca, 2025-04-08:
OK @AdityaHPatwardhan I don't change now my library, because working on production in many systems. However, I will consider a change in future systems. For now, I will manage as I have so far.

closed 2025-04-08 by MacWyznawca