espressif/esp-idf #15694: esp32 IP101 eth first connect to route cannot be ping

wensenl, 2025-03-31:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

esp-idf-v4.4.8

### Espressif SoC revision.

esp32

### Operating System used.

macOS

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

esp32

### Power Supply used.

USB

### What is the expected behavior?

when device power, esp32 IP101 can be ping 

### What is the actual behavior?

can not be ping.

### Steps to reproduce.

1. use IP101 eth0
2. ping esp32 device

### Debug Logs.

```plain

```

### More Information.

wensenl, 2025-03-31:
debug log

[03301137异常esp32.log](https://github.com/user-attachments/files/19537829/03301137.esp32.log)

kostaond, 2025-04-03:
@wensenl do you have DHCP server properly configured? Try to follow the following steps:
1) Properly configure DHCP server on your PC and test with other device.
2) Connect ESP device directly to the PC.
3) Capture traffic using Wireshark or similar tool.
4) Check if the ESP device sends DHCP request and if the PC responses with DHCP offer.

One last note, unfortunately ESP-IDF v4.4 is no longer supported... Please check [Release Support Schedule](https://github.com/espressif/esp-idf?tab=readme-ov-file#esp-idf-release-support-schedule) and consider to upgrade to newer version.

kostaond, 2025-04-15:
@wensenl have you been able to identify and resolve the issue?