espressif/esp-idf #15696: ESP32-C6 os_mbuf_dup() fault/crash

kenh-sq, 2025-03-31:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Using release/v5.4 branch.

I'm currently trying to debug an issue where a call to `os_mbuf_dup()` while handling a GATT server write is faulting on occasion (we duplicate the mbuf chain to pass it on to a different task for processing). Typically, it is a `Store access fault` stopping in ROM (MEPC = 0x4002145E). I know all of the msys/mbuf functions are stored in the ROM for the ESP32-C6.

If I do a similar, but not quite, sequence of `os_msys_get_pkthdr()` and `os_mbuf_appendfrom()` instead of the `os_mbuf_dup()` I don't get a crash. The reason we had/have used `os_mbuf_dup()` is that we had existing code developed on ESP32-C3 that uses this and has never shown this issue.

This is occurring somewhat randomly, but always when the mbuf chain is at least a chain length of 2. Below is a log excerpt of the chain-of-2 mbuf fields prior to the crash:
```
W (101360) spBleGattServiceWriteToQueue: os_mbuf_len( ctxt->om ): 454
W (101360) spBleGattServiceWriteToQueue: 0: om:                   0x4084A078
W (101360) spBleGattServiceWriteToQueue: 0: ->om_data:            0x4084A0B7
W (101361) spBleGattServiceWriteToQueue: 0: ->om_flags:           0x00 
W (101361) spBleGattServiceWriteToQueue: 0: ->om_pkthdr_len:      44
W (101361) spBleGattServiceWriteToQueue: 0: ->om_len:             193
W (101362) spBleGattServiceWriteToQueue: 0: ->om_omp:             0x40839F58
W (101363) spBleGattServiceWriteToQueue: 0: ->->omp_databuf_len:  240
W (101363) spBleGattServiceWriteToQueue: 0: ->->omp_pool:         0x40839F28
W (101363) spBleGattServiceWriteToQueue: 0: ->->->mp_block_size:  256
W (101363) spBleGattServiceWriteToQueue: 0: ->->->mp_num_blocks:  0
W (101364) spBleGattServiceWriteToQueue: 0: ->->->mp_num_free:    1
W (101364) spBleGattServiceWriteToQueue: 0: ->->->mp_min_free:    0
W (101364) spBleGattServiceWriteToQueue: 0: ->->->mp_flags:       0x03
W (101364) spBleGattServiceWriteToQueue: 0: ->->->mp_membuf_addr: 0x00000000
W (101365) spBleGattServiceWriteToQueue: 0: ->->->mp_list (next): 0x00000000
W (101365) spBleGattServiceWriteToQueue: 0: ->->->slh_first:      0x00000000
W (101365) spBleGattServiceWriteToQueue: 0: ->->->name:           host_mbuf_1
W (101366) spBleGattServiceWriteToQueue: 0: ->->omp_next:         0x40839F98
W (101366) spBleGattServiceWriteToQueue: 0: ->om_next:            0x40849D00
W (101366) spBleGattServiceWriteToQueue: 1: om:                   0x40849D00
W (101366) spBleGattServiceWriteToQueue: 1: ->om_data:            0x40849D10
W (101367) spBleGattServiceWriteToQueue: 1: ->om_flags:           0x00 
W (101367) spBleGattServiceWriteToQueue: 1: ->om_pkthdr_len:      0
W (101367) spBleGattServiceWriteToQueue: 1: ->om_len:             261
W (101368) spBleGattServiceWriteToQueue: 1: ->om_omp:             0x40839F98
W (101368) spBleGattServiceWriteToQueue: 1: ->->omp_databuf_len:  304
W (101368) spBleGattServiceWriteToQueue: 1: ->->omp_pool:         0x40839F68
W (101368) spBleGattServiceWriteToQueue: 1: ->->->mp_block_size:  320
W (101369) spBleGattServiceWriteToQueue: 1: ->->->mp_num_blocks:  0
W (101369) spBleGattServiceWriteToQueue: 1: ->->->mp_num_free:    1
W (101369) spBleGattServiceWriteToQueue: 1: ->->->mp_min_free:    0
W (101369) spBleGattServiceWriteToQueue: 1: ->->->mp_flags:       0x03
W (101370) spBleGattServiceWriteToQueue: 1: ->->->mp_membuf_addr: 0x00000000
W (101370) spBleGattServiceWriteToQueue: 1: ->->->mp_list (next): 0x00000000
W (101370) spBleGattServiceWriteToQueue: 1: ->->->slh_first:      0x00000000
W (101370) spBleGattServiceWriteToQueue: 1: ->->->name:           host_mbuf_2
W (101371) spBleGattServiceWriteToQueue: 1: ->->omp_next:         0x00000000
W (101371) spBleGattServiceWriteToQueue: 1: ->om_next:            0x00000000
```

I could not find any source code references to `host_mbuf_1` or `host_mbuf_2`. Are these allocated by the BLE controller? Is the controller assuming that other tasks (ie, the host) wouldn't try to do a `os_mbuf_dup()` using the mbuf's it allocated?

I originally started this issue with the intent of asking how I could use compiled mbuf functions instead of using the ROM-based mbuf functions so I could better debug this (custom libble_app.a?). After not finding the mempool names, I'm more confident its the case of the controller not expecting some other task to allocate mbufs from its pool(s).

kenh-sq, 2025-04-07:
If I assume the implementation of os_mbuf_dup() in the ESP32-C6 ROM is the same implementation as in \bt\host\nimble\nimble\porting\nimble\src\os_mbuf.c, then I believe I understand what is causing this bug.

Should I move/open this issue in esp-nimble even though its an issue that also affects the ESP32-C6 ROM implementation?

zhaoweiliang2021, 2025-05-14:
[libble_app.zip](https://github.com/user-attachments/files/20199552/libble_app.zip)
@kenh-sq 1. Unzip the provided libble_app.zip file. Will get a libble_app.a file(based on idf release/v5.4 commit fe75355314cf7beb5a7bfa40a31b0b653cca7fe7 ).
2. Replace the existing libble_app.a file in the specified directory(esp-idf/components/bt/controller/lib_esp32c6/esp32c6-bt-lib/esp32c6).
3. Rebuild your project and perform the test again.
If the issue persists, please provide further details.
We look forward to your updates.
Thanks.

kenh-sq, 2025-05-14:
@zhaoweiliang2021 - I apologize, but I cannot find commit fe75355314cf7beb5a7bfa40a31b0b653cca7fe7 on esp-idf repo. Is it acceptable to test against current release/v5.4 (commit fd37cd46b203786114b6f2793d62fba6da437a62) ?

zhaoweiliang2021, 2025-05-15:
@kenh-sq Maybe the related commit has not been synchronized yet. This lib is also applicable to commit( https://github.com/espressif/esp-idf/commit/fd37cd46b203786114b6f2793d62fba6da437a62).

kenh-sq, 2025-05-15:
@zhaoweiliang2021 For the libble_app.a you provided, I am now getting a different breaking assert when I use `os_mbuf_dup()` within my GATT characteristic write callback:
```
BLE assert: line 488 in function r_os_mbuf_ctrl_mbuf_cnt_decrease, param: 0x0, 0x0
assert failed: osi_assert_wrapper bt.c:482 (0)
Core  0 register dump:
MEPC    : 0x4080090c  RA      : 0x4080eaf0  SP      : 0x408484b0  GP      : 0x40816c54  
TP      : 0x40848710  T0      : 0x37363534  T1      : 0x7271706f  T2      : 0x33323130  
S0/FP   : 0x00000099  S1      : 0x00000001  A0      : 0x408484ec  A1      : 0x408185d5  
A2      : 0x00000001  A3      : 0x00000029  A4      : 0x00000001  A5      : 0x40823000  
A6      : 0x0000000c  A7      : 0x76757473  S2      : 0x00000009  S3      : 0x408485ee  
S4      : 0x408185d4  S5      : 0x00000000  S6      : 0x00000000  S7      : 0x00000000  
S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000  
T3      : 0x6e6d6c6b  T4      : 0x6a696867  T5      : 0x66656463  T6      : 0x62613938  
MSTATUS : 0x00001881  MTVEC   : 0x40800001  MCAUSE  : 0x00000002  MTVAL   : 0x00000000  
MHARTID : 0x00000000  

Stack memory:
408484b0: 0x42021f3a 0x42021fc2 0x420b7f44 0x40814dba 0x408187d4 0x420b7f44 0x40818094 0x420aa65b
408484d0: 0x408187e4 0x408484e4 0x408187e8 0x420a7828 0x408185d4 0x00323834 0x40833a3c 0x65737361
408484f0: 0x66207472 0x656c6961 0x6f203a64 0x615f6973 0x72657373 0x72775f74 0x65707061 0x74622072
40848510: 0x343a632e 0x28203238 0xff002930 0x4001924e 0x00000000 0x00000000 0x00000000 0x42021fd6
40848530: 0x00380030 0x4084c380 0x00000000 0x4200f790 0x00000000 0x00000000 0x4084855c 0x42025590
40848550: 0x00000000 0x4084c380 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
40848570: 0x00000000 0x00000000 0x00000000 0x00000000 0x40823700 0x40823000 0x4084c380 0x400193a6
40848590: 0x40812160 0x40812158 0x40848610 0x00020000 0x00000000 0x40800000 0x000fffff 0x408485c4
408485b0: 0x4081c000 0x00000000 0x00000000 0x40000000 0x00000000 0x000001e8 0x00000000 0x00000000
408485d0: 0x40823700 0x40823000 0x4084c380 0x40802356 0x40823700 0x40823000 0x4084c380 0x42051f36
408485f0: 0x00000000 0x00001efd 0x00000000 0x00000000 0x4081c000 0x00001efd 0x4084c380 0x42051f4c
40848610: 0xfffffffe 0x4081b969 0x00000000 0x42051f72 0x4081c000 0x00001efd 0x4081b8f0 0x42013a88
40848630: 0xa5a5a5a5 0xa5a5a5a5 0x1908b100 0x2aea5400 0x00002710 0x00000064 0x00000000 0x54000000
40848650: 0x00000000 0x00000000 0x00000000 0x00000058 0x5e76aa86 0x00600000 0x00600c37 0x00000000
40848670: 0x00000000 0x00000000 0x00000000 0x01000100 0x40000100 0x00000000 0x4084c380 0x40450000
40848690: 0x00000000 0x40520000 0x00000000 0x40240000 0x00000000 0x00000000 0x47ae147b 0x3f847ae1
408486b0: 0x00000000 0x00000000 0x00000000 0x00010000 0x00010001 0x00000000 0x00000000 0x00001f1b
408486d0: 0x00000000 0x00000000 0x00000000 0x40811ec8 0x00000000 0x00000000 0x00000000 0x00000000
408486f0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
40848710: 0xa5a5a5a5 0xa5a5a5a5 0x00000150 0x40848590 0x00001f1b 0x40819d70 0x40819d70 0x4084871c
40848730: 0x40819d68 0x00000004 0x4083e33c 0x4083e33c 0x4084871c 0x00000000 0x00000015 0x40847718
40848750: 0x70747173 0x00000000 0x00000000 0x00000000 0x40848710 0x00000015 0x00000001 0x00000000
40848770: 0x00000000 0x00000000 0x408258f4 0x4082595c 0x408259c4 0x00000000 0x00000000 0x00000001
40848790: 0x00000000 0x00000000 0x00000000 0x4200713a 0x00000000 0x00000000 0x00000000 0x00000000
408487b0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
408487d0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
408487f0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
40848810: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
40848830: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
40848850: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00001000
40848870: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
40848890: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
```

At commit 79290027545ba8c67bb56017ac799a3554579fbd, I do not get this crash. It is the changes made to the library at commit dd9dbfccd615346abd882f89e90c7e190aa7cd65 where this becomes broken.

From debug trace, it is occurring in call to `r_os_mbuf_free_chain()` in `ble_l2cap_remove_rx()` in `ble_hs_hci_evt_acl_process()`. I could not find source code for `os_mbuf_ctrl_mbuf_cnt_decrease()`, so I don't know what the assert is regarding.

It looks like the name of the mbuf pool that is used for GATT write packets has changed. Previously, it was `host_mbuf_1` and/or `host_mbuf_2`, but this pool was `lec_hmb_mp1` and appears to have the same configuration that `host_mbuf_1` had previously as noted in my original comment on this issue.

Relative to that, please review the pool configuration details reported:
* `mp_num_blocks` is 0.
* `mp_num_free` is 1.

Because `mp_num_free` is non-zero, `os_mbuf_get()`, called from `os_mbuf_dup()`, will take a buffer from the pool, but why is `mp_num_blocks` 0? In my project configuration, I have `CONFIG_BT_NIMBLE_MSYS_1_BLOCK_COUNT` and `CONFIG_BT_LE_MSYS_1_BLOCK_COUNT` set to 64, so I would expect `mp_num_blocks` to be that value. Is this buffer not from the MSYS_1 pool? That used to be the case in code that we had running on ESP32c3 in release/v5.2.

Reproducing this new issue should be possible by:
1. Set up GATT characteristic with write functionality.
2. Make call to `os_mbuf_dup()` in write callback.
3. Send characteristic write to device.

zhaoweiliang2021, 2025-05-16:
@kenh-sq 
This check was added later, so it does not exist in commit 7929002.
Judging from the position of the assert, the recorded number of get/free of the msys buffer does not match, which leads to the assert.
It looks like there might be some issues with the msys buffer count statistics.
[libble_app.zip](https://github.com/user-attachments/files/20237753/libble_app.zip)
The attachment is based on commit 7929002 which fixes the os_mbuf_dup issue. You can continue testing.

Could you provide a minimal demo of the issue so we can check if there are other issues in your real usage?
Thanks

kenh-sq, 2025-05-16:
@zhaoweiliang2021 
I have been running your most recent provided libble_app.a against commit 79290027545ba8c67bb56017ac799a3554579fbd for a couple hours now without issue. I confirmed I could still get the issue with the original libble_app.a at that commit.

I will need to get back to you on setting up demonstration of `r_os_mbuf_ctrl_mbuf_cnt_decrease()` issue. My initial attempt at modifying bleprph example is not showing the issue, so I will need to see what other configuration differences exist.

However, I believe the original issue is resolved with your fix. Here is a way to reproduce the original issue with bleprph example:
1. At commit fd37cd46b203786114b6f2793d62fba6da437a62
2. In `gatt_svr_write()`, add `os_mbuf_dup(om);` before `om_len = OS_MBUF_PKTLEN(om);` line.
3. With nRF Connect, connect to device and send 384 bytes to characteristic.

Result: Store access fault 100% of time (for me).

4. Change libble_app.a to version provided in comment 4 above (https://github.com/espressif/esp-idf/issues/15696#issuecomment-2878530194); rebuild.
5. Retry sending 384 bytes with nRF Connect.

Result: Does not fault.

kenh-sq, 2025-05-23:
@zhaoweiliang2021 

Reproduction for `r_os_mbuf_ctrl_mbuf_cnt_decrease()` issue:
1. Use bleprph example project.
2. In `gatt_svr_write()`, add the following lines before `om_len = OS_MBUF_PKTLEN(om);` line:
```c
struct os_mbuf *om2 = os_mbuf_dup(om);
os_mbuf_free_chain(om2);
```
3. Write data to characteristic.

The underlying issue is that there is no "increase" when `r_os_mbuf_dup()` is called, so calls to `os_mbuf_free_chain()` on a duplicated chain will eventually fault.

zhaoweiliang2021, 2025-05-26:
[libble_app.zip](https://github.com/user-attachments/files/20437085/libble_app.zip)
@kenh-sq 
1. Unzip the provided libble_app.zip file. Will get a libble_app.a file(based on idf release/v5.4 commit fd37cd46b203786114b6f2793d62fba6da437a62 ).
2. Replace the existing libble_app.a file in the specified directory(esp-idf/components/bt/controller/lib_esp32c6/esp32c6-bt-lib/esp32c6).
3. Rebuild your project and perform the test again.
If the issue persists, please provide further details.
Thanks.