espressif/esp-idf #15697: v5.4.1 wrong folder

ae-5th, 2025-03-31:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.4.1

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

PowerShell

### What is the expected behavior?

Installation without problem.

### What is the actual behavior?

During installation there will be message that there is a problem with Python, and in output box there will be message that it is missing folder esp-idf-v5.4.1.

To overcome this problem, until it is fixed. First time installation will create folder esp-idf. So it is necessary to rename it to esp-idf-v5.4.1, and start process again. After installation has been finished, there will be two folders. You can delete esp-idf folder, and esp-idf-v5.4.1 will be renamed to v5.4.1 by installation process.

### Steps to reproduce.

1. Configure ESP-IDF extension
2. Express
3. Select v5.4.1
4. Install

### Build or installation Logs.

```plain

```

dobairoland, 2025-04-01:
@brianignacio5 Vscode installer expects that the ZIP contains a directory in format `esp-idf-v5.4.1`? Can you confirm this please?

This time we had to create a ZIP manually and it looks I made a mistake of putting there directory with name `esp-idf`. Is there an easy fix on your side or we need to upload a new ZIP?

rishi9699, 2025-04-01:
> Yes it does expect this format to download the zip file from extension setup. Fixing can be done but it would not be available until next release of extension.
> 
> Since others releases follow this format I believe the easier solution would be reupload the file.
> 
> Does this behavior also occur in the IDF Installer? I think I just replicated the install behavior in the extension.

I agree with @brianignacio5 , since the older versions of the extension and also all the older releases of esp-idf follow the older naming convention of the directory, the better (also faster?) solution is to make changes here rather than the extension.

dobairoland, 2025-04-01:
@ae-5th @rishi9699 Could you please try again? The ZIP file has been replaced.

rishi9699, 2025-04-01:
> [@ae-5th](https://github.com/ae-5th) [@rishi9699](https://github.com/rishi9699) Could you please try again? The ZIP file has been replaced.

I'm getting a different error now, (doing it through the vscode extension method)

```
Downloaded v5.4.1 (release version). Extracting...

Extracted /home/rishi/esp/esp-idf-v5.4.1.zip in /home/rishi/esp.

spawn xtensa-esp-elf-gdb-no-python ENOENTspawn riscv32-esp-elf-gdb-no-python ENOENTspawn xtensa-esp-elf-gcc ENOENTspawn riscv32-esp-elf-gcc ENOENTspawn esp32ulp-elf-as ENOENTspawn openocd ENOENTError checking xtensa-esp-elf-gdb version
Error: spawn xtensa-esp-elf-gdb-no-python ENOENT
Error checking riscv32-esp-elf-gdb version
Error: spawn riscv32-esp-elf-gdb-no-python ENOENT
Error checking xtensa-esp-elf version
Error: spawn xtensa-esp-elf-gcc ENOENT
Error checking riscv32-esp-elf version
Error: spawn riscv32-esp-elf-gcc ENOENT
Error checking esp32ulp-elf version
Error: spawn esp32ulp-elf-as ENOENT
Error checking openocd-esp32 version
Error: spawn openocd ENOENT
```

dobairoland, 2025-04-01:
@rishi9699 This doesn't seem to be releated to the ZIP file. @brianignacio5, any idea?

brianignacio5, 2025-04-02:
Yes, this seems to be an issue with the setup itself. Please open a GitHub issue with required troubleshooting information in the https://github.com/espressif/vscode-esp-idf-extension/issues/new/choose

dobairoland, 2025-04-02:
@ae-5th Have your issue been resolved?

closed 2025-04-02 by ae-5th