espressif/esp-idf #15698: Graceful Disconnect ?

diplfranzhoepfinger, 2025-04-01:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

When i do a Reboot, 
or press the Reset key, 

in a Situation where the ESP32 is a AP, and a Windows is connected to this AP, 
Windows thinks after Reboot it is still connected. 

BUT ESP32 has no more a assigned IP Address for this Windows PC. 

is there a Way to send to this Windows a graceful "Disconnect" so it will re-connect after reboot. ? 

Thanks, 
Franz

zhangyanjiaoesp, 2025-04-16:
@diplfranzhoepfinger 
When the softAP reboot, it will not send disconnect event to the STA, so the STA thinks after Reboot it is still connected. But if the STA sends data to the softAP, it will trigger the disconnect process, the STA can then choose to reconnect or connect to a different AP.

zhangyanjiaoesp, 2025-04-16:
@boarchuz 
You can add  a delay before calling `esp_wifi_init()` in the application layer.

f-hoepfinger-hr-agrartechnik, 2025-05-01:
> [@boarchuz](https://github.com/boarchuz) You can add a delay before calling `esp_wifi_init()` in the application layer.

this might be the Way. the Reboot is "too fast" ... right ?

zhangyanjiaoesp, 2025-05-07:
Yes, the reboot process is fast.
 @boarchuz @f-hoepfinger-hr-agrartechnik If no more question, can we close this?

f-hoepfinger-hr-agrartechnik, 2025-05-08:
close

zhangyanjiaoesp, 2025-05-08:
@diplfranzhoepfinger Can we close this?

closed 2025-06-10 by Alvin1Zhang