espressif/esp-idf #15701: Poor P2P Ethernet throughput (IDFGH-15004)

robopp55, 2025-04-01:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I'm seeing really poor Ethernet throughput and I'm at a loss as to why. I have a ESP32-Ethernet-Kit running the Ethernet iPerf example project ([https://github.com/espressif/esp-idf/tr ... rnet/iperf](https://github.com/espressif/esp-idf/tree/master/examples/ethernet/iperf)). The ESP32-Ethernet-Kit is directly connected to my PC via RJ45 to a 2.5Gbps USB ethernet adapter. Running an iPerf test I'm seeing ~41Mbps in both directions when testing TCP throughput. To sanity check my results, I found a second USB ethernet adapter and ran the same test - 95Mbps consistently. The only modification I've made to the ESP32 ethernet iPerf example is to set the ESP's IP address to static. Here are the basic modifications to ethernet_iperf_main.c:

```
ESP_ERROR_CHECK(esp_netif_dhcpc_stop(eth_netif));
 
esp_netif_ip_info_t ip;
memset(&ip, 0 , sizeof(esp_netif_ip_info_t));
       
ip.ip.addr = esp_ip4addr_aton("192.168.4.10");
ip.netmask.addr = esp_ip4addr_aton("255.255.255.0");
ip.gw.addr = esp_ip4addr_aton("192.168.4.1");
 
ESP_ERROR_CHECK(esp_netif_set_ip_info(eth_netif, &ip));
```

Is an Ethernet throughput of 41% of link budget expected? That seems really really low to me. I'd expect to see ~95Mbps.

TIA!

kostaond, 2025-04-03:
Please check if your "2.5Gbps USB Ethernet adapter" is USB3.0 capable and check if it is connected to USB3.0 port. I can confirm the performance drop to ~40Mbps when USB2.0 is used. It could be USB2.0 hub or PC port in path or the USB adapter itself.

robopp55, 2025-04-08:
Hi @kostaond,
Thank you for your reply. The USB ethernet adapter I'm using is USB-C plugged directly into a USB-C port on my laptop and I just confirmed that it's a USB version 3.0 device via the USBDeview application. Additionally, I've confirmed using a second USB Ethernet adapter and two iPerf instances on my laptop that the port is capable of 100Mbps.

kostaond, 2025-04-08:
Strange... Just to confirm, you are able to reach ~95 Mbps with one USB adapter but with the same setup but different USB adapter you are able to achieve just ~41 Mbps. Is that right?

robopp55, 2025-04-08:
Hi @kostaond ,
Not sure I understand you question. Hopefully this clarifies things for you -

Setup 1 - USB-C to Ethernet adapter connected to an ESP32-Ethernet-Kit. I can only get ~41Mbps using iPerf running on my Windows laptop and the ESP32-Ethernet-Kit. In an effort to try to figure out why my throughput was so slow I tested using setup 2 below.

Setup 2 - Same USB-C Ethernet adapter connected to a USB type A Ethernet adapter. Both adapters are connected to the same laptop. Running two instances of iPerf on my laptop I'm seeing 95Mbps transfer rates.

I'd expect the ESP32-Ethernet-Kit can do 95Mbps, but that doesn't appear to be the case and I'm trying to figure out why before I abandon the ESP32.

robopp55, 2025-04-08:
Hi @kostaond,

That's correct. 

I just ran the setup 1 test using the other USB adapter connected to the ESP32-Ethernet-Kit and I'm now seeing seeing 70Mbps. However, it's a USB2.0 device. So it appears that my USB-C Ethernet adapter is throttling the network? That's surprising. Have you seen this before?

closed 2025-04-09 by kostaond